dariuszseweryn/RxAndroidBle #191: Service Discovery Times Out After BluetoothGatt Cache Refresh

kenwdelong, 2017-05-09:
### Summary
I am having issues with some phones caching services and characteristics, so I implemented the BluetoothGatt cache refresh noted here https://github.com/Polidea/RxAndroidBle/issues/146.  However, now I cannot perform service discovery.<br />
#### Steps to reproduce actual result
This is my connection code:
```java
mBleConnection = mBleDevice.establishConnection(false)
		.takeUntil(mDisconnectTriggerSubject.get())
		.compose(new ConnectionSharingAdapter())
		.subscribeOn(Schedulers.io());

Subscription sub = mBleConnection
		.flatMap(rxBleConnection -> rxBleConnection.queue(new GattCacheRefreshOperation()))
		.observeOn(Schedulers.io())
		.subscribe(v -> {
			discoverServicesAndCharacteristics();
		}, throwable -> {
			mServiceDiscoveryCompleteSubject.onError(throwable);
		});
```
The service discovery method calls `mBleConnection.flatMap(RxBleConnection::discoverServices)` and goes on to process those results.

My refresh operation class implements `RxBleRadioOperationCustom` and looks like
```java
@NonNull
@Override
public Observable<Void> asObservable(BluetoothGatt bluetoothGatt,
                         RxBleGattCallback rxBleGattCallback,
                         Scheduler scheduler) throws Throwable {

	return Observable.fromCallable(() -> refreshDeviceCache(bluetoothGatt))
			.subscribeOn(scheduler)
			;
}

private Void refreshDeviceCache(BluetoothGatt gatt) {
	// from http://stackoverflow.com/questions/22596951/how-to-programmatically-force-bluetooth-low-energy-service-discovery-on-android
	try {
		BluetoothGatt localBluetoothGatt = gatt;
		Method localMethod = localBluetoothGatt.getClass().getMethod("refresh", new Class[0]);
		if (localMethod != null) {
			boolean bool = ((Boolean) localMethod.invoke(localBluetoothGatt, new Object[0])).booleanValue();
			Timber.i("Gatt cache refresh successful: [%b]", bool);
		}
	}
	catch (Exception localException) {
		Timber.e("An exception occured while refreshing device");
	}
	return null;
}
```

#### Actual result
The connection code was working fine until I added the cache refresh.  Then I get roughly 20 second timeout and an exception.  Here are the logs:
```
05-09 12:08:08.308 9098-9225/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(994668495)
05-09 12:08:08.308 9098-9173/com.hatchbaby.rest.qa D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(994668495)
05-09 12:08:08.358 9098-9098/com.hatchbaby.rest.qa D/BluetoothGatt: connect() - device: C8:E3:82:B1:4D:9E, auto: false
05-09 12:08:08.358 9098-9098/com.hatchbaby.rest.qa D/BluetoothGatt: registerApp()
05-09 12:08:08.358 9098-9098/com.hatchbaby.rest.qa D/BluetoothGatt: registerApp() - UUID=e659a5de-c47f-452d-90d7-28805113200a
05-09 12:08:08.358 9098-9114/com.hatchbaby.rest.qa D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
05-09 12:08:08.408 9098-9098/com.hatchbaby.rest.qa I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@16efa7f5 time:82316050
05-09 12:08:08.728 9098-9115/com.hatchbaby.rest.qa D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=C8:E3:82:B1:4D:9E
05-09 12:08:08.728 9098-9115/com.hatchbaby.rest.qa D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
05-09 12:08:08.728 9098-9313/com.hatchbaby.rest.qa I/Nightlight: [pool-3-thread-1] Got state [RxBleConnectionState{CONNECTED}] in Nightlight{mName='Ken Batch 2', mMacAddress='C8:E3:82:B1:4D:9E', identity='1a3b5bfb'}
05-09 12:08:08.738 9098-9313/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED (944499391)
05-09 12:08:08.748 9098-9173/com.hatchbaby.rest.qa D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(994668495)
05-09 12:08:08.748 9098-9173/com.hatchbaby.rest.qa D/RxBle#Radio:  STARTED (944499391)
05-09 12:08:08.748 9098-9098/com.hatchbaby.rest.qa D/BluetoothGatt: refresh() - device: C8:E3:82:B1:4D:9E
05-09 12:08:08.748 9098-9098/com.hatchbaby.rest.qa I/GattCacheRefreshOperation: [main] Gatt cache refresh successful: [true]
05-09 12:08:08.748 9098-9251/com.hatchbaby.rest.qa I/Nightlight: [RxIoScheduler-6] Starting service discovery
05-09 12:08:08.748 9098-9173/com.hatchbaby.rest.qa D/RxBle#Radio: FINISHED (944499391)
05-09 12:08:08.748 9098-9413/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED RxBleRadioOperationServicesDiscover(675644243)
05-09 12:08:08.758 9098-9173/com.hatchbaby.rest.qa D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(675644243)
05-09 12:08:08.758 9098-9098/com.hatchbaby.rest.qa D/BluetoothGatt: discoverServices() - device: C8:E3:82:B1:4D:9E
05-09 12:08:09.038 9098-9098/com.hatchbaby.rest.qa D/NightlightsActivity: [main] com.hatchbaby.rest.ui.NightlightsActivity@15c63b37 onDestroy
05-09 12:08:28.758 9098-9173/com.hatchbaby.rest.qa D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(675644243)
05-09 12:08:28.758 9098-9318/com.hatchbaby.rest.qa E/Nightlight: [RxIoScheduler-7] Error discovering Services & Characteristics
                                                                 BleGattCallbackTimeoutException{macAddress=C8:E3:82:B1:4D:9E, bleGattOperationType=BleGattOperation{description='SERVICE_DISCOVERY'}}
                                                                     at com.polidea.rxandroidble.internal.operations.RxBleRadioOperationServicesDiscover$1.call(RxBleRadioOperationServicesDiscover.java:69)
                                                                     at com.polidea.rxandroidble.internal.operations.RxBleRadioOperationServicesDiscover$1.call(RxBleRadioOperationServicesDiscover.java:63)
                                                                     at rx.internal.operators.OnSubscribeDefer.call(OnSubscribeDefer.java:46)
                                                                     at rx.internal.operators.OnSubscribeDefer.call(OnSubscribeDefer.java:35)
                                                                     at rx.Observable.unsafeSubscribe(Observable.java:10346)
                                                                     at rx.internal.operators.OperatorTimeoutBase$TimeoutSubscriber.onTimeout(OperatorTimeoutBase.java:200)
                                                                     at rx.internal.operators.OperatorTimeout$1$1.call(OperatorTimeout.java:41)
                                                                     at rx.internal.schedulers.EventLoopsScheduler$EventLoopWorker$2.call(EventLoopsScheduler.java:189)
                                                                     at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:55)
                                                                     at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:422)
                                                                     at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                                                                     at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:152)
                                                                     at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:265)
                                                                     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
                                                                     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
                                                                     at java.lang.Thread.run(Thread.java:818)
```
As noted in the documentation for `RxBleRadioOperationCustom`, it seems like the radio queue lock is not released and the service discovery can't proceed.  However, the observable in the refresh operation certainly emits onCompleted() (I have put log statements in `doOnCompleted()` to verify).  Also, looking at the RxBle logs above, it seems that the radio operation associated with the cache clearing (944499391) has completed before the service discovery begins.
<br />
#### Expected result
Service discovery will proceed and not hang.<br />

dariuszseweryn, 2017-05-11:
Hello @kenwdelong 

Actually your custom operation is finished successfully and the service discovery starts as intended as you can see here:

    05-09 12:08:08.748 9098-9173/com.hatchbaby.rest.qa D/RxBle#Radio: FINISHED (944499391)
    05-09 12:08:08.748 9098-9413/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED RxBleRadioOperationServicesDiscover(675644243)
    05-09 12:08:08.758 9098-9173/com.hatchbaby.rest.qa D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(675644243)
    05-09 12:08:08.758 9098-9098/com.hatchbaby.rest.qa D/BluetoothGatt: discoverServices() - device: C8:E3:82:B1:4D:9E
    05-09 12:08:09.038 9098-9098/com.hatchbaby.rest.qa D/NightlightsActivity: [main] com.hatchbaby.rest.ui.NightlightsActivity@15c63b37 onDestroy
    05-09 12:08:28.758 9098-9173/com.hatchbaby.rest.qa D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(675644243)

Unfortunately the Android BLE stack behaviour is not documented in regards to using of `BluetoothGatt.refresh()` function. That is the main reason why this functionality was not yet introduced to the library.

As you can see in the logs Android does not notify the app about the end of the process which is a bogus behaviour and is caused by improper usage of `BluetoothGatt.refresh()` function (as far as I know).

If you have time - feel free to investigate what is going on and post your findings here.

kenwdelong, 2017-05-11:
Any suggestions on the best way to dig into this?  Logging is all the way up, but not much help...

dariuszseweryn, 2017-05-11:
Unfortunately I cannot help much at this moment. I just remember that after refresh the `ServiceDiscoveryManager` should not check `BluetoothGatt.getServices()` for cached services.

dariuszseweryn, 2017-05-16:
@kenwdelong Any news? Were you able to overcome the difficulties?

kenwdelong, 2017-05-16:
No, I was not.  I put the cache clearing just after connecting, just before service discovery, and just before disconnecting.  In the first two cases, service discovery would hang.  In the last case, the error rate went way up for the subsequent connection.

We are using Nordic's DFU library for firmware upgrades.  I noticed in their code that they call this method after the upgrade completes (with comment noting that services and characteristics will be different after the DFU), so I tried clearing the cache just before doing a firmware upgrade.  That seems to have helped a lot.

I have log levels all the way up but didn't see much useful.  I was going to see whether I could debug into it (although I suppose that would drastically change all timings), but ran out of time.  I suppose I could also look at the bluetooth hci log.  I'm not sure where else to look.

ivanlardis, 2017-06-07:
@kenwdelong, hello, i have same problem. Please can you show me an example of the code where you solved this problem

kenwdelong, 2017-06-07:
I was unable to solve it.  As noted above, I only clear the cache now just before a firmware upgrade, and that seems to be somewhat benign.  

I wasn't sure how to debug further, except maybe the BLE HCI logs, but didn't have time for that.

dariuszseweryn, 2017-06-07:
@kenwdelong Could you describe the exact sequence of what you were doing? Maybe a log with the setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE);`?

kenwdelong, 2017-06-08:
Here is what I get on VERBOSE.  Service discovery just never returns.  We have a 15-second timeout in the app so customers don't hang; generally service discovery returns in 1-3 seconds without the cache clear operation.
```
06-08 16:35:05.978 31163-31163/com.hatchbaby.rest.qa I/NightlightManager: [main] Created new nightlight, Nightlight{mName='Ken Batch 2', mMacAddress='C8:E3:82:B1:4D:9E', identity='9bb5505'}
06-08 16:35:05.978 31163-31163/com.hatchbaby.rest.qa I/NightlightManager: [main] NightlightManager.connect() to Nightlight{mName='Ken Batch 2', mMacAddress='C8:E3:82:B1:4D:9E', identity='9bb5505'}
06-08 16:35:05.980 31163-31163/com.hatchbaby.rest.qa I/Nightlight: [main] Nightlight.connect()
06-08 16:35:05.992 31163-31163/com.hatchbaby.rest.qa I/SplashActivity: [main] Connecting to default nightlight: [C8:E3:82:B1:4D:9E]
06-08 16:35:05.998 31163-31224/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED RxBleRadioOperationScanApi21(115264771)
06-08 16:35:05.998 31163-31213/com.hatchbaby.rest.qa D/RxBle#Radio:  STARTED RxBleRadioOperationScanApi21(115264771)
06-08 16:35:06.001 31163-31163/com.hatchbaby.rest.qa I/BluetoothAdapter: getBluetoothLeScanner
06-08 16:35:06.003 31163-31233/com.hatchbaby.rest.qa E/OpenGLRenderer: allen debug liyu Key: 5242945
06-08 16:35:06.003 31163-31163/com.hatchbaby.rest.qa D/BluetoothAdapter: STATE_ON
06-08 16:35:06.003 31163-31233/com.hatchbaby.rest.qa E/OpenGLRenderer: allen debug liyu Key: 34359738371
06-08 16:35:06.022 31163-31178/com.hatchbaby.rest.qa D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5 mClientIf=0
06-08 16:35:06.026 31163-31213/com.hatchbaby.rest.qa D/RxBle#Radio: FINISHED RxBleRadioOperationScanApi21(115264771)
06-08 16:35:06.171 31163-31163/com.hatchbaby.rest.qa I/Nightlight: [main] Got state [RxBleConnectionState{DISCONNECTED}] in Nightlight{mName='Ken Batch 2', mMacAddress='C8:E3:82:B1:4D:9E', identity='9bb5505'}
06-08 16:35:06.171 31163-31163/com.hatchbaby.rest.qa I/Nightlight: [main] Scanned device
06-08 16:35:06.174 31163-31163/com.hatchbaby.rest.qa I/Nightlight: [main] Got state [RxBleConnectionState{CONNECTING}] in Nightlight{mName='Ken Batch 2', mMacAddress='C8:E3:82:B1:4D:9E', identity='9bb5505'}
06-08 16:35:06.183 31163-31163/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(156254365)
06-08 16:35:06.184 31163-31213/com.hatchbaby.rest.qa D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(156254365)
06-08 16:35:06.189 31163-31163/com.hatchbaby.rest.qa V/RxBle#BleConnectionCompat: Connecting without reflection
06-08 16:35:06.190 31163-31163/com.hatchbaby.rest.qa D/BluetoothGatt: connect() - device: C8:E3:82:B1:4D:9E, auto: false
06-08 16:35:06.190 31163-31163/com.hatchbaby.rest.qa D/BluetoothGatt: registerApp()
06-08 16:35:06.190 31163-31163/com.hatchbaby.rest.qa D/BluetoothGatt: registerApp() - UUID=a161ade6-480f-47d9-9861-43d0dd65c9b4
06-08 16:35:06.200 31163-31177/com.hatchbaby.rest.qa D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
06-08 16:35:06.200 31163-31163/com.hatchbaby.rest.qa I/BluetoothAdapter: getBluetoothLeScanner
06-08 16:35:06.205 31163-31163/com.hatchbaby.rest.qa D/BluetoothAdapter: STATE_ON
06-08 16:35:06.575 31163-31178/com.hatchbaby.rest.qa D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=C8:E3:82:B1:4D:9E
06-08 16:35:06.576 31163-31178/com.hatchbaby.rest.qa D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
06-08 16:35:06.591 31163-31279/com.hatchbaby.rest.qa I/Nightlight: [pool-2-thread-1] Got state [RxBleConnectionState{CONNECTED}] in Nightlight{mName='Ken Batch 2', mMacAddress='C8:E3:82:B1:4D:9E', identity='9bb5505'}
06-08 16:35:06.595 31163-31279/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED (89878948)
06-08 16:35:06.611 31163-31213/com.hatchbaby.rest.qa D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(156254365)
06-08 16:35:06.612 31163-31213/com.hatchbaby.rest.qa D/RxBle#Radio:  STARTED (89878948)
06-08 16:35:06.629 31163-31163/com.hatchbaby.rest.qa D/BluetoothGatt: refresh() - device: C8:E3:82:B1:4D:9E
06-08 16:35:06.632 31163-31163/com.hatchbaby.rest.qa I/GattCacheRefreshOperation: [main] Gatt cache refresh successful: [true]
06-08 16:35:06.634 31163-31223/com.hatchbaby.rest.qa I/Nightlight: [RxIoScheduler-2] Starting service discovery (name is [Ken Batch 2])
06-08 16:35:06.634 31163-31213/com.hatchbaby.rest.qa D/RxBle#Radio: FINISHED (89878948)
06-08 16:35:06.640 31163-31288/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED RxBleRadioOperationServicesDiscover(38437862)
06-08 16:35:06.641 31163-31213/com.hatchbaby.rest.qa D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(38437862)
06-08 16:35:06.648 31163-31163/com.hatchbaby.rest.qa D/BluetoothGatt: discoverServices() - device: C8:E3:82:B1:4D:9E
06-08 16:35:20.995 31163-31269/com.hatchbaby.rest.qa W/NightlightManager: [RxComputationScheduler-4] Connection process failed! java.util.concurrent.TimeoutException
06-08 16:35:20.995 31163-31269/com.hatchbaby.rest.qa I/NightlightManager: [RxComputationScheduler-4] NightlightManager.disconnect() for Nightlight{mName='Ken Batch 2', mMacAddress='C8:E3:82:B1:4D:9E', identity='9bb5505'}
06-08 16:35:20.996 31163-31269/com.hatchbaby.rest.qa D/NightlightManager: [RxComputationScheduler-4] Disconnecting from Nightlight{mName='Ken Batch 2', mMacAddress='C8:E3:82:B1:4D:9E', identity='9bb5505'}
06-08 16:35:20.996 31163-31269/com.hatchbaby.rest.qa I/Nightlight: [RxComputationScheduler-4] Disconnecting: Nightlight.disconnect()
[... cut ...]
06-08 16:35:21.005 31163-31269/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(58317343)
06-08 16:35:21.009 31163-31269/com.hatchbaby.rest.qa I/Nightlight: [RxComputationScheduler-4] Nightlight state: RxBleConnectionState{DISCONNECTED}
06-08 16:35:21.009 31163-31269/com.hatchbaby.rest.qa I/NightlightManager: [RxComputationScheduler-4] mDisconnectObservable set to null
06-08 16:35:21.010 31163-31269/com.hatchbaby.rest.qa I/Nightlight: [RxComputationScheduler-4] Nightlight state: RxBleConnectionState{DISCONNECTED}
```

dariuszseweryn, 2017-06-09:
I see. And could you paste the logs from what you are actually using now and what works for you?

dariuszseweryn, 2017-07-17:
@kenwdelong If you have not moved on yet — could you check what would happen if you would use this:
```
@NonNull
@Override
public Observable<Void> asObservable(BluetoothGatt bluetoothGatt,
                         RxBleGattCallback rxBleGattCallback,
                         Scheduler scheduler) throws Throwable {

	return Observable.fromCallable(() -> refreshDeviceCache(bluetoothGatt))
			.delay(500, TimeUnit.MILLISECONDS) // << *** Add this line ***
			.subscribeOn(scheduler)
			;
}

private Void refreshDeviceCache(BluetoothGatt gatt) {
	// from http://stackoverflow.com/questions/22596951/how-to-programmatically-force-bluetooth-low-energy-service-discovery-on-android
	try {
		BluetoothGatt localBluetoothGatt = gatt;
		Method localMethod = localBluetoothGatt.getClass().getMethod("refresh", new Class[0]);
		if (localMethod != null) {
			boolean bool = ((Boolean) localMethod.invoke(localBluetoothGatt, new Object[0])).booleanValue();
			Timber.i("Gatt cache refresh successful: [%b]", bool);
		}
	}
	catch (Exception localException) {
		Timber.e("An exception occured while refreshing device");
	}
	return null;
}
```

kenwdelong, 2017-07-19:
Wow.  That makes it work.  Without the delay, it reliably hangs.  With the delay, service discovery succeeds!

dariuszseweryn, 2017-07-19:
Yeah, that is the problem. Apparently `BluetoothGatt.refresh()` works as everything else - it's asynchronous but there is no callback to be sure when it is ready for next operations. This may be further complicated by how the Android OS works with bonded devices. Either way - adding a reference to the original issue #53 so the above code will be discoverable.
I plan to add this functionality to the core library in `1.4.0` or `1.5.0` when the new API will be ready #239. The most simple to implement would be to refresh the gatt either right after the connecting is kicked off / connected or right after disconnection. Refreshing the gatt in the middle of communication would need significant refactoring and probably a totally new approach to the API to make it clear to the users of what is actually going on and how it should be handled.
Closing for now as the issue is resolved by adding a delay. If you will have anything to add - post here #53

closed 2017-07-19 by dariuszseweryn

zcelaloglu, 2018-07-23:
@kenwdelong is there a way to implement this solution as soon as the first search/connection fails?

AFAIS, you implement this after the connection is successful