dariuszseweryn/RxAndroidBle #193: Weird behavior on API >=23

marciogranzotto, 2017-05-11:
### Summary
<br /> When scanning for devices I'm getting devices with the wrong mac address on API >=23
#### Preconditions
<br /> Use a API >=23 device
#### Steps to reproduce actual result
```
 fun scanAppliances(): Observable<Appliance> {
        return rxBleClient.scanBleDevices()
                .flatMap flatMap@ { scanResult ->
                    val uuids = AdvertisedServiceUUIDExtractor().extractUUIDs(scanResult.scanRecord)
                    val uuid = uuids.firstOrNull { servicesUUIDs.contains(it) } ?: return@flatMap null //this just filters out unwanted uuids
                    val bleDevice = scanResult.bleDevice
                    Timber.tag("scanAppliances")
                    Timber.d("--------------------------------------------------------\nnew scanResult")
                    Timber.d("device name: ${bleDevice.name} with macAddrs: ${bleDevice.macAddress} rssi: ${scanResult.rssi} UUID: $uuid")
                    val appliance = Appliance( //this is just my internal POJO
                            bleDevice.name,
                            uuid.toString(),
                            null,
                            bleDevice.macAddress
                    )
                    Timber.d("appliance created")
                    return@flatMap Observable.just(appliance)
                }
    }
```
#### Actual result
<br />
When scanning for devices on API 22 I only get one reading of each device, just like on this log:
[image]
This is what I'm expecting.
#### Expected result
<br /> When scanning on API 23 and API 25 devices, I get a endless stream of readings of the same device. The bigger problem is that sometimes the device appears with the wrong mac address. Usings `nF Connect` app I can see that the wrong address actually are from another devices that are near
[image]

marciogranzotto, 2017-05-12:
Maybe this is related to #74 ?

dariuszseweryn, 2017-05-12:
I do not think that is related. Some devices are returning only one scanned advertisement per device address. Other devices have a really poor performance of BLE. It is hard to compare unless you have the same device type with two different OS versions.

marciogranzotto, 2017-05-12:
The biggest problem is not really the number of scans, but the mac address that some times is wrong.
But I cannot figure out what is the problem, I've tested it on around 10 devices and only 2 or 3 had this problem. For example, on the Nexus 5X on 7.1.1 it works properly but on the Nexus 6P on 7.1.1 I get this problem.

dariuszseweryn, 2017-05-12:
Are you sure that you did not clone the advertisement in `nRF Connect` by an accident?

marciogranzotto, 2017-05-12:
Yes, the the wrong addresses are always different. Most of the time they show on nRF Connect as a nameless Apple device, but been in a office full of macs and iPhones, that can mean nothing.

dariuszseweryn, 2017-05-12:
Looks like a development environment where people are mocking advertisement packets. Either way I do not think it is a library problem.

marciogranzotto, 2017-05-12:
Not really, I'm the only one dealing with bluetooth right now, I can guarantee that there's no mocking advertisement packets here.
I'll test @gryffindor14 branch that implements Scan Compat and see if it fixes my problem

dariuszseweryn, 2017-05-12:
As far as I know on Android 5.0+ the old API is using the new scan under the hood anyway but give it a try. I expect to have the new API scan implemented in the library somewhere next week. At least in the Snapshot.

marciogranzotto, 2017-05-15:
Unfortunately that branch didn't fixed my problem. I've also got the same problem using the sample scanning app provided in this repo.

dariuszseweryn, 2017-05-15:
Unfortunately it does not look like something that the library may fix if it works on another device properly. I cannot do more than suggest you to open an issue on Google's issue tracker: https://issuetracker.google.com/issues?q=nexus%206p%20ble%20scan

Closing for now. If you will have more info or an idea what we could do from the library side - feel free to comment / reopen.

Best Regards

closed 2017-05-15 by dariuszseweryn