dariuszseweryn/RxAndroidBle #195: Characteristic change event not received

gryffindor14, 2017-05-18:
### Summary
<br />
A characteristic change event is logged but never received through the subscribed observable when a CharacteristicChangeEvent is fired while the notification is being setup

#### Preconditions
<br />
A RxBleDevice is connected and has a characteristic with indication property
#### Steps to reproduce actual result
1. connect to a BluetoothDevice
2.setup indications

Observable.combineLatest(
connection.setupIndication(UUID.fromString(00008a82-0000-1000-8000-00805f9b34fb)),
connection.setupIndication(UUID.fromString(00008a91-0000-1000-8000-00805f9b34fb)),
(indication1, indication2) -> { return indication1;})
.flatMap(observable-> observable)
.doOnNext(bytes -> { log.d(bytes) })
.subscribe();

3. As soon as onDescriptor write is fired a onChangeEvent happens that the observable never sees 
#### Actual result
<br />
 No log
#### Expected result
<br />
 Log of bytes

Actual Log:
Galaxy s6 6.0.1
D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(149819350)
 D/BluetoothGatt: onClientConnParamsChanged() - Device=98:7B:F3:6B:F9:99 interval=6 status=0
D/RxBle#Radio:   QUEUED RxBleRadioOperationServicesDiscover(8272968)
 D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(8272968)
 D/BluetoothGatt: discoverServices() - device: 98:7B:F3:6B:F9:99
 D/BluetoothGatt: onSearchComplete() = Device=98:7B:F3:6B:F9:99 Status=0
D/RxBle#BluetoothGatt: onServicesDiscovered status=0
D/BluetoothGatt: setCharacteristicNotification() - uuid: 00008a82-0000-1000-8000-00805f9b34fb enable: true
 D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(72475096)
D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(8272968)
D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(72475096)
D/BluetoothGatt: onClientConnParamsChanged() - Device=98:7B:F3:6B:F9:99 interval=39 status=0
D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
**D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00008a82-0000-1000-8000-00805f9b34fb**
D/RxBle#Radio: FINISHED RxBleRadioOperationDescriptorWrite(72475096)

dariuszseweryn, 2017-05-22:
Hello @gryffindor14 
Could you elaborate more on your use case?

gryffindor14, 2017-05-22:
Hey @dariuszseweryn, thanks for taking a look at this. For us, yes, receiving the indication is important to the transmission. 

Our use case is that when we subscribe to the characteristic it fires a characteristic change event with a value. We then write to another characteristic which causes us to receive another characteristic change event with another value that we combine into a key. 

I have had better results when writing the client descriptor myself while starting to observe the change event from RxBleGattCallback.

connection.setupNotification(characteristic, NotificationSetupMode.COMPAT)
    .flatmap(observable-> Observable.combineLatest(observable, writeClientDescriptor(connection), 
          return bytes; //from observable
)

dariuszseweryn, 2017-05-22:
That is an unusual use case. Usually the notification is either triggered by a characteristic write or it is a constant stream of notifications in which situation it is not fatal if one of them is missed. 

To be honest I do not have an idea how to implement your use case with the current contract which is `Returned observable will emit Observable<byte[]> once the notification setup has been completed.`.
Since your device is emitting the first value right after the notification is established — it may be missed due to a race condition. Unfortunately buffering of the notifications is not an option because then it is possible that the library would cause out of memory exceptions.

dariuszseweryn, 2017-05-22:
@gryffindor14 Feel free to add any thoughts you may have. I will think about your use case but I cannot tell when (or how) it will be possible to address it correctly. I do not consider it a top priority as it is quite easy to workaround.

Curiosity — does something like this work for you?
```
connection.setupNotification(uuid)
                .flatMap(observable -> Observable.merge(
                        observable,
                        connection.readCharacteristic(uuid)
                ))
                // ...
```
So after establishing the notification you would re-read the characteristic.

gryffindor14, 2017-05-22:
The characteristic doesn't have a read property associated with it so unfortunately that will not work. The workaround I mention in my last comment seems to be working very well.

Related to that workaround, is there any way to remove the dependency on dagger or a way to generate the classes from the command line. I am porting my library to Xamarin and getting ClassNotFoundExceptions for the generated dagger components. 

Version 1.1.0 works with Xamarin fine, but the setCharacteristicNotification method with the NotificationSetupMode value is not available until version 1.2.0 which also bundles the Dagger components

dariuszseweryn, 2017-05-22:
Not sure. Support for Xamarin was not intended when creating this library.
Mentioning @uKL so he will be aware of the issue.

gryffindor14, 2017-05-22:
Thanks very much for your help. I got the xamarin binding working, I was just missing a dependency jar. 

The workaround is working for me and I will probably be using it as my base for setting up notifications and indications so I don't miss any in the future. I'm going to close this for now

closed 2017-05-22 by gryffindor14

mswokk, 2017-12-05:
Hi @dariuszseweryn 
I agree it's unusual that using BLE notification for important messaging.
But regardless it, registering some callback function before requesting data is much more make sense.
So I modified some part of rxandroidble and it fixed my App problem.

**I hope you to review my idea and please answer so that I can make a pull request.**
[Here is my change](https://github.com/mswokk/RxAndroidBle/commit/a4e1cd8089e86ae7960d736f47c4ea9e65a505c1) (I'm quite new to RX programming, not so sure my change is best)

* Appendix: my problem was..
I build an RN app with react-native-ble-plx.
I expect to receive 5 characteristic change consequentially from BLE device by notification request.
On iPhone, it always works well. 
But on Android, app always missed first change due to fails to check changedCharacteristicSerializedPublishRelay.hasObservers() in NotificationAndIndicationManager.java
After that fix, now android also works well.

Thank you.

dariuszseweryn, 2017-12-05:
Hello @mswokk 
Feel free to open a PR where we could discuss it. Here it is not very visible—it is an old and already closed issue.
Best Regards

shuaisunny, 2018-08-11:
@gryffindor14  @mswokk  i'm meet the same issue on rxandroidble version 1.1.0，can you tell me how to fix this? preferably with 1.1.0, 1.2.0 is also ok.That code is incomplete, i can't got u well, thanks.

gryffindor14, 2018-08-11:
It will require 1.2.0. I believe I posted how I got it to work in the
thread. It requires setting up the notification with the NotificationCompat
option.
connection.setupNotification(uuid, NotificationSetupMode.COMPAT)
.flatmap { notification -> return
notification.mergeWith(connection.writeDescriptor(descriptor_uuid,
BluetoothGattDescriptor.ENABLE_NOTIFICATIONS).ignoreElements())

note the ignoreElements() method at the end. This merges the notification
byte[] stream with the descriptorWrite byte[] stream but the write for the
descriptor is ignored.

combineLatest() also works as mentioned in the thread.
connection.writeDescriptor is a completeable in the rxjava2 version and
therefore does not emit any values. Good luck!

Griffin