dariuszseweryn/RxAndroidBle #196: scanBleDevices() with multiple service UUIDs doesn't discover any specified device

ipodib, 2017-05-19:
### Summary
<br /> When I trying to discover devices with multiple service UUIDs, it doesn't discover any of the specified  services.
#### Steps to reproduce actual result
1. Scan devices using several UUIDs, for example:
`scanBleDevices("uuid1", "uuid2")`
2. Turn only one related device, for example device that has `uuid1`.
#### Actual result
Devices is not discovered at all.
#### Expected result
Device should be discovered, as I set UUIDs that allowed to be filtered.

Note:
When I don't specify UUIDs, device is discovered successfully.

uKL, 2017-05-19:
Does the device advertise both UUIDs?

ipodib, 2017-05-19:
@uKL I'm not sure what do you mean. I have two different devices, in my case it's health devices (activity tracker and blood pressure). Each device has it's own service UUID. I want discover only those two devices. And when I specify those two service UUIDs it's not discovered at all.

Hope I answered to your question.

uKL, 2017-05-19:
Both in the Android's API and in our library it is required that all UUID are matched for each filtered device. This is a standard Android's behavior but may be not so clear, so maybe we should add it to the documentation. Thanks for pointing that out.

For now, I see some options for you:

1. Scan without filters and use RxJava's `filter`. Parse the scan record yourself. You can take a look at the `UUIDUtil` class but keep in mind that it's internal.
2. As a small spoiler, we'll soon introduce more advanced filtering with the API based on API 21 but with compatibility package, which may be more suitable for you.

ipodib, 2017-05-19:
Thanks for information. First option may work for me for now. 

But I keept debugging and found interesting thing in RxBleRadioOperationScan:
```
if (!isFilterDefined || uuidUtil.extractUUIDs(scanRecord).containsAll(filterUuids)) {
    RxBleRadioOperationScan.this.onNext(new RxBleInternalScanResult(device, rssi, scanRecord));
}
```

It looks like `uuidUtil.extractUUIDs(scanRecord).containsAll(filterUuids)` always returns false in my case. Is that correct way of filtering devices? Or I'm looking in wrong direction?

uKL, 2017-05-19:
Could you provide example UUIDs of your devices and your filter setting?

ipodib, 2017-05-19:
Yes, sure. My filtering settings is:

```
"00001808-0000-1000-8000-00805f9b34fb"
"00001810-0000-1000-8000-00805f9b34fb"
```

And device UUID device is:
"00001810-0000-1000-8000-00805f9b34fb"

I assume containsAll function that used in RxBleRadioOperationScan class will not work in this case.

uKL, 2017-05-19:
Well, `extractUUID` returns `00001810-0000-1000-8000-00805f9b34fb` and your filter setting requires that both `00001808-0000-1000-8000-00805f9b34fb` and `00001810-0000-1000-8000-00805f9b34fb` are present, which are not.

What's the problem then? :)

ipodib, 2017-05-19:
Now I understand that feature. I was confused, so I assumed that it will discover all devices and make filtering automatically. Nevertheless I can achieve my goal by using RxJava's `filter`.

Thanks.

closed 2017-05-19 by uKL