dariuszseweryn/RxAndroidBle #197: What could cause hundreds of "scan result" reports?

RobLewis, 2017-05-22:
### Summary
Scan apparently doesn't terminate and keeps generating results. 
<br />
#### Preconditions
Start a scan
<br />
#### Steps to reproduce actual result
1. Start a scan
2. Try to stop it by unsubscribing from scan
3. 
#### Actual result
See log: hundreds of scan results of the same MAC address 
<br />
#### Expected result
One result. 
<br />
Log (it goes on for a long time): 

```
05-21 18:25:55.876 28182-28182/io.aeroscope.aeroscope D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(1065921514)
05-21 18:25:55.877 28182-28230/io.aeroscope.aeroscope D/RxBle#Radio:  STARTED RxBleRadioOperationScan(1065921514)
05-21 18:25:55.884 28182-28182/io.aeroscope.aeroscope D/MainActivity: Finished startScan()
05-21 18:25:55.885 28182-28182/io.aeroscope.aeroscope D/BluetoothAdapter: startLeScan(): null
05-21 18:25:55.888 28182-28182/io.aeroscope.aeroscope D/BluetoothAdapter: 331935856: getState(). Returning 12
05-21 18:25:55.890 28182-28182/io.aeroscope.aeroscope D/BluetoothAdapter: 331935856: getState(). Returning 12
05-21 18:25:55.893 28182-28182/io.aeroscope.aeroscope D/BluetoothAdapter: 331935856: getState(). Returning 12
05-21 18:25:55.899 28182-28199/io.aeroscope.aeroscope D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=2
05-21 18:25:55.912 28182-28182/io.aeroscope.aeroscope I/BluetoothLeScanner: startRegisteration: mLeScanClients={android.bluetooth.BluetoothAdapter$2@1b92d924=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@11879c8d, android.bluetooth.BluetoothAdapter$2@18ca146e=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@e9b61a5}
05-21 18:25:55.914 28182-28230/io.aeroscope.aeroscope D/RxBle#Radio: FINISHED RxBleRadioOperationScan(1065921514)
05-21 18:25:55.941 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:55.948 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462259589768
05-21 18:25:55.957 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:55.967 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462276372529
05-21 18:25:55.974 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:55.982 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462293058001
05-21 18:25:55.990 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.000 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462309690719
05-21 18:25:56.010 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.016 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462326370077
05-21 18:25:56.025 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.033 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462343051496
05-21 18:25:56.043 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.049 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462359740865
05-21 18:25:56.061 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.065 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462376415093
05-21 18:25:56.077 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.082 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462393097528
05-21 18:25:56.093 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.098 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462409783752
05-21 18:25:56.110 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.115 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462426466431
05-21 18:25:56.126 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.132 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462443155544
05-21 18:25:56.143 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.148 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462459837393
05-21 18:25:56.160 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.165 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462476521133
05-21 18:25:56.176 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.182 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462493207081
05-21 18:25:56.193 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.198 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462509894291
05-21 18:25:56.209 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.215 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462526587513
05-21 18:25:56.226 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.232 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462543261908
05-21 18:25:56.242 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.249 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462559949183
05-21 18:25:56.260 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 requestNextVsync
05-21 18:25:56.265 28182-28209/io.aeroscope.aeroscope D/OpenGLRenderer: DisplayEventReceiver 0xa3b23050 latestVsyncEvent 995462576635866
05-21 18:26:06.066 28182-28199/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=CC:C6:B5:9F:9A:F8, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=[00001530-1212-efde-1523-785feabcd123], mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=DFU Targ], mRssi=-71, mTimestampNanos=1053780009469668}
[... 7 lines cut ...]
05-21 18:26:06.174 28182-28200/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=CC:C6:B5:9F:9A:F8, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=[00001530-1212-efde-1523-785feabcd123], mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=DFU Targ], mRssi=-74, mTimestampNanos=1053780119316360}
05-21 18:26:06.175 28182-28199/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=CC:C6:B5:9F:9A:F8, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=[00001530-1212-efde-1523-785feabcd123], mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=DFU Targ], mRssi=-74, mTimestampNanos=1053780120830129}
05-21 18:26:06.206 28182-28201/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=CC:C6:B5:9F:9A:F8, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=[00001530-1212-efde-1523-785feabcd123], mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=DFU Targ], mRssi=-75, mTimestampNanos=1053780151046283}
[... cut ...] ecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=DFU Targ], mRssi=-70, mTimestampNanos=1053781175652206}
05-21 18:26:07.231 28182-28201/io.aeroscope.aeroscope D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=CC:C6:B5:9F:9A:F8, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=[00001530-1212-efde-1523-785feabcd123], mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=DFU Targ], mRssi=-70, mTimestampNanos=1053781177022283}
```

dariuszseweryn, 2017-05-22:
Hi @RobLewis 
Show your code

dariuszseweryn, 2017-05-24:
@RobLewis Ping? Anything you could add?

dariuszseweryn, 2017-05-29:
Closing due to inactivity. Feel free to open when more information will be provided.

closed 2017-05-29 by dariuszseweryn