dariuszseweryn/RxAndroidBle #198: 1.3.0-SNAPSHOT ScanResult and wifi scan results

sebirdman, 2017-05-24:
When trying out the new snapshot I noticed that 

import com.polidea.rxandroidble.RxBleScanResult;
became 
com.polidea.rxandroidble.scan.ScanResult

In my same class i'm using android.net.wifi.ScanResult and now have to specify exactly which ScanResult i'm using, slightly annoying.

is there any reason for the change from RxBleScanResult to ScanResult?

dariuszseweryn, 2017-05-24:
It is a complete copy of a class `android.bluetooth.le.ScanResult` so if you would use the original Android BLE API you would also end up with specifying a full package. :)

`RxBleScanResult` is a class that was used in the old scan API. They are named different to indicate that they are different. Plus I am trying to drop the Smurf Naming Convention. ;)

sebirdman, 2017-05-25:
Ah i see, ok that makes sense :P

closed 2017-05-25 by sebirdman