dariuszseweryn/RxAndroidBle #200: Suggest to add a method for the "getConnectedDevices"

ZacharyTech, 2017-05-27:
Hello, I suggest you add a function "getConnectedDevices" to the RxBleClient class inside the project, this method is used to connect the device  
[See](https://developer.android.com/reference/android/bluetooth/BluetoothManager.html#getConnectedDevices(int))

dariuszseweryn, 2017-05-29:
Hello @ZacharyTech 
Could you describe what would you need this feature for? You can now track state of individual `RxBleDevice`s via `.observeConnectionStateChanges()`.

ZacharyTech, 2017-05-29:
Use the phone settings -> Bluetooth connection device (BLE device + Android phone), then it will not be able to scan this device.
As long as the standard BLE equipment can be used to connect the above way!

This question is actually a supplement to [How to get an existing connection](https://github.com/Polidea/RxAndroidBle/issues/163)

dariuszseweryn, 2017-05-29:
AFAIK the Settings -> Bluetooth shows only Bluetooth Classic devices which is not directly related to BLE connection and you would still need to establish it.

ZacharyTech, 2017-05-29:
I am sure it is BLE, using the device is  [DIJ OSMO Mobile](http://www.dji.com/osmo-mobile/info) + Samsung S6.

dariuszseweryn, 2017-05-29:
Ok. Neverthless what would you expect from the suggested method? You can already get all connected devices via `BluetoothAdapter.getDefaultAdapter().getConnectedDevices()`

ZacharyTech, 2017-05-29:
Although through the `BluetoothAdapter.getDefaultAdapter (). GetConnectedDevices ()` can be obtained to the connected device，but i don't know  how to associate it with `RxAndridBle` and operate it!

dariuszseweryn, 2017-05-29:
You can use `BluetoothDevice.getAddress()` which returns the Bluetooth Mac Address String and then use `RxBleClient.getBleDevice(bluetoothMacAddressString)`. There is nothing more that can be done.

ZacharyTech, 2017-05-29:
How to read or write, I use `RxBleClient.getBleDevice (macAddress) .establishConnection (autoConnect) `will appear `BleAlreadyConnectedException`

dariuszseweryn, 2017-05-29:
You must have already established the connection to your device via calling `.establishConnection()`.
You have to manage your connection and unsubscribe from `.establishConnection()` when you want the connection to be closed.
For more information check out the samples, documentation and Stackoverflow for tag `rxandroidble`.

ZacharyTech, 2017-05-29:
Okay, thank you very much.

dariuszseweryn, 2017-05-29:
You are welcome. Closing.

closed 2017-05-29 by dariuszseweryn