dariuszseweryn/RxAndroidBle #201: Read ACTION_STATE_CHANGED ?

baadillahnabil, 2017-05-28:
hi, i'm still learning this, but how can i know when users turn off their bluetooth and do something about it or when users turn on their bluetooth. in regular way is checking ACTION_STATE_CHANGED , but i don't know how to do that with this library

Thank you before for this great library, really really helpful ;)

dariuszseweryn, 2017-05-29:
Hello @baadillahnabil 
If you would search this repository for phrase `ACTION_STATE_CHANGED` you would find a public class: `RxBleAdapterStateObservable` which should do exactly what you want.

baadillahnabil, 2017-05-29:
oh ya i found it now, thank you, i wasn't able to find that (my bad), thanks again

closed 2017-05-29 by baadillahnabil

dariuszseweryn, 2017-05-29:
No problem. :) I will give it more visibility by adding a hint on the Readme page.

baadillahnabil, 2017-05-29:
cool, thanks for that too :)