dariuszseweryn/RxAndroidBle #202: Error message：BleCannotSetCharacteristicNotificationException{bluetoothGattCharacteristic=XXXXX····}

edifangyi, 2017-05-29:
### I can write information via Demo, but I can not listen to the information my device has sent
<br />
I get through a lot of BLEDemo to UUID, for example
notify：0003cdd1-0000-1000-8000-00805f9b0131
write：0003cdd2-0000-1000-8000-00805f9b0131

In other Demo can be heard on the equipment to send the data can also be written. But in this project, I can only write, can not listen

I have not changed the project code

<br />
#### Steps to reproduce actual result
1.
2.
3.

Other Demo can be monitored, this project can not
<br />

一加3T
7.1.1

dariuszseweryn, 2017-05-29:
Hello @edifangyi 
Does your characteristic has a `PROPERTY_NOTIFY`?

edifangyi, 2017-05-29:
@dariuszseweryn yes，，look here
[image]

</br>
I clicked Notify  -> CONNECT - >REGISTER NOTIFICATIONS : 
</br>

BleCannotSetCharacteristicNotificationException{bluetoothGattCharacteristic=0003cdd1-0000-1000-8000-00805f9b0131}

dariuszseweryn, 2017-05-29:
Does your notify characteristic have a Client Characteristic Config descriptor?

edifangyi, 2017-05-29:
@dariuszseweryn I'm sorry, I did not read your meaning. However, I use the Bluetooth manufacturers to provide the Demo can use the monitor

http://ww1.sinaimg.cn/large/648b709cly1fg2i5uzpjsj20u01hcjsw.jpg

dariuszseweryn, 2017-05-29:
I think that your characteristic (`0003cdd1-0000-1000-8000-00805f9b0131`) does not have a Client Characteristic Config descriptor (`00002902-0000-1000-8000-00805f9b34fb`) which causes the error.

If that is the case you could use function `RxBleConnection.setupNotification(UUID, NotificationSetupMode.COMPAT)`

edifangyi, 2017-05-29:
@dariuszseweryn  Thank you, the problem is solved. I will ask the Bluetooth chip manufacturers in this area related issues

closed 2017-05-29 by dariuszseweryn

Andy4ward, 2018-11-30:
@dariuszseweryn I've been looking into how i can call the RxBleConnection.setupNotifcation method from my react native project with very little success.

I can go into the android package and set all monitor notifcations to COMPAT but that just causes errors elsewhere.

Should it be possible to get to the the setupNotification function by importing NativeModules from React Native?

dariuszseweryn, 2019-03-27:
@Andy4ward This issue should be now fixed in the RN library