dariuszseweryn/RxAndroidBle #203: Clarification of Location Services and Permission requirements

RobLewis, 2017-05-29:
### Summary
Seeking clear statement of how to handle Location Services for Android 6+
<br />
#### Preconditions
App that works with Android 5 and fails on Android 6 due to Location Services requirement
<br />
#### Steps to reproduce actual result
1. Create app for Android 5
2. Nothing in Manifest about permissions
3. Try to run on Android 6
#### Actual result
Fails due to missing Location Services
<br />
#### Expected result
Fails, but how to address this? 
<br />

Following up on issue #106...
I've added "uses permission" to the Manifest for `BLUETOOTH`, `BLUETOOTH_ADMIN`, and `ACCESS_COARSE_LOCATION`. According to Android docs, the first 2 aren't "dangerous" and so are granted automatically. The third must be approved by the user: at install time for Android < 6, and at runtime for Android ≥ 6. I added the recommended `ContextCompat.checkSelfPermission()` call for Coarse Location to my `onStart()`, with the callback handler. Am I right that this will only be invoked if the Android version requires it? 

But now, what to do about Location Services? Google's [docs on the SettingsApi](https://developers.google.com/android/reference/com/google/android/gms/location/SettingsApi) aren't crystal clear. I'm told to create a GoogleApiClient that supports LocationServices.API, and "add all of the LocationRequests that the app will be using". What do I need here to make RxAndroidBle work? There is also this: "If the client is using BLE scans to derive location, it can request that BLE be enabled by calling `setNeedBle(boolean)`". 

Can someone provide a minimal example of code to handle this sufficient to let RxAndroidBle function? Ideally, one that works on any version of Android?

dariuszseweryn, 2017-05-29:
Hello @RobLewis 

I suggest you to look for the informations you need on the internet.

Runtime Permissions: There are plenty of great articles to be found via Google on how to handle that.

Location Services: https://stackoverflow.com/questions/33045581/location-needs-to-be-enabled-for-bluetooth-low-energy-scanning-on-android-6-0 or other. Feel free to google up your questions.

RobLewis, 2017-05-29:
As I mentioned, I think I have handled the requesting coarse location permission issue. 
The remaining question is _what exactly does RxAndroidBle need enabled to scan for devices?_ The Google docs suggest that there isn't one single "Location Services" function that can be switched on or off, but rather a number of different functions that fall into that category: GPS_PROVIDER, NETWORK_PROVIDER, Google Play Services. Everything my research has found has to do with the Android device determining its own location, which would seem to have little to do with scanning for BLE peripherals. 
Is there no straightforward answer to my question? 
And is there another issue coming up when you switch over to API 21-style scanning?

dariuszseweryn, 2017-05-29:
As you can read in the link I have posted above - some devices using Android 6+ need to have Location Services enabled to return any BLE scans. It does not matter which provider is enabled as long as the Location Services are on. It does not matter on Android 6+ wether you use the old or the new API - you will not get any scans without Location Services being on.

RobLewis, 2017-05-29:
Thanks, I believe that is the answer I needed.

closed 2017-05-29 by dariuszseweryn