dariuszseweryn/RxAndroidBle #204: how add CR or LF when write ?

baadillahnabil, 2017-05-29:
i want to send or write in string or text with LF , how can i do that ?

Thanks before :)

dariuszseweryn, 2017-05-29:
What is CR or LF?

baadillahnabil, 2017-05-29:
CR=Carriage Return , LF = Line Feed (Newline) 
but i want to use only LF

dariuszseweryn, 2017-05-29:
You would need to send the `byte` representation of the character you want.
http://lmgtfy.com/?q=utf-8+carriage+return+code
http://lmgtfy.com/?q=utf-8+newline+code
http://lmgtfy.com/?q=java+string+to+utf+8+bytes

baadillahnabil, 2017-05-29:
okay i got it working now, thanks, but i got another problem, i can only send 20 characters only inside string that converted to bytes using `string.getBytes()` and the pass it to `writeCharacteristic` 

i have this string : 
`String value = ("thislineisstopatthisinthats \n");`

then write it : 
`if (isBluetoothConnected()) {
            connectionObservable
                    .flatMap(rxBleConnection -> rxBleConnection.writeCharacteristic(characteristicUUID, value.getBytes(StandardCharsets.UTF_8)))
                    .observeOn(AndroidSchedulers.mainThread())
                    .subscribe(bytes -> onWriteSuccess(), this::onWriteFailure);
            }`

but the string is only sent for first 20 characters only, that make me so confused

RobLewis, 2017-05-29:
What is the confusion? All normal Bluetooth packets are 20 bytes. Of course you must also be aware that some UTF-8 characters are more than 1 byte. If you want guaranteed 1-byte characters you can use `.getBytes( StandardCharsets.US_ASCII )`. It works for me.

baadillahnabil, 2017-05-30:
@RobLewis thanks for your answer. i confused because i don't know that, can bluetooth packet size be about 30-40 bytes ? i need to send a string that contains about 30-40 characters at one time. i'm very new about bluetooth :(

dariuszseweryn, 2017-05-30:
@baadillahnabil http://lmgtfy.com/?q=ble+send+more+than+20+bytes
Your problem is not related to the library but is a general BLE question and there are many answers available on the internet. Closing.

closed 2017-05-30 by dariuszseweryn

baadillahnabil, 2017-05-30:
yeah, thanks for that, but i got it working using `.createNewLongWriteBuilder()` 

thanks again :)

dariuszseweryn, 2017-05-30:
Long Write is just performing multiple standard writes of 20 bytes each. Alternatively you can request a larger MTU but it is depending on your peripheral. Best Regards.