dariuszseweryn/RxAndroidBle #205: Battery notification not being received

chuks, 2017-05-29:
### Battery notification not received
<br />
#### Connect to device, set notifications for battery level change and custom characteristic
<br />
I am using a custom peripheral, which exposes a custom service & characteristic. I am trying to setup notification on characteristic value changes for both the custom characteristic, and the battery level changes. When using the AOSP API directly, I can set notification for both, and I receive changes on both characteristics as expected. When I use this RxAndroidBle API, I no longer get notifications on the battery level changes. I do get the notification on the custom characteristic, though. I wonder why this is? See code below:

<br />

`List<BluetoothGattCharacteristic> characteristics = new ArrayList<>();
characteristics.add(mSwitchCharacteristic);             // the custom Characteristic
characteristics.add(mBatteryLevelCharacteristic);   // battery level characteristic

Subscription notificationSubscription = connectionObservable
    .flatMap(rxBleConnection -> setupNotifications(rxBleConnection, characteristics))
    .doOnNext(notificationObservable -> {
        Log.i(TAG,"setupAllNotifications - Notification has been set up");
    })
    .subscribe(
        pair -> {
            Log.i(TAG,"setupAllNotifications - received notifications...");
            
            UUID uuid    = pair.first.getUuid();
            byte[] bytes = pair.second;
            BluetoothGattCharacteristic characteristic = null;

            if (mSwitchCharacteristic.getUuid().toString().equals(uuid.toString())){
                Log.i(TAG,"setupAllNotifications - received Switch char...");

                characteristic = mSwitchCharacteristic;
            }
            else if (mBatteryLevelCharacteristic.getUuid().toString().equals(uuid.toString())) {
                Log.i(TAG, "setupAllNotifications - received Battery char...");

                characteristic = mBatteryLevelCharacteristic;
            }

            if (characteristic != null) {
                Log.i(TAG,"setupAllNotifications - got characteristics, process..");

                onNotificationReceived(characteristic, bytes);      // just a method that processes the value
            }
            else{
                Log.i(TAG,"setupAllNotifications - failed to get characteristics, ignore.");
            }
        },
        throwable -> {
            Log.iTAG,"setupAllNotifications - throwable: "+throwable.getLocalizedMessage());

            onNotificationSetupFailure(characteristic, throwable);   // just a method that processes the value
        }
);`

<br />
and the setupNotifications() method is as follows:
`private Observable<Pair<BluetoothGattCharacteristic, byte[]>> setupNotifications(RxBleConnection connection,
                                                                                     List<BluetoothGattCharacteristic> characteristics) {
        Helper.getInstance().debug(TAG,"setupNotifications");

        return Observable.from(characteristics) // deal with every characteristic separately
                .filter(characteristic -> (characteristic.getProperties() & BluetoothGattCharacteristic.PROPERTY_NOTIFY) != 0) // filter characteristics that have notify property
                .flatMap(characteristic -> connection
                                .setupNotification(characteristic) // setup notification for each
                                .flatMap(observable -> observable), // to get the raw bytes from notification
                        Pair::new); // merge characteristic with byte[] to keep track from which characteristic the bytes came
    }`

<br />
#### Actual result
- Battery level changes are not received in the onNotificationReceived() method.
- Changes to the custom characteristic are received in the onNotificationReceived() method
<br />

#### Expected result
- changes to battery level should be received in the onNotificationReceived() method.
- changes to the custom characteristic should be received in the onNotificationReceived() method
<br />

dariuszseweryn, 2017-05-29:
Hello @chuks 
Please set `RxBleLog.setLogLevel(RxBleLog.VERBOSE)` and paste the logs from your application.

chuks, 2017-05-29:
@dariuszseweryn  thanks, here are the relevant logs with Verbose. Using this API, I am receiving the custom characteristic changes from the notifications but not the battery level changes.
<br/>
`05-29 12:45:36.852 20244-20244/com.testapp.rxtestapp D/RxBle#Radio:   QUEUED RxBleRadioOperationScanApi21(65643822)
05-29 12:45:36.852 20244-20302/com.testapp.rxtestapp D/RxBle#Radio:  STARTED RxBleRadioOperationScanApi21(65643822)
05-29 12:45:36.874 20244-20323/com.testapp.rxtestapp I/Adreno: QUALCOMM build                   : 6818200, Idb2b4cb785
                                                                  Build Date                       : 11/18/16
                                                                  OpenGL ES Shader Compiler Version: XE031.09.00.04
                                                                  Local Branch                     : N25
                                                                  Remote Branch                    : 
                                                                  Remote Branch                    : 
                                                                  Reconstruct Branch               : 
05-29 12:45:36.881 20244-20323/com.testapp.rxtestapp I/OpenGLRenderer: Initialized EGL, version 1.4
05-29 12:45:36.881 20244-20323/com.testapp.rxtestapp D/OpenGLRenderer: Swap behavior 1
05-29 12:45:36.890 20244-20244/com.testapp.rxtestapp D/BluetoothAdapter: isLeEnabled(): ON
05-29 12:45:36.894 20244-20257/com.testapp.rxtestapp D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5 mClientIf=0
05-29 12:45:36.899 20244-20302/com.testapp.rxtestapp D/RxBle#Radio: FINISHED RxBleRadioOperationScanApi21(65643822)
05-29 12:45:37.128 20244-20249/com.testapp.rxtestapp I/art: Do partial code cache collection, code=20KB, data=56KB
05-29 12:45:37.128 20244-20249/com.testapp.rxtestapp I/art: After code cache collection, code=20KB, data=56KB
05-29 12:45:37.128 20244-20249/com.testapp.rxtestapp I/art: Increasing code cache capacity to 256KB
05-29 12:45:37.586 20244-20244/com.testapp.rxtestapp D/EventBus: No subscribers registered for event class com.testapp.rxtestapp.broadcasts.NetworkBroadcast
05-29 12:45:37.586 20244-20244/com.testapp.rxtestapp D/EventBus: No subscribers registered for event class org.greenrobot.eventbus.NoSubscriberEvent
05-29 12:45:37.662 20244-20249/com.testapp.rxtestapp I/art: Compiler allocated 4MB to compile java.lang.String com.testapp.rxtestapp.utils.APIHelper.makeRequest(java.lang.String, java.lang.String, java.lang.String)
05-29 12:45:41.788 20244-20278/com.testapp.rxtestapp V/FA: Inactivity, disconnecting from the service
05-29 12:45:43.297 20244-20244/com.testapp.rxtestapp I/com.testapp.rxtestapp.app.Rxtestapp: getRxBleClient
05-29 12:45:43.323 20244-20302/com.testapp.rxtestapp D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(106250063)
05-29 12:45:43.324 20244-20244/com.testapp.rxtestapp D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(106250063)
05-29 12:45:43.336 20244-20244/com.testapp.rxtestapp V/RxBle#BleConnectionCompat: Connecting without reflection
05-29 12:45:43.337 20244-20244/com.testapp.rxtestapp D/BluetoothGatt: connect() - device: C5:82:E1:4D:93:32, auto: false
05-29 12:45:43.337 20244-20244/com.testapp.rxtestapp D/BluetoothGatt: registerApp()
05-29 12:45:43.338 20244-20244/com.testapp.rxtestapp D/BluetoothGatt: registerApp() - UUID=25978641-894b-49f8-b612-dbf167edf0ef
05-29 12:45:43.340 20244-20244/com.testapp.rxtestapp D/EventBus: No subscribers registered for event class com.testapp.rxtestapp.bus_events.RxtestappConnectionEvent
05-29 12:45:43.340 20244-20257/com.testapp.rxtestapp D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
05-29 12:45:43.340 20244-20244/com.testapp.rxtestapp D/EventBus: No subscribers registered for event class org.greenrobot.eventbus.NoSubscriberEvent
05-29 12:45:43.739 20244-20256/com.testapp.rxtestapp D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=C5:82:E1:4D:93:32
05-29 12:45:43.742 20244-20256/com.testapp.rxtestapp D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
05-29 12:45:43.769 20244-20302/com.testapp.rxtestapp D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(106250063)
05-29 12:45:43.769 20244-20244/com.testapp.rxtestapp D/RxBle#Radio:   QUEUED RxBleRadioOperationServicesDiscover(64721463)
05-29 12:45:43.770 20244-20302/com.testapp.rxtestapp D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(64721463)
05-29 12:45:43.774 20244-20244/com.testapp.rxtestapp D/BluetoothGatt: discoverServices() - device: C5:82:E1:4D:93:32
05-29 12:45:44.747 20244-20257/com.testapp.rxtestapp D/BluetoothGatt: onSearchComplete() = Device=C5:82:E1:4D:93:32 Status=0
05-29 12:45:44.749 20244-20257/com.testapp.rxtestapp D/RxBle#BluetoothGatt: onServicesDiscovered status=0
05-29 12:45:44.753 20244-20302/com.testapp.rxtestapp D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(64721463)
05-29 12:45:44.754 20244-20244/com.testapp.rxtestapp I/com.testapp.rxtestapp.app.Rxtestapp: getRxBleClient
05-29 12:45:44.760 20244-20302/com.testapp.rxtestapp D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(185798987)
05-29 12:45:44.761 20244-20244/com.testapp.rxtestapp D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(185798987)
05-29 12:45:44.761 20244-20244/com.testapp.rxtestapp I/com.testapp.rxtestapp.app.Rxtestapp: getRxBleClient
[... cut ...]
05-29 12:46:21.949 20244-20302/com.testapp.rxtestapp D/RxBle#Radio:  STARTED RxBleRadioOperationScanApi21(51051310)
05-29 12:46:21.950 20244-20244/com.testapp.rxtestapp D/RxBle#Radio:   QUEUED RxBleRadioOperationScanApi21(51051310)
05-29 12:46:21.957 20244-20244/com.testapp.rxtestapp D/BluetoothAdapter: isLeEnabled(): ON
05-29 12:46:21.966 20244-20257/com.testapp.rxtestapp D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5 mClientIf=0
05-29 12:46:21.975 20244-20302/com.testapp.rxtestapp D/RxBle#Radio: FINISHED RxBleRadioOperationScanApi21(51051310)`

chuks, 2017-05-29:
@dariuszseweryn  See attached log file, couldn't find a way to format the logs into a readable format. I'm doing a scan, connecting to the device, and writing notification for battery level changes and the custom characteristic changes

[testlogs.txt](https://github.com/Polidea/RxAndroidBle/files/1036555/testlogs.txt)

dariuszseweryn, 2017-05-29:
Everything is fine judging from the logs you have pasted in the comment above. Are you sure that battery notifications are being sent at all? It seems that the Android BLE Stack is not seeing them (no `D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=00002a19-0000-1000-8000-00805f9b34fb` anywhere to be seen.

chuks, 2017-05-29:
@dariuszseweryn Yes, that's the problem I'm seeing. It looks like both notifications are set. I can see this line twice for after I send the notifications, indicating that both notifications have been sent:
FINISHED RxBleRadioOperationDescriptorWrite

But I only receive the changes for the custom characteristics (I can see this line whenever I trigger a change onCharacteristicChanged characteristic=0000fad4-0eea-45d1-a44a-bb3ce36fced6), and not from the battery changes.

chuks, 2017-05-29:
How else can I verify? This is the code I'm using to combine both characteristics:
`private Observable<Pair<BluetoothGattCharacteristic, byte[]>> setupNotifications(RxBleConnection connection,
                                                                                     List<BluetoothGattCharacteristic> characteristics) {
        Helper.getInstance().debug(TAG,"setupNotifications");

        return Observable.from(characteristics) // deal with every characteristic separately
                .filter(characteristic -> (characteristic.getProperties() & BluetoothGattCharacteristic.PROPERTY_NOTIFY) != 0) // filter characteristics that have notify property
                .flatMap(characteristic -> connection
                                .setupNotification(characteristic) // setup notification for each
                                .flatMap(observable -> observable), // to get the raw bytes from notification
                        Pair::new); // merge characteristic with byte[] to keep track from which characteristic the bytes came
    }`

<br>
where the List is just <br>
`List<BluetoothGattCharacteristic> characteristics = new ArrayList<>();
            characteristics.add(mSwitchCharacteristic);             => this is the custom characteristic
            characteristics.add(mBatteryLevelCharacteristic);   => this is the battery characteristic
`

dariuszseweryn, 2017-05-29:
In the logs you have pasted I do not see any logs from your part of the application. Are you sure that you have run the app from the code you have posted in the original post?

chuks, 2017-05-29:
Yes, it's the same code. I had to clean up the logs and remove some proprietary info before posting the logs. Also, here's something I find interesting. If I write only the custom characteristics (i.e. without the battery), I get the notifications as expected. If I do the reverse, i.e. send notifications for only battery changes without the custom characteristics, I do not get anything.

chuks, 2017-05-29:
Using the default ASOP API for the same process, I can set both notifications and receive changes on both, so I'm wondering if the code above is wrong?

dariuszseweryn, 2017-05-29:
I cannot really tell. The logs from the library do look good. It seems that the Android BLE stack is not receiving the notifications from the battery. I wonder what would happen if you would switch the place of the characteristics when adding to the `List`.

chuks, 2017-05-29:
" I wonder what would happen if you would switch the place of the characteristics when adding to the List"

That's what I did. I added battery first, then the other one, but got the same behavior.  Yes, the logs from the library do look good. What if I don't combine setting both notification, i.e. if I set them separately with two separate calls?

chuks, 2017-05-29:
@dariuszseweryn ok --it is working now. It takes a very long time before the battery notifications start coming in. No code changes needed. I think the delay must be from the peripheral side, since the custom starts immediately. Can close this, thanks!

dariuszseweryn, 2017-05-30:
FYI You can close it by yourself next time. :)

closed 2017-05-30 by dariuszseweryn