dariuszseweryn/RxAndroidBle #209: Release 1.3.0?

kenwdelong, 2017-06-02:
Just a quick question: any thoughts on the timeline for release 1.3.0?

dariuszseweryn, 2017-06-05:
I would like to make it somewhere this week.

closed 2017-06-05 by dariuszseweryn

dariuszseweryn, 2017-06-06:
@kenwdelong Feel free to test `1.3.0-SNAPSHOT` in the mean time. I do not expect any changes to the API but it would be easier to fix bugs before the release.

kenwdelong, 2017-06-06:
That was my plan for today!!  :-)

kenwdelong, 2017-06-06:
I've integrated the SNAPSHOT build, and so far so good.  We'll do some more extensive testing tomorrow.

BTW, why did you comment out setReportDelay() etc in ScanSettingsBuilder?

dariuszseweryn, 2017-06-07:
Because that is not yet implemented in a backwards-compatible way. This is a place for pull requests from the community if anyone will need this functionality — from my experience implemented features are the most used ones.

kenwdelong, 2017-06-09:
After a day of testing, so far everything is working fine.

dariuszseweryn, 2017-06-09:
Good to hear that. I plan to make an official release today.

dariuszseweryn, 2017-06-09:
Released.

kenwdelong, 2017-06-09:
Excellent, thanks!