dariuszseweryn/RxAndroidBle #210: Can't write to characteristic

lorenzowoodridge, 2017-06-02:
I'm currently trying to write to a characteristic via an instance of RxBleConnection like I have done many times before, but have been unsuccessful on multiple devices.

Snippet of code:
`rxBleConnection.writeCharacteristic(CHARACTERISTIC_CONTROL, new byte[COMMAND_START])
                .subscribe(bytes -> {
                   Log.d(TAG, "here");
                }, throwable -> {
                    Log.d(TAG, "here1");
                });
`

Error logs:
06-02 17:13:07.376 10639-14392/mobile.airturn.com.airturn D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(61635033)
06-02 17:13:07.377 10639-14068/mobile.airturn.com.airturn D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(61635033)
06-02 17:13:37.378 10639-10987/mobile.airturn.com.airturn D/AirTurnPrivateAPI: here1
06-02 17:13:37.379 10639-14068/mobile.airturn.com.airturn D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(61635033)
06-02 17:13:37.385 10639-10655/mobile.airturn.com.airturn D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=47c17590-a9c1-11e3-a5e2-0800200c9a66 status=133
06-02 17:13:37.386 10639-10655/mobile.airturn.com.airturn D/BluetoothGatt: onClientConnectionState() - status=22 clientIf=5 device=00:07:80:F3:5D:B0
06-02 17:13:37.386 10639-10655/mobile.airturn.com.airturn D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=22
06-02 17:13:37.387 10639-10950/mobile.airturn.com.airturn D/AirTurnAPI: observeStateChanges: BleGattCharacteristicException{macAddress=00:07:80:F3:5D:B0, status=133 (0x85 -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h), bleGattOperationType=BleGattOperation{description='CHARACTERISTIC_WRITE'}}
06-02 17:13:37.387 10639-10950/mobile.airturn.com.airturn D/AirTurnAPI: RxBleConnectionState{DISCONNECTED}
06-02 17:13:37.387 10639-14175/mobile.airturn.com.airturn D/BluetoothGatt: setCharacteristicNotification() - uuid: 362f71a0-6c96-11e3-981f-0800200c9a66 enable: false
06-02 17:13:37.390 10639-10639/mobile.airturn.com.airturn D/MainSettingsFragment: Disconnect triggered!
06-02 17:13:37.395 10639-14175/mobile.airturn.com.airturn D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(46384973)
06-02 17:13:37.396 10639-14068/mobile.airturn.com.airturn D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(46384973)
06-02 17:13:37.397 10639-10950/mobile.airturn.com.airturn D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(111378946)

I believe the issue may be related to this"
https://stackoverflow.com/questions/25888817/android-bluetooth-status-133-in-oncharacteristicwrite

dariuszseweryn, 2017-06-05:
Why do you believe so? Have you tried to manually change the write type of the characteristic? Did it helped?

lorenzowoodridge, 2017-06-05:
I've tried to change the write type of the characteristic and that did not work, but I'm thinking I need to set the value of a descriptor, which to my knowledge the library does not give me access to the instance of the BluetoothGatt in order to do so.

I'm also in contact with the firmware engineer to see if there is something else that I'm missing.

dariuszseweryn, 2017-06-12:
@lorenzowoodridge Any neews?

lorenzowoodridge, 2017-06-12:
Not sure why it's working now but it is!

dariuszseweryn, 2017-06-12:
Have you changed anything?

lorenzowoodridge, 2017-06-13:
It was definitely something minor that I was doing wrong but I'm not sure what. Not an issue with the library!

closed 2017-06-13 by lorenzowoodridge