dariuszseweryn/RxAndroidBle #211: Keep connection between activities?

baadillahnabil, 2017-06-03:
Hi, 
straight to example: 

i have an `Activity A` -> this activity is the about selecting a device from devices (in recyclerview) and establish a connection when selected, i have done this part successfully. and then after the device is selected it goes to `Activity B` where all the main menu is there, but i want to keep the connection between that activities. so my questions is how can i make that happen ?

please help me, i've been struggling to find how to do this.

Thanks before :)

logalleg, 2017-06-05:
You probably want the Bluetooth actions occurring in a Service, and that Service using broadcasts for various activities, fragments, etc to listen in on.

dariuszseweryn, 2017-06-05:
Hello @baadillahnabil 
This is a general android / rxjava architecture question — not an issue or a feature request with the library. 
I have seen that you have already posted the same question on stackoverflow (https://stackoverflow.com/questions/44353812/android-keep-bluetooth-connection-between-activities) which is a better place to ask. 
Closing.

closed 2017-06-05 by dariuszseweryn

9otFound, 2022-09-17:
Hey,
May you please help me with how you fixed the issue?
I mean how you kept connections between activities
I couldn't find any solution for that :(

dariuszseweryn, 2022-09-18:
@9otFound There are many different approaches to that (eg. Android Service, a singleton object) and every one of them boils down to something that's lifecycle is longer than a single Activity.