dariuszseweryn/RxAndroidBle #213: Introduce a more concise connection interface

dariuszseweryn, 2017-06-06:
### Summary
Current connection interface `RxBleConnection` consists almost entirely of functions that return various `Observable`s where other reactive types would be more appropriate i.e.
* `Observable<byte[]> readCharacteristic(UUID;` vs `Single<byte[]> readCharacteristic(UUID);`
* `Observable<byte[]> writeCharacteristic(UUID, byte[]);` vs `Completable writeCharacteristic(UUID, byte[]);`

RobLewis, 2017-06-06:
I agree this would be a nice enhancement. Maybe as part of the change to RxJava 2?

dariuszseweryn, 2017-06-06:
I wanted to play around with other stuff (like encryption and binding) which I do not have a lot to do before going to `RxJava 2` because of the opportunity to drop unwanted APIs then. Do you think that `RxJava 2` would be better to chase now?

This is quite an easy change once there will be a method returning the new interface.

streetsofboston, 2017-10-12:
I agree with @RobLewis 

I would focus more on moving to using RxJava 2 while changing the code to use the correct observable-types at the same time.

The fact that there is no RxJava 2 version (yet) of RxAndroidBLE may become more and more a roadblock for folks that'll prevent them from using this great library.