dariuszseweryn/RxAndroidBle #215: Duplicate declarations of permission android.permission.ACCESS_COARSE_LOCATION with different maxSdkVersions

LordParsley, 2017-06-09:
### Summary
An RxAndroidBle manifest permission for ACCESS_COARSE_LOCATION is causing a duplicate permission problem preventing submission on Google Play if an app already includes the permission as a "uses-permission" instead of "uses-permission-sdk-23".

[image]

#### Steps to reproduce actual result

If your app targets permissions lower than SDK 23, then

<br />1. Include in your manifest:
```
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
```

<br />2. Build with RxAndroidBle as a gradle dependency.
<br />3. Your output manifest will have duplicate permissions:
```
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION" />
```

<br />4. Google Play will reject an upload due to "Duplicate declarations of permission android.permission.ACCESS_COARSE_LOCATION with different maxSdkVersions."

#### Confirmation of Source

Manifest merger release report:

```
uses-permission-sdk-23
ADDED from [com.polidea.rxandroidble:rxandroidble:1.1.0] /Users/Mew/.android/
```

#### Workaround

The app consuming the dependency may adjust its own manifest to use a different maxSdkVersion:

```
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="22"/>
```

dariuszseweryn, 2017-06-09:
Hello,

Have you read the `README`? Recently we have added there a paragraph:

### Permissions
RxAndroidBle already provides all the necessary bluetooth permissions for you. Recently, Google has started checking these when releasing to the Play Store. If you have ACCESS_COARSE_LOCATION set manually you may run into an issue where your permission do not merge with RxAndroidBle's, resulting in a failure to upload to the Play Store This permission is only required on SDK 23+. If you need this permission on a lower version of Android use:

```
<uses-permission
  android:name="android.permission.ACCESS_COARSE_LOCATION"
  android:maxSdkVersion="22"/>
```

LordParsley, 2017-06-09:
@dariuszseweryn thank you – sorry, I see the warning added in https://github.com/Polidea/RxAndroidBle/commit/56854d9b2ca88fbb97053d8f1909c4fe31665e12

dariuszseweryn, 2017-06-09:
Yup. Does it fix your problem?

RobLewis, 2017-06-11:
Can I ask a question in concrete terms? 

- I'm targeting API 21 and testing on an Android 5.0.1 tablet. I have not declared any permissions in the manifest. The app works OK. 
- On an Android 6 tablet, the app aborts with the permissions issue. 
- What is the best way to solve this and maintain compatibility with different Android versions?

dariuszseweryn, 2017-06-11:
@RobLewis The issue in the topic is related to uploading an `.apk` file to the Play Store which is checking the `AndroidManifest.xml`. Local builds are not affected.

As for your question `What is the best way to solve this and maintain compatibility with different Android versions?` you need to read about `Runtime Permissions`. There is a lot of good articles accessible via Google search for that so there is no sense in me trying to explain it here.

LordParsley, 2017-06-12:
@dariuszseweryn Yes, thanks, I used the workaround to upload to Google Play successfully (didn't realise you had documented it already when I created the issue).

dariuszseweryn, 2017-06-12:
Good. Closing then.

closed 2017-06-12 by dariuszseweryn

vanlooverenkoen, 2023-05-09:
The documentation & comments in this issue did not work for me: 

This is what actually worked for me!
```
    <uses-permission
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        tools:node="replace" />
    <uses-permission
        android:name="android.permission.ACCESS_FINE_LOCATION"
        tools:node="replace" />

    <uses-permission-sdk-23
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        tools:node="remove" />
    <uses-permission-sdk-23
        android:name="android.permission.ACCESS_FINE_LOCATION"
        tools:node="remove" />
```