dariuszseweryn/RxAndroidBle #216: Is v. 1.3.0 scanning backward-compatible with earlier versions?

RobLewis, 2017-06-11:
I note that the new ReadMe has introduced `ScanSettings.Builder()` in the scanning API. Is this required, and will it break code using earlier versions? 
Also, are constants like `ScanSettings.CALLBACK_TYPE_ALL_MATCHES` documented somewhere? And what form do the optional filters take? 

P.S. Like the new logo!

dariuszseweryn, 2017-06-11:
Hello @RobLewis 
> I note that the new ReadMe has introduced ScanSettings.Builder() in the scanning API. Is this required, and will it break code using earlier versions?

No (at least yet). The library is using a `Semantic Versioning` and deprecated APIs will continue to work until version `2.0.0`.

> Also, are constants like ScanSettings.CALLBACK_TYPE_ALL_MATCHES documented somewhere? And what form do the optional filters take?

`ScanSettings` and `ScanFilter` classes are copied directly from Android AOSP project and you can find the documentation for those classes using google.

RobLewis, 2017-06-12:
I see that `ScanSettings.Builder` is invoked with an anonymous inner class as the first (and perhaps only) argument to `.scanBleDevices()`. What is the syntax for adding `ScanFilter`s? Is it a second argument to `.scanBleDevices()`?

dariuszseweryn, 2017-06-12:
```
ScanFilter filter = new ScanFilter.Builder()
//  set wanted filters
  .build()
```
It is copy-pasted from the original Android API. You can add zero or more `ScanFilter`s to a scan. Unlike the old API the new one will emit `ScanResults` even if only one `ScanFilter` will match (in case you add many) which is inline with the original Android API behaviour. Does it answer your question?

RobLewis, 2017-06-12:
Thanks, but no, it doesn't. My question is: where in the code does the filter definition go? I see how to _create_ the filter object, but I don't see where to put it in the code. The `ScanSettings` object is shown as an argument to the `.scanBleDevices()` function. Is the `ScanFilter` object a second argument to this function (an overload)?

dariuszseweryn, 2017-06-12:
As you can see in the interface of `RxBleClient` the `.scanBleDevices(ScanSettings, ScanFilter...)` function is accepting `ScanFilter`s as `varargs`. You can add none or an arbitrary number of `ScanFilter`s to the function after the `ScanSettings`

In the `sample` project you can see an example with `ScanSettings` and one `ScanFilter`:
```
            scanSubscription = rxBleClient.scanBleDevices(
                    new ScanSettings.Builder()
                            .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                            .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                            .build(),
                    new ScanFilter.Builder()
                            // add custom filters if needed
                            .build()
            )
```
but you can add more:
```
            scanSubscription = rxBleClient.scanBleDevices(
                    new ScanSettings.Builder()
                            .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                            .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                            .build(),
                    new ScanFilter.Builder()
                            // add custom filters if needed
                            .build(),
                    new ScanFilter.Builder()
                            // add custom filters if needed
                            .build(),
                    new ScanFilter.Builder()
                            // add custom filters if needed
                            .build(),
                    new ScanFilter.Builder()
                            // add custom filters if needed
                            .build()
            )
```
Does it answer your question?

RobLewis, 2017-06-12:
Perfectly clear, thank you. I guess I should learn to examine your source code more closely.

dariuszseweryn, 2017-06-12:
There are Javadocs, samples and the API is written in plain old java - nothing fancy. If you are using Android Studio - you can use keyboard shortcuts to peek Javadoc (cmd + j by default if I remember correctly) and what parameters a function accepts (cmd + p).
You're welcome. Closing.

closed 2017-06-12 by dariuszseweryn