dariuszseweryn/RxAndroidBle #217: Characteristics Showing up as Descriptors

lorenzowoodridge, 2017-06-12:
I recently did an over the air update to my Bluetooth device and now some of the characteristics are not showing up as characteristics, instead they are showing up as descriptors. I thought the issue was with the Bluetooth cache, so I wiped the cache but that didn't work. I then used other devices and that did not work either.

I just tested the device with our iOS version of the app and it is able to read and write to those characteristics as it should (instead of mistaking them for descriptors), so I am thinking it may be an issue with the service discovery or some other bug. I've written some logs as follows that get called on a successful discoverServices():

`// debugging
for (BluetoothGattService s : services.getBluetoothGattServices()) {
    Log.d(TAG, "Service UUID: " + s.getUuid());

    for (BluetoothGattCharacteristic c : s.getCharacteristics()) {
        Log.d(TAG, "----- characteristic UUID: " + c.getUuid());

        for (BluetoothGattDescriptor d : c.getDescriptors()) {
            Log.d(TAG, "-------- descriptors UUID: " + d.getUuid());
        }
    }
}`

And a snippet of output is as follows:

`----- characteristic UUID: fba0a8af-3d4b-4aab-8e4c-4112b0561425
------- descriptors UUID: 00002901-0000-1000-8000-00805f9b34fb
------- descriptors UUID: 00002803-0000-1000-8000-00805f9b34fb
------- descriptors UUID: 640c0d80-9b4f-11e3-a5e2-0800200c9a66
------- descriptors UUID: 00002901-0000-1000-8000-00805f9b34fb
------- descriptors UUID: 00002803-0000-1000-8000-00805f9b34fb
------- descriptors UUID: 49aa8950-a40d-11e3-a5e2-0800200c9a66
------- descriptors UUID: 00002901-0000-1000-8000-00805f9b34fb
------- descriptors UUID: 00002803-0000-1000-8000-00805f9b34fb
------- descriptors UUID: 7b51af7d-f28a-48be-8081-6a647e6249eb
------- descriptors UUID: 00002901-0000-1000-8000-00805f9b34fb
------- descriptors UUID: 00002803-0000-1000-8000-00805f9b34fb
------- descriptors UUID: db3207d4-97ff-4497-ad1d-0ba1bccc56ba
-------- descriptors UUID: 00002901-0000-1000-8000-00805f9b34fb`

Many of those descriptors are actually supposed to be characteristics.

lorenzowoodridge, 2017-06-12:
Our firmware developer just reported this back actually:

"Looks like Android turns characteristics into descriptors after the first 30 characteristics!

I moved the last characteristics up and sure enough, it just cuts off any after 30. Great."

dariuszseweryn, 2017-06-13:
Thanks for the report. Unfortunately as you have mentioned it is a bug in Android OS and there is nothing we could do on the library side. Maybe it would be possible to workaround this issue by using reflection and reworking the discovered items. May be worth investigating if someone has time. As for now — closing.

closed 2017-06-13 by dariuszseweryn

lorenzowoodridge, 2017-06-13:
I agree that this is not a bug with the library, but it is definitely worth bringing to everyone's attention :)

dariuszseweryn, 2017-06-13:
Yup, no doubts about it. Thanks again for the info! :)