dariuszseweryn/RxAndroidBle #218: when unsubscribe during long-write, long-write doesn't "finish" and disconnect never "started"

DogsForArms, 2017-06-13:
Hello!  I have been noticing that if "unsubscribe" is called when a write is ongoing, or a subscribe to change is subscribing, the queued "Disconnect" is not finished.  In my case I am noticing it during a long-write.  My application is observing a characteristic-change for response, and long-writing.  Here is an example of a "unsubscribe" while long-write

D/rxble-writeupdate: 1) observing 297ff612-9a62-4068-9a0c-0a21c4146187
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicLongWrite(99139398)
D/RxBle#Radio: FINISHED RxBleRadioOperationDescriptorWrite(85537306)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicLongWrite(99139398)
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/rxble-: - calling unsubscribe here -
D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(77589682)
D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(60563966)
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=297ff612-9a62-4068-9a0c-0a21c4146187 status=0

So i start off with "RxBleRadioOperationDescriptorWrite(85537306)" is ongoing, but now it is finished.  The long write starts, the disconnect is queued, the long write is disabled but the disconnect never starts.  I can see my BLE device says I'm still connected, but subsequent attempts to scan for devices never actually "start".  It requires an app reboot then the state is back to normal.  

I was seeing the same behavior when I was waiting for `connection.setupNotification(characteristic)` to let me know when I am monitoring changes to a characteristic, although it is a shorter window.

So as far as steps to reproduce go, I'd say try unsubscrib mid a very "long" write.  I do not have an ack strategy (do I need one?).  Thanks for your time.

DogsForArms, 2017-06-13:
Thanks for the quickness Darius!  Let me know if I can provide details or help provide more specific code to reproduce.

dariuszseweryn, 2017-06-13:
I will create a fix tomorrow and I will ask you to test it. ;)

dariuszseweryn, 2017-06-14:
This is now fixed and should be available shortly in `1.3.1-SNAPSHOT`. Please try it out.

dariuszseweryn, 2017-06-14:
This is actually a more general problem with unsubscribing from operations while they were running. Should be fixed now but it may use a refactor later on. Please test it and share if it helped.

Best Regards

closed 2017-06-16 by dariuszseweryn

DogsForArms, 2017-07-19:
Thanks very much!  I'll update now and let you know what I see.  Great work!