dariuszseweryn/RxAndroidBle #220: Not compatible with Bluez v5

Pascal66, 2017-06-14:
After upgraded my phone because it's necessary for one device,
The app using your library cant connect to it
What can i do ?

### Disclaimer
Before you will post a new issue to the RxAndroidBle library consider if you are filling a bug/feature request or a general question 
about:
* Usage of RxJava
* Usage of Bluetooth Low Energy
* Usage of RxAndroidBle
* Weird behaviour of the peripheral you are using
* Other not directly related to a new feature request or a bug

If any of the above points seems to match your use case — consider using Google or creating a question on www.stackoverflow.com with 
a tag `rxandroidble` where it will be easier to access by other people with similar questions. Issues that are not bugs or feature requests
will be closed.

### Summary

#### Preconditions

#### Steps to reproduce actual result
<br />1.
<br />2.
<br />3.
#### Minimum code snippet reproducing the issue

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`

#### Actual result

#### Expected result

dariuszseweryn, 2017-06-14:
Hello @Pascal66
Please read the disclaimer on your post and at least provide more information because in the current it is impossible to tell anything.

Pascal66, 2017-06-15:
Ok I've a custom rom 4.4.4 with bluez stack for bluetooth.
I need to use an app wich I know using your library, it unable to get the ble device.
When I look at my bluetooth devices, I can saw all
not sure how to explain better

Pascal66, 2017-06-15:
Logcat of my bluez stack on android 4.4.4 : 
The ble device is well reconized as you see below
The problem seems to be is the use of your library wich cant talk with this ble device.
Unfortunatly, I cant look at this proprietary app and cant watch wheress the problem.

```
06-14 17:54:21.085 4302-4356/? D/BlueZ: external/bluetooth/bluez/android/hal-bluetooth.c:handle_device_found() 
06-14 17:54:21.085 4302-4356/? D/BlueZ: external/bluetooth/bluez/android/hal-bluetooth.c:device_props_to_hal() prop[0]: type=BT_PROPERTY_BDADDR len=6 val=00:a0:50:c9:53:a8
06-14 17:54:21.095 4302-4356/? D/BlueZ: external/bluetooth/bluez/android/hal-bluetooth.c:device_props_to_hal() prop[1]: type=BT_PROPERTY_TYPE_OF_DEVICE len=4 val=BT_DEVICE_DEVTYPE_BLE
06-14 17:54:21.095 4302-4356/? D/BlueZ: external/bluetooth/bluez/android/hal-bluetooth.c:device_props_to_hal() prop[2]: type=BT_PROPERTY_REMOTE_RSSI len=4 val=211
06-14 17:54:21.095 4302-4356/? D/BlueZ: external/bluetooth/bluez/android/hal-bluetooth.c:device_props_to_hal() prop[3]: type=BT_PROPERTY_BDNAME len=14 val=Flipr 0021B863
06-14 17:54:21.125 4302-4356/? D/BlueZ: external/bluetooth/bluez/android/hal-bluetooth.c:handle_device_state_changed() 
06-14 17:54:21.125 4302-4356/? D/BlueZ: external/bluetooth/bluez/android/hal-bluetooth.c:device_props_to_hal() prop[0]: type=BT_PROPERTY_REMOTE_RSSI len=4 val=177
06-14 17:54:21.135 28736-28736/? D/BluetoothEventManager: DeviceFoundHandler created new CachedBluetoothDevice: 00:A0:50:C9:53:A8
```

dariuszseweryn, 2017-06-16:
Unfortunately I cannot tell anything without more information from inside the application with `RxBleLog.setLogLevel(RxBleLog.VERBOSE)` being set.
The library supports the official Android BLE API. I do not know what exactly your custom ROM changes and how does it work with the Android API.
> What can i do ?

You can fork the library source and try to make it work with your setup. Alternatively you could try to use the vanilla Android API but I suspect that it will not work as well since the library relies on that.

As for now I close this issue since we only support official API. If you will get more relevant information that could shed some light on what is actually happening — feel free to comment — I will definitely read what you will add. I cannot promise more.

Best Regards

closed 2017-06-16 by dariuszseweryn

Pascal66, 2017-06-16:
New logcat, hope it can give you some clues, because i'm near to an headache about this strange behaviour:
```
06-16 17:27:21.709 32381-32381/com.goflipr.flipr I/dalvikvm: Could not find method android.view.View.<init>, referenced from method com.wang.avi.AVLoadingIndicatorView.<init>
06-16 17:27:21.709 32381-32381/com.goflipr.flipr W/dalvikvm: VFY: unable to resolve direct method 22391: Landroid/view/View;.<init> (Landroid/content/Context;Landroid/util/AttributeSet;II)V
06-16 17:27:21.709 32381-32381/com.goflipr.flipr D/dalvikvm: VFY: replacing opcode 0x70 at 0x0001
06-16 17:27:21.709 32381-32381/com.goflipr.flipr I/dalvikvm: Could not find method android.view.View.drawableHotspotChanged, referenced from method com.wang.avi.AVLoadingIndicatorView.drawableHotspotChanged
06-16 17:27:21.709 32381-32381/com.goflipr.flipr W/dalvikvm: VFY: unable to resolve virtual method 22411: Landroid/view/View;.drawableHotspotChanged (FF)V
06-16 17:27:21.709 32381-32381/com.goflipr.flipr D/dalvikvm: VFY: replacing opcode 0x6f at 0x0000
06-16 17:27:21.829 32381-32381/com.goflipr.flipr I/dalvikvm: Could not find method android.app.Activity.shouldShowRequestPermissionRationale, referenced from method com.tbruyelle.rxpermissions2.RxPermissions.shouldShowRequestPermissionRationaleImplementation
06-16 17:27:21.829 32381-32381/com.goflipr.flipr W/dalvikvm: VFY: unable to resolve virtual method 141: Landroid/app/Activity;.shouldShowRequestPermissionRationale (Ljava/lang/String;)Z
06-16 17:27:21.829 32381-32381/com.goflipr.flipr D/dalvikvm: VFY: replacing opcode 0x6e at 0x000d
06-16 17:27:21.829 32381-32381/com.goflipr.flipr I/dalvikvm: Could not find method android.app.Activity.checkSelfPermission, referenced from method com.tbruyelle.rxpermissions2.RxPermissionsFragment.isGranted
06-16 17:27:21.829 32381-32381/com.goflipr.flipr W/dalvikvm: VFY: unable to resolve virtual method 96: Landroid/app/Activity;.checkSelfPermission (Ljava/lang/String;)I
06-16 17:27:21.829 32381-32381/com.goflipr.flipr D/dalvikvm: VFY: replacing opcode 0x6e at 0x0004
06-16 17:27:21.829 32381-32381/com.goflipr.flipr I/dalvikvm: Could not find method android.content.pm.PackageManager.isPermissionRevokedByPolicy, referenced from method com.tbruyelle.rxpermissions2.RxPermissionsFragment.isRevoked
06-16 17:27:21.829 32381-32381/com.goflipr.flipr W/dalvikvm: VFY: unable to resolve virtual method 704: Landroid/content/pm/PackageManager;.isPermissionRevokedByPolicy (Ljava/lang/String;Ljava/lang/String;)Z
06-16 17:27:21.829 32381-32381/com.goflipr.flipr D/dalvikvm: VFY: replacing opcode 0x6e at 0x0010
06-16 17:27:21.829 32381-32381/com.goflipr.flipr I/dalvikvm: Could not find method android.app.Fragment.onRequestPermissionsResult, referenced from method com.tbruyelle.rxpermissions2.RxPermissionsFragment.onRequestPermissionsResult
06-16 17:27:21.829 32381-32381/com.goflipr.flipr W/dalvikvm: VFY: unable to resolve virtual method 254: Landroid/app/Fragment;.onRequestPermissionsResult (I[Ljava/lang/String;[I)V
06-16 17:27:21.829 32381-32381/com.goflipr.flipr D/dalvikvm: VFY: replacing opcode 0x6f at 0x0000
06-16 17:27:21.829 32381-32381/com.goflipr.flipr I/dalvikvm: Could not find method com.tbruyelle.rxpermissions2.RxPermissionsFragment.requestPermissions, referenced from method com.tbruyelle.rxpermissions2.RxPermissionsFragment.requestPermissions
06-16 17:27:21.829 32381-32381/com.goflipr.flipr W/dalvikvm: VFY: unable to resolve virtual method 44684: Lcom/tbruyelle/rxpermissions2/RxPermissionsFragment;.requestPermissions ([Ljava/lang/String;I)V
06-16 17:27:21.829 32381-32381/com.goflipr.flipr D/dalvikvm: VFY: replacing opcode 0x6e at 0x0002
06-16 17:27:21.959 32381-32441/com.goflipr.flipr D/dalvikvm: GC_FOR_ALLOC freed 1322K, 16% free 7361K/8756K, paused 54ms, total 54ms
06-16 17:27:22.420 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ╔════════════════════════════════════════════════════════════════════════════════════════
06-16 17:27:22.430 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ║ Thread: main
06-16 17:27:22.430 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ╟────────────────────────────────────────────────────────────────────────────────────────
06-16 17:27:22.440 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ║ LoadedApk$ReceiverDispatcher$Args.run  (LoadedApk.java:759)
06-16 17:27:22.440 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ║    ScanActivity$2.onReceive  (ScanActivity.java:138)
06-16 17:27:22.440 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ╟────────────────────────────────────────────────────────────────────────────────────────
06-16 17:27:22.440 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ║ STATE_TURNING_ON
06-16 17:27:22.440 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ╚════════════════════════════════════════════════════════════════════════════════════════
06-16 17:27:23.020 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ╔════════════════════════════════════════════════════════════════════════════════════════
06-16 17:27:23.030 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ║ Thread: main
06-16 17:27:23.030 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ╟────────────────────────────────────────────────────────────────────────────────────────
06-16 17:27:23.030 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ║ LoadedApk$ReceiverDispatcher$Args.run  (LoadedApk.java:759)
06-16 17:27:23.030 32381-32381/com.goflipr.flipr D/PRETTYLOGGER: ║    ScanActivity$2.onReceive  (ScanActivity.java:134)
[... cut ...]
06-16 17:27:35.092 32381-32381/com.goflipr.flipr I/Choreographer: Skipped 600 frames!  The application may be doing too much work on its main thread.
06-16 17:27:35.112 32381-32381/com.goflipr.flipr D/EventBus: No subscribers registered for event class com.goflipr.flipr.model.ble.BleError
06-16 17:27:35.112 32381-32381/com.goflipr.flipr D/EventBus: No subscribers registered for event class org.greenrobot.eventbus.NoSubscriberEvent
06-16 17:28:10.126 32381-32381/com.goflipr.flipr D/dalvikvm: GC_EXPLICIT freed 734K, 17% free 7312K/8756K, paused 6ms+8ms, total 80ms

```

dariuszseweryn, 2017-06-16:
From the log above I can only tell that `BluetoothAdapter.startLeScan()` happened to block the function call for 10 seconds. It is a custom ROM and this implementation seems to work differently than the original.