dariuszseweryn/RxAndroidBle #222: Can't Subscribe to Characteristic if Not Paired

lorenzowoodridge, 2017-06-14:
Whenever I try to set up a characteristic notification on a device that has yet to be paired it fails, but when I do set up the notification on a device that is paired it succeeds.

Code:

```
connectionSubscription = rxBleDevice.establishConnection(false)
                .observeOn(Schedulers.io())
                .flatMap(rxBleConnection1 -> {
                    Log.d(TAG, "connection made");
                    this.rxBleConnection = rxBleConnection1;

                    return rxBleConnection.setupNotification(CHARACTERISTIC_STATE_UUID);
                })
                .doOnNext(notificationObservable-> {
                    Log.d(TAG, "notifications setup");

                    airTurnCallback.deviceConnected(new AirTurnDevice(rxBleDevice));
                })
                .flatMap(notificationObservable -> notificationObservable)
                .subscribe(this::onStateChange, throwable -> {
                    Log.d(TAG, "Throwable: " + throwable);
                });
```

Logs:

```
D/BluetoothGatt: connect() - device: 88:6B:0F:40:FD:4C, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=73e33511-4341-4b5a-b7fd-a40032d25256
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=88:6B:0F:40:FD:4C
D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
D/AirTurnAPI: RxBleConnectionState{CONNECTED}
D/AirTurnAPI: connection made
D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(344003475)
D/RxBle#Radio:   QUEUED RxBleRadioOperationServicesDiscover(505411897)
D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(505411897)
D/BluetoothGatt: discoverServices() - device: 88:6B:0F:40:FD:4C
D/BluetoothGatt: onSearchComplete() = Device=88:6B:0F:40:FD:4C Status=0
D/RxBle#BluetoothGatt: onServicesDiscovered status=0
D/BluetoothGatt: setCharacteristicNotification() - uuid: 362f71a0-6c96-11e3-981f-0800200c9a66 enable: true
D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(479503017)
D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(505411897)
D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(479503017)
D/RxBle#Radio: FINISHED RxBleRadioOperationDescriptorWrite(479503017)
D/BluetoothGatt: setCharacteristicNotification() - uuid: 362f71a0-6c96-11e3-981f-0800200c9a66 enable: false
D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(810194717)
D/AirTurnAPI: Throwable: BleCannotSetCharacteristicNotificationException{bluetoothGattCharacteristic=362f71a0-6c96-11e3-981f-0800200c9a66, reason=CANNOT_WRITE_CLIENT_CHARACTERISTIC_CONFIG_DESCRIPTOR (see Javadoc for more comment)}
D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(810194717)
D/AirTurnAPI: RxBleConnectionState{DISCONNECTED}
D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(218007577)
D/ConnectFragment: AirTurnReceiver onReceive - com.airturn.broadcast.devicedisconnected
D/ConnectFragment: state: DISCONNECTED
D/DeviceView: DISCONNECTED
D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
D/RxBle#Radio: FINISHED RxBleRadioOperationDescriptorWrite(810194717)
D/RxBle#Radio:  STARTED RxBleRadioOperationDisconnect(218007577)
D/BluetoothManager: getConnectionState()
D/BluetoothManager: getConnectedDevices
D/BluetoothGatt: cancelOpen() - device: 88:6B:0F:40:FD:4C
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=88:6B:0F:40:FD:4C
D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=7
D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(218007577)
```

dariuszseweryn, 2017-06-16:
Hello @lorenzowoodridge 
Does this characteristic needs encryption? Can you give more information about properties / permissions it (and the descriptor) has?

lorenzowoodridge, 2017-06-16:
From our gatt.xml specification:
<properties read="true" notify="true" authenticated_read="true"/>

And my debug output:
----- characteristic UUID: 362f71a0-6c96-11e3-981f-0800200c9a66
----- characteristic properties: 18
----- characteristic permissions: 0
-------- descriptor UUID: 00002902-0000-1000-8000-00805f9b34fb
-------- descriptor permissions: 0
-------- descriptor UUID: 00002901-0000-1000-8000-00805f9b34fb
-------- descriptor permissions: 0

dariuszseweryn, 2017-06-16:
It looks weird - none of the listed items have any permissions. Anyway so far it does not seem to be a problem with the library so far. Feel free to investigate further if you think otherwise.

lorenzowoodridge, 2017-06-16:
It works perfectly if the device has already been paired previously.

What I thought was interesting is that the logs indicate setCharacteristicNotification is called twice

```
D/BluetoothGatt: setCharacteristicNotification() - uuid: 362f71a0-6c96-11e3-981f-0800200c9a66 enable: true
D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(479503017)
D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(505411897)
D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(479503017)
D/RxBle#Radio: FINISHED RxBleRadioOperationDescriptorWrite(479503017)
D/BluetoothGatt: setCharacteristicNotification() - uuid: 362f71a0-6c96-11e3-981f-0800200c9a66 enable: false
```

dariuszseweryn, 2017-06-16:
It is not called twice but reverted after the descriptor write fails - then the whole `.setupNotification()` function fails and the notification is tear down to not saturate the limit of active notifications.

dariuszseweryn, 2017-06-16:
Maybe the peripheral is returning different permissions if it is already paired?

Ezor, 2017-06-18:
Hi, i get the same things :
```
06-18 17:44:56.316 19237-19258/fr.ecomsoft.pocketpro D/RxBle#BluetoothGatt: onServicesDiscovered status=0
06-18 17:44:56.316 19237-19319/fr.ecomsoft.pocketpro D/BluetoothGatt: setCharacteristicNotification() - uuid: 18cda784-4bd3-4370-85bb-bfed91ec86af enable: true
06-18 17:44:56.321 19237-19319/fr.ecomsoft.pocketpro D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(1110660456)
06-18 17:44:56.321 19237-19250/fr.ecomsoft.pocketpro D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(1110487864)
06-18 17:44:56.321 19237-19250/fr.ecomsoft.pocketpro D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(1110660456)
06-18 17:44:56.346 19237-19237/fr.ecomsoft.pocketpro D/BluetoothGatt: writeDescriptor() - uuid: 00002902-0000-1000-8000-00805f9b34fb
06-18 17:44:56.726 19237-19249/fr.ecomsoft.pocketpro D/BluetoothGatt: onDescriptorWrite() - Device=EC:FE:7E:15:91:3A UUID=18cda784-4bd3-4370-85bb-bfed91ec86af
06-18 17:44:56.731 19237-19249/fr.ecomsoft.pocketpro D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=128
06-18 17:44:56.736 19237-19250/fr.ecomsoft.pocketpro D/RxBle#Radio: FINISHED RxBleRadioOperationDescriptorWrite(1110660456)
06-18 17:44:56.736 19237-19319/fr.ecomsoft.pocketpro D/BluetoothGatt: setCharacteristicNotification() - uuid: 18cda784-4bd3-4370-85bb-bfed91ec86af enable: false
06-18 17:44:56.741 19237-19319/fr.ecomsoft.pocketpro D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(1110764936)
06-18 17:44:56.741 19237-19250/fr.ecomsoft.pocketpro D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(1110764936)
06-18 17:44:56.746 19237-19237/fr.ecomsoft.pocketpro D/BluetoothGatt: writeDescriptor() - uuid: 00002902-0000-1000-8000-00805f9b34fb
06-18 17:44:56.771 19237-19319/fr.ecomsoft.pocketpro D/dalvikvm: GC_FOR_ALLOC freed 1369K, 30% free 6034K/8532K, paused 19ms, total 19ms
06-18 17:44:56.771 19237-19319/fr.ecomsoft.pocketpro E/TimeActivity: Error
                                                                     BleCannotSetCharacteristicNotificationException{bluetoothGattCharacteristic=18cda784-4bd3-4370-85bb-bfed91ec86af, reason=CANNOT_WRITE_CLIENT_CHARACTERISTIC_CONFIG_DESCRIPTOR (see Javadoc for more comment)}
                                                                         at com.polidea.rxandroidble.internal.connection.NotificationAndIndicationManager$10.call(NotificationAndIndicationManager.java:215)
                                                                         at com.polidea.rxandroidble.internal.connection.NotificationAndIndicationManager$10.call(NotificationAndIndicationManager.java:212)
                                                                         at rx.internal.operators.OperatorOnErrorResumeNextViaFunction$4.onError(OperatorOnErrorResumeNextViaFunction.java:140)
                                                                         at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:266)
                                                                         at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:818)
                                                                         at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:579)
                                                                         at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:568)
                                                                         at rx.internal.operators.OperatorMerge$InnerSubscriber.onError(OperatorMerge.java:852)
                                                                         at rx.internal.operators.OnSubscribeCreate$BaseEmitter.onError(OnSubscribeCreate.java:109)
                                                                         at rx.internal.util.ObserverSubscriber.onError(ObserverSubscriber.java:39)
                                                                         at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:153)
                                                                         at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
                                                                         at rx.internal.operators.OperatorUnsubscribeOn$1.onError(OperatorUnsubscribeOn.java:47)
                                                                         at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.onError(OperatorSubscribeOn.java:80)
                                                                         at rx.internal.operators.OnSubscribeDoOnEach$DoOnEachSubscriber.onError(OnSubscribeDoOnEach.java:87)
                                                                         at rx.internal.operators.OnSubscribeCreate$BaseEmitter.onError(OnSubscribeCreate.java:109)
                                                                         at com.polidea.rxandroidble.internal.util.RadioReleasingEmitterWrapper.onError(RadioReleasingEmitterWrapper.java:44)
                                                                         at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:153)
                                                                         at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
[... cut ...]
06-18 17:44:56.836 19237-19248/fr.ecomsoft.pocketpro D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=EC:FE:7E:15:91:3A
06-18 17:44:56.836 19237-19248/fr.ecomsoft.pocketpro D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
06-18 17:44:56.836 19237-19237/fr.ecomsoft.pocketpro D/BluetoothGatt: close()
06-18 17:44:56.836 19237-19237/fr.ecomsoft.pocketpro D/BluetoothGatt: unregisterApp() - mClientIf=5
06-18 17:44:56.836 19237-19250/fr.ecomsoft.pocketpro D/RxBle#Radio: FINISHED RxBleRadioOperationDisconnect(1110319560)
```

Any idea ?

dariuszseweryn, 2017-06-18:
@Ezor Check what `status=128` means. It is probably that you try to set too many notifications (Android has limits) or a notification of a wrong type. It is probably not related to the topic of this issue.

Ezor, 2017-06-18:
@dariuszseweryn I just fixed the problem by using `setupIndication` instead of `setupNotification`. Thanks anyway

lorenzowoodridge, 2017-06-19:
The permissions are the same even when the device has already been paired before

dariuszseweryn, 2017-06-19:
I think that you may have a bug on the peripheral side. I can see that your characteristic supports reads and notifies. What I do not see is that your `Client Characteristic Configuration Descriptor` supports writes. This may be causing the problem.

Either way. Currently the library is not supporting encrypted characteristics so it is up to the user to work it out.

lorenzowoodridge, 2017-06-21:
I don't believe the characteristic is encrypted, and I'm fairly certain the bug is not on the peripheral side because I can subscribe to characteristics just fine both after pairing for the first time and also dealing with an already paired device.

dariuszseweryn, 2017-07-27:
To sum up:
* Out of the box support for bonding and usage of encrypted characteristics is not yet available — there is an already opened issue for it though (#35 and #47) — Once a bulletproof solution will be developed we will add it to the library. If anyone would like to contribute — you are welcome. :)
* For user's that will try to perform an unsupported operation on a specific characteristic a warning log will be displayed (#224). When a new API (#239) will be introduced the check will throw an error by default (so people will not be surprised why they get an error if they try to i.e. read a characteristic that is not readable) but the user will be able to suppress it for a given connection.

Closing.

closed 2017-07-27 by dariuszseweryn