dariuszseweryn/RxAndroidBle #223: No scanResults in background service after successful read/write operation (starting from 1.2.3)

lukevr, 2017-06-18:
### Summary

I.m using service for the continuous background scan and operations with BLE devices and everything worked fine before 1.2.3 release. Now, after successful read/write operation on find BLE device, I am stopping to receive scan results. If I downgrade rxandroidble to 1.2.2  - everything works fine.

#### Preconditions
public class MyService extends Service {
...
Observable<RxBleScanResult> scanResults = rxBleClient
                .scanBleDevices()
                .share();

...

beaconSubscription = scanResults.
                observeOn(AndroidSchedulers.mainThread()).
                subscribe(rxBleScanResult -> {
                            if (sharedPref.getBoolean("smart_settings", false)) {
                                framesCounter++;
                            }
                            checkBeaconCandidat(rxBleScanResult);
                        },
                        throwable -> {
                            Log.e(TAG, throwable.toString());
                            throwable.printStackTrace();
                        });
...

writeSubscription = rxBleScanResult.getBleDevice().establishConnection(false)
                .flatMap(rxBleConnection -> rxBleConnection.readCharacteristic(characteristicUuid)
                        .doOnNext(bytes -> {
                            bytesToWrite[0] = getSomeBytesToWriteFromReadResults();
                        })
                        .flatMap(bytes -> rxBleConnection.writeCharacteristic(characteristicUuid, bytesToWrite[0])))
                // retry 3 times in case of error
                //        .retryWhen(errors -> errors.zipWith(Observable.range(1, 3), (n, i) -> i))
                .retryWhen(new Func1<Observable<? extends Throwable>, Observable<?>>() {
                    @Override
                    public Observable<?> call(Observable<? extends Throwable> errorNotification) {
                        return errorNotification
                                .flatMap(new Func1<Throwable, Observable<?>>() {
                                    @Override
                                    public Observable<?> call(Throwable throwable) {
                                        Log.e(TAG, "connect retrying... !!!!!!!!!!!!!!!!!!!");
                                        return Observable.timer(5, TimeUnit.SECONDS);
                                    }
                                });
                    }
                })
                .subscribe(
                        writeBytes -> {
                            // All is OK - reset subscription and Enter event
                            Log.e(TAG, "Write SUCCESS to " + characteristicUuid);
                            resetEnter();
                        },
                        throwable -> {
                            Log.e(TAG, throwable.toString());
                            throwable.printStackTrace();
                            // Reset Enter subscription second time even in case of exception
                            Log.e(TAG, "Write FAIL!!!!!!!!!!!!!!!!!!!!! ");
                            resetEnter();
                        }
                );

....
private void resetEnter() {

        if (writeSubscription != null && !writeSubscription.isUnsubscribed())
            writeSubscription.unsubscribe();
        writeSubscription = null;
    }
[... 71 lines cut ...]
D/RxBle#BluetoothGatt: onServicesDiscovered status=0
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(163596181)
D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(202142970)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(163596181)
W/BluetoothGatt: onCharacteristicRead() - Device=B8:27:EB:B6:A2:34 handle=12 Status=0
D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=ffffffff-ffff-ffff-ffff-fffffffffff1 status=0
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(165270862)
D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(163596181)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(165270862)
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=ffffffff-ffff-ffff-ffff-fffffffffff1 status=0
D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(24210304)
D/RxBle#Radio:   QUEUED RxBleRadioOperationScan(138526636)
D/BluetoothGatt: onClientConnectionState() - status=19 clientIf=7 device=B8:27:EB:B6:A2:34
D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=19
D/BluetoothMapService: onReceive
D/BluetoothMapService: onReceive: android.bluetooth.device.action.ACL_DISCONNECTED
I/TrustAgent.Tracker: [BluetoothConnectionTracker] Bluetooth disconnect broadast for MyApp B8:27:EB:B6:A2:34
D/BluetoothAdapter: isLeEnabled(): ON
D/BtGatt.GattService: stopScan() - queue size =2
D/BtGatt.GattService: unregisterClient() - clientIf=6
D/BtGatt.GattService: onScanFilterParamsConfigured() - clientIf=6, status=0, action=1, availableSpace=30
... 
**and no scan results available in the service**

**I've tried with 1.3.0 - the same result, scanning is not recovered after read/write operation.**

dariuszseweryn, 2017-06-19:
Try version `1.3.1` and post the logs.

lukevr, 2017-06-22:
Hi, sorry for the late answer, I have tested with 1.3.1 and seems it works in background service again.
Here are logs:

```
D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(203378203)
D/RxBle#Radio:  STARTED RxBleRadioOperationConnect(203378203)
D/BluetoothGatt: connect() - device: B8:27:EB:B6:A2:34, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=12c86bab-8c1c-48f2-bbce-073d1bc0bba3
D/BtGatt.GattService: registerClient() - UUID=12c86bab-8c1c-48f2-bbce-073d1bc0bba3
D/BtGatt.GattService: onClientRegistered() - UUID=12c86bab-8c1c-48f2-bbce-073d1bc0bba3, clientIf=11
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
D/BtGatt.GattService: clientConnect() - address=B8:27:EB:B6:A2:34, isDirect=true
D/bt_btif_config: btif_get_address_type: Device [b8:27:eb:b6:a2:34] address type 0
D/bt_btif_config: btif_get_device_type: Device [b8:27:eb:b6:a2:34] type 2
D/BluetoothAdapter: isLeEnabled(): ON
D/BtGatt.GattService: stopScan() - queue size =3
D/BtGatt.GattService: unregisterClient() - clientIf=10
D/BtGatt.ScanManager: configureRegularScanParams() - queue=2
D/BtGatt.ScanManager: configureRegularScanParams() - ScanSetting Scan mode=0 mLastConfiguredScanSetting=2
D/BtGatt.ScanManager: configureRegularScanParams - scanInterval = 8000configureRegularScanParams - scanWindow = 800
D/BtGatt.GattService: onScanParamSetupCompleted : 0
W/bt_smp: smp_br_connect_callback is called on unexpected transport 2
W/bt_btif: bta_dm_acl_change info: 0x0
D/bt_btif_dm: remote version info [b8:27:eb:b6:a2:34]: 0, 0, 0
E/bt_btif: bta_gattc_cache_load: can't open GATT cache file /data/misc/bluetooth/gatt_cache_b827ebb6a234 for reading, error: No such file or directory
E/BluetoothRemoteDevices: state12newState0
D/BtGatt.GattService: onConnected() - clientIf=11, connId=11, address=B8:27:EB:B6:A2:34
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=11 device=B8:27:EB:B6:A2:34
D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
I/TrustAgent.Tracker: [BluetoothConnectionTracker] Bluetooth connect broadast for PicPic B8:27:EB:B6:A2:34
D/RxBle#Radio:   QUEUED RxBleRadioOperationServicesDiscover(193464282)
D/RxBle#Radio: FINISHED RxBleRadioOperationConnect(203378203)
D/RxBle#Radio:  STARTED RxBleRadioOperationServicesDiscover(193464282)
D/BluetoothGatt: discoverServices() - device: B8:27:EB:B6:A2:34
D/BtGatt.GattService: discoverServices() - address=B8:27:EB:B6:A2:34, connId=11
W/bt_bta_gattc: bta_gattc_explore_srvc no more services found
D/BtGatt.GattService: onSearchCompleted() - connId=11, status=0
D/bt_bta_gattc: bta_gattc_get_gatt_db
D/BtGatt.GattService: onGetGattDb() - address=B8:27:EB:B6:A2:34
D/BtGatt.GattService: got service with UUID=00001800-0000-1000-8000-00805f9b34fb
D/BtGatt.GattService: got characteristic with UUID=00002a00-0000-1000-8000-00805f9b34fb
D/BtGatt.GattService: got characteristic with UUID=00002a01-0000-1000-8000-00805f9b34fb
D/BtGatt.GattService: got service with UUID=00001801-0000-1000-8000-00805f9b34fb
D/BtGatt.GattService: got characteristic with UUID=00002a05-0000-1000-8000-00805f9b34fb
D/BtGatt.GattService: got descriptor with UUID=00002902-0000-1000-8000-00805f9b34fb
D/BtGatt.GattService: got service with UUID=1d038ab3-45f4-4b96-9915-22ec29b75d58
D/BtGatt.GattService: got characteristic with UUID=ffffffff-ffff-ffff-ffff-fffffffffff1
D/BluetoothGatt: onSearchComplete() = Device=B8:27:EB:B6:A2:34 Status=0
D/RxBle#BluetoothGatt: onServicesDiscovered status=0
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicRead(43664172)
D/RxBle#Radio: FINISHED RxBleRadioOperationServicesDiscover(193464282)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicRead(43664172)
I/bt_btif_gatt: set_read_value len = 21 
W/BluetoothGatt: onCharacteristicRead() - Device=B8:27:EB:B6:A2:34 handle=12 Status=0
D/RxBle#BluetoothGatt: onCharacteristicRead characteristic=ffffffff-ffff-ffff-ffff-fffffffffff1 status=0
E/JetBeepLog: Read IV from ffffffff-ffff-ffff-ffff-fffffffffff1
D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(152824366)
D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicRead(43664172)
D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(152824366)
E/bt_att: value resp op_code = ATT_RSP_PREPARE_WRITE len = 22
I/NuPlayerDecoder: [audio] saw output EOS
E/bt_att: value resp op_code = ATT_RSP_PREPARE_WRITE len = 18
D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=ffffffff-ffff-ffff-ffff-fffffffffff1 status=0
D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(226994713)
D/RxBle#Radio:   QUEUED RxBleRadioOperationScanApi21(118409100)
D/RxBle#Radio: FINISHED RxBleRadioOperationCharacteristicWrite(152824366)
D/RxBle#Radio:  STARTED RxBleRadioOperationDisconnect(226994713)
D/BluetoothManager: getConnectionState()
D/BluetoothManager: getConnectedDevices
[... 27 lines cut ...]
W/bt_btif: bta_gattc_conn_cback() - cif=8 connected=0 conn_id=8 reason=0x0016
W/bt_btif: bta_gattc_conn_cback() - cif=9 connected=0 conn_id=9 reason=0x0016
W/bt_btif: bta_gattc_conn_cback() - cif=10 connected=0 conn_id=10 reason=0x0016
I/bt_btm_sec: btm_sec_disconnected clearing pending flag handle:64 reason:22
E/BluetoothRemoteDevices: state12newState1
D/BluetoothMapService: onReceive
D/BluetoothMapService: onReceive: android.bluetooth.device.action.ACL_DISCONNECTED
I/TrustAgent.Tracker: [BluetoothConnectionTracker] Bluetooth disconnect broadast for MyApp B8:27:EB:B6:A2:34
```
**Scan starts..
[... cut ...]
D/BtGatt.GattService: registerClient() - UUID=b1aebcf2-8a1c-4711-8c9d-3e85585183a2
D/BtGatt.GattService: onClientRegistered() - UUID=b1aebcf2-8a1c-4711-8c9d-3e85585183a2, clientIf=6
D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6 mClientIf=0
D/BtGatt.GattService: start scan with filters
D/BtGatt.ScanManager: handling starting scan
D/BtGatt.ScanManager: configureRegularScanParams() - queue=3
D/BtGatt.ScanManager: configureRegularScanParams() - ScanSetting Scan mode=2 mLastConfiguredScanSetting=2
```

dariuszseweryn, 2017-06-22:
There was a bug in between `1.2.3` and `1.3.0` where unsubscribing from operations would cause the library to get stuck. It was fixed in `1.3.1`. Good that it works now.

closed 2017-06-22 by dariuszseweryn