dariuszseweryn/RxAndroidBle #227: Scanning Ble

leof23, 2017-06-20:
### Summary
Hi, I'm developing an app for reading a ble device, I'm using a Switch to control the Scan event, if I pass the switch between on/off quickly, sometimes I received results but later, after some calls to the event, I don't receive more results, I'm using a Samsung Galaxy tab s2 with Android 7, and using Kotlin for my code
#### Minimum code snippet reproducing the issue
My Scan method
`fun scanDevices(name: String) {
        Log.i("arrive", "to the method")
        if (!connected) {
            Log.i("arrive", "to the if")
            val scanSubscription = rxBleClient.scanBleDevices(
                    ScanSettings.Builder()
                            .setScanMode(ScanSettings.MATCH_MODE_AGGRESSIVE)
                            .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                            .build(),
                    ScanFilter.empty())
                    .doOnSubscribe { Log.i("arrive","do subscribe") }
                    .doOnError { throwable -> throwable.printStackTrace() }
                    .doOnUnsubscribe { Log.w("arrive", "do unSubscribe") }
                    .subscribe(
                            { rxBleScanResult ->
                                Log.i("arrive", "scan")
                                if (rxBleScanResult?.bleDevice?.name == name) {
                                    device = rxBleScanResult.bleDevice
                                    Log.i("found", "device")
                                    found()
                                }
                            }
                    ) { throwable -> throwable.printStackTrace() }
            compositeSubscriptions.add(scanSubscription)
        }
    }`
#### Logs from the application running
06-20 11:22:38.842 21148-21148/com.cartier.croxel.cartierApp D/ViewRootImpl@232e253[MainActivity]: ViewPostImeInputStage processPointer 0
06-20 11:22:38.898 21148-21148/com.cartier.croxel.cartierApp D/ViewRootImpl@232e253[MainActivity]: ViewPostImeInputStage processPointer 1
06-20 11:22:38.907 21148-21148/com.cartier.croxel.cartierApp I/arrive: to the method
06-20 11:22:38.907 21148-21148/com.cartier.croxel.cartierApp I/arrive: to the if
06-20 11:22:38.908 21148-21148/com.cartier.croxel.cartierApp I/arrive: do subscribe
06-20 11:22:38.925 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:38.926 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:38.926 21148-21148/com.cartier.croxel.cartierApp D/BluetoothLeScanner: Start Scan
06-20 11:22:38.928 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:38.929 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:38.931 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:38.932 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:39.037 21148-21162/com.cartier.croxel.cartierApp D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6 mClientIf=0
06-20 11:22:40.499 21148-21198/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-20 11:22:40.499 21148-21198/com.cartier.croxel.cartierApp D/ScanRecord: first manudata for manu ID
06-20 11:22:40.500 21148-21148/com.cartier.croxel.cartierApp I/arrive: scan
06-20 11:22:40.500 21148-21198/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-20 11:22:40.500 21148-21198/com.cartier.croxel.cartierApp D/ScanRecord: first manudata for manu ID
06-20 11:22:40.506 21148-21148/com.cartier.croxel.cartierApp I/arrive: scan
06-20 11:22:40.510 21148-21162/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-20 11:22:40.510 21148-21162/com.cartier.croxel.cartierApp D/ScanRecord: first manudata for manu ID
06-20 11:22:40.511 21148-21148/com.cartier.croxel.cartierApp I/arrive: scan
06-20 11:22:40.573 21148-21164/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-20 11:22:40.573 21148-21164/com.cartier.croxel.cartierApp D/ScanRecord: first manudata for manu ID
06-20 11:22:40.573 21148-21148/com.cartier.croxel.cartierApp I/arrive: scan
06-20 11:22:40.584 21148-21198/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-20 11:22:40.585 21148-21198/com.cartier.croxel.cartierApp D/ScanRecord: first manudata for manu ID
06-20 11:22:40.585 21148-21148/com.cartier.croxel.cartierApp I/arrive: scan
06-20 11:22:40.611 21148-21162/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-20 11:22:40.611 21148-21162/com.cartier.croxel.cartierApp D/ScanRecord: first manudata for manu ID
06-20 11:22:40.612 21148-21148/com.cartier.croxel.cartierApp I/arrive: scan
06-20 11:22:40.650 21148-21164/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-20 11:22:40.651 21148-21164/com.cartier.croxel.cartierApp D/ScanRecord: first manudata for manu ID
06-20 11:22:40.651 21148-21148/com.cartier.croxel.cartierApp I/arrive: scan
06-20 11:22:40.670 21148-21198/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-20 11:22:40.670 21148-21198/com.cartier.croxel.cartierApp D/ScanRecord: first manudata for manu ID
06-20 11:22:40.671 21148-21148/com.cartier.croxel.cartierApp I/arrive: scan
06-20 11:22:40.711 21148-21162/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-20 11:22:40.711 21148-21162/com.cartier.croxel.cartierApp D/ScanRecord: first manudata for manu ID
[... 395 lines cut ...]
06-20 11:22:47.386 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:47.387 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:47.389 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:47.389 21148-21148/com.cartier.croxel.cartierApp D/BluetoothLeScanner: Stop Scan
06-20 11:22:48.844 21148-21148/com.cartier.croxel.cartierApp D/ViewRootImpl@232e253[MainActivity]: ViewPostImeInputStage processPointer 0
06-20 11:22:48.978 21148-21148/com.cartier.croxel.cartierApp D/ViewRootImpl@232e253[MainActivity]: ViewPostImeInputStage processPointer 1
06-20 11:22:48.980 21148-21148/com.cartier.croxel.cartierApp I/arrive: to the method
06-20 11:22:48.980 21148-21148/com.cartier.croxel.cartierApp I/arrive: to the if
06-20 11:22:48.980 21148-21148/com.cartier.croxel.cartierApp I/arrive: do subscribe
06-20 11:22:48.988 21148-21148/com.cartier.croxel.cartierApp V/InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@1a98123 nm : com.cartier.croxel.cartierApp ic=null
06-20 11:22:48.988 21148-21148/com.cartier.croxel.cartierApp I/InputMethodManager: [IMM] startInputInner - mService.startInputOrWindowGainedFocus
06-20 11:22:48.990 21148-21148/com.cartier.croxel.cartierApp D/InputTransport: Input channel constructed: fd=80
06-20 11:22:48.990 21148-21148/com.cartier.croxel.cartierApp D/InputTransport: Input channel destroyed: fd=84
06-20 11:22:49.000 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:49.002 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:49.002 21148-21148/com.cartier.croxel.cartierApp D/BluetoothLeScanner: Start Scan
06-20 11:22:49.003 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:49.004 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:49.006 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:49.008 21148-21148/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-20 11:22:49.113 21148-21164/com.cartier.croxel.cartierApp D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6 mClientIf=0
#### Actual result
As you can see in the log, in the final part the method only reach to the doOnSubscribe, but never scan again until i turn the switch on/off again
#### Expected result
My spected result is whenever i activate the switch, I need to scan util I can found the device

dariuszseweryn, 2017-06-20:
Hello @leof23 
You are probably hitting [this issue](https://blog.classycode.com/undocumented-android-7-ble-behavior-changes-d1a9bd87d983)
If that is the case the library can do nothing to prevent this behaviour. It may be feasible to emit a new `BleScanException` in that case or the library would need to throttle scan requests on Android 7. I will think about possible solutions.

leof23, 2017-06-20:
yes, that was my issue, thanks for the answer

dariuszseweryn, 2017-06-20:
I will keep this issue opened as a reminder for me.

closed 2017-06-29 by dariuszseweryn

dariuszseweryn, 2017-07-04:
Is available in `1.4.0-SNAPSHOT`

darrinps, 2018-09-11:
I'm seeing something similar to this:

   `Failed to establish connection Undocumented scan throttle (code 2147483646), suggested retry date is Wed Jul 11 11:18:06 PDT 2018
                                                                                         com.polidea.rxandroidble2.exceptions.BleScanException: Undocumented scan throttle (code 2147483646), suggested retry date is Wed Jul 11 11:18:06 PDT 2018
                                                                                             at com.polidea.rxandroidble2.internal.scan.ScanPreconditionsVerifierApi24.verify(ScanPreconditionsVerifierApi24.java:51)
                                                                                             at com.polidea.rxandroidble2.RxBleClientImpl$1.call(RxBleClientImpl.java:118)
                                                                                             at com.polidea.rxandroidble2.RxBleClientImpl$1.call(RxBleClientImpl.java:115)
                                                                                             at io.reactivex.internal.operators.observable.ObservableDefer.subscribeActual(ObservableDefer.java:32)
                                                                                             at io.reactivex.Observable.subscribe(Observable.java:12051)
                                                                                             at io.reactivex.internal.operators.observable.ObservableDoOnEach.subscribeActual(ObservableDoOnEach.java:42)
                                                                                             at io.reactivex.Observable.subscribe(Observable.java:12051)
                                                                                             at io.reactivex.internal.operators.observable.ObservableFilter.subscribeActual(ObservableFilter.java:30)
                                                                                             at io.reactivex.Observable.subscribe(Observable.java:12051)
                                                                                             at io.reactivex.internal.operators.observable.ObservableElementAtSingle.subscribeActual(ObservableElementAtSingle.java:37)
                                                                                             at io.reactivex.Single.subscribe(Single.java:3310)
                                                                                             at io.reactivex.internal.operators.single.SingleTimeout.subscribeActual(SingleTimeout.java:53)
                                                                                             at io.reactivex.Single.subscribe(Single.java:3310)
                                                                                             at io.reactivex.internal.operators.single.SingleDoOnError.subscribeActual(SingleDoOnError.java:35)
                                                                                             at io.reactivex.Single.subscribe(Single.java:3310)
                                                                                             at io.reactivex.internal.operators.single.SingleMap.subscribeActual(SingleMap.java:34)
                                                                                             at io.reactivex.Single.subscribe(Single.java:3310)
                                                                                             at io.reactivex.internal.operators.single.SingleDefer.subscribeActual(SingleDefer.java:43)
                                                                                             at io.reactivex.Single.subscribe(Single.java:3310)`

It's happening on multiple devices running on various versions of Android up to and including those running Android Pie.  What I think would be nice, is if a flag could be set for RxAndroidBle that tells it to automatically scan at the suggested retry time. That time seems like it's around 10 seconds or so into the future most of the time.

dariuszseweryn, 2018-09-13:
Feel free to add a feature request though I think it is simple to create it without bloating the library's code base by utilising `.retryWhen()` operator and triggering the retry when the time will pass

devDeejay, 2019-11-11:
Make sure you are not calling the `scanBluetoothDevices()` method too frequently

vegeta2102, 2024-03-21:
@devDeejay 
@dariuszseweryn 
What do you mean about "too frequently" calling the `scanBluetoothDevices()`
I am using `retryWhen()` to retry scan when exception for every 5 seconds.

It's look like below

```kotlin

.scanBleDevices(scanSetting, scanFilters)
.retryWhen {
                                it.flatMap { error ->
                                    when (error) {
                                        is BleScanException -> {
                                            Observable.timer(5000, TimeUnit.MILLISECONDS)
                                        }

```

and sometimes I got same error in this issue.

vegeta2102, 2024-03-21:
Actually , I got this error 

```

com.polidea.rxandroidble2.exceptions.BleScanException: Scan failed because application registration failed (code 6)
com.polidea.rxandroidble2.exceptions.BleScanException: Undocumented scan throttle (code 2147483646), suggested retry date is Thu Mar 21 11:03:02 GMT+09:00 2024

```

dariuszseweryn, 2024-03-21:
Hey @vegeta2102 
> What do you mean about "too frequently" calling the scanBluetoothDevices()

Please check the merged PR above description. It has a good description and a link to source where I learned about this behaviour. 

TL;DR — one can start up to 5 scans every 30 seconds.

vegeta2102, 2024-05-08:
@dariuszseweryn 
Totally understand ! Thank you