dariuszseweryn/RxAndroidBle #229: Asus Scanning issue

leof23, 2017-06-21:
### Summary
Hi, I'm having a weird behaviour when I scanning for BLE devices, I'm testing with 3 different devices, a Samsung Galaxy Tab 2 with Android 7, a Samsung Galaxy S4 with Android 5.0.1, and a tablet Asus P01Z with Android 5.0.2, my version of the library is 1.3.1 and I'm using Kotlin for my code
#### Preconditions
try to Scan for BLE devices
#### Minimum code snippet reproducing the issue
My Scan method
`    fun scanDevices(name: String) {
        Log.i("arrive", "to the method")
        if (!connected) {
            Log.i("arrive", "to the if")
            val scanSubscription = rxBleClient.scanBleDevices(
                    ScanSettings.Builder()
                            .setScanMode(ScanSettings.MATCH_MODE_AGGRESSIVE)
                            .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                            .build(),
                    ScanFilter.empty()
                    )
                    .doOnSubscribe { Log.i("arrive","do subscribe") }
                    .doOnError { throwable -> throwable.printStackTrace() }
                    .doOnUnsubscribe { Log.w("arrive", "do unSubscribe") }
                    .subscribe(
                            { rxBleScanResult ->
                                Log.i("arrive", "start scanning")
                                if (rxBleScanResult != null) {
                                    Log.i("arrive", "rxBleScanResult is not null")
                                    if (rxBleScanResult.bleDevice != null) {
                                        Log.i("arrive", "bleDevice is not null")
                                        if (rxBleScanResult.bleDevice.name != null) {
                                            Log.i("arrive", "name is not null")
                                            if (rxBleScanResult.bleDevice.name == name) {
                                                device = rxBleScanResult.bleDevice
                                                Log.i("found", "device")
                                                found()
                                            }
                                        }
                                    }
                                }
                            },{ throwable -> throwable.printStackTrace() }
                    )
            compositeSubscriptions.add(scanSubscription)
        }
    }`
#### Logs from the application running
# Samsung Galaxy tab 2(Android 7) Log:
06-21 11:34:40.698 20701-20701/com.cartier.croxel.cartierApp D/ViewRootImpl@5407fdf[MainActivity]: ViewPostImeInputStage processPointer 0
06-21 11:34:40.785 20701-20701/com.cartier.croxel.cartierApp D/ViewRootImpl@5407fdf[MainActivity]: ViewPostImeInputStage processPointer 1
06-21 11:34:40.793 20701-20701/com.cartier.croxel.cartierApp I/arrive: to the method
06-21 11:34:40.793 20701-20701/com.cartier.croxel.cartierApp I/arrive: to the if
06-21 11:34:40.793 20701-20701/com.cartier.croxel.cartierApp I/arrive: do subscribe
06-21 11:34:40.803 20701-20701/com.cartier.croxel.cartierApp V/InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@7327299 nm : com.cartier.croxel.cartierApp ic=null
06-21 11:34:40.804 20701-20701/com.cartier.croxel.cartierApp I/InputMethodManager: [IMM] startInputInner - mService.startInputOrWindowGainedFocus
06-21 11:34:40.805 20701-20701/com.cartier.croxel.cartierApp D/InputTransport: Input channel constructed: fd=86
06-21 11:34:40.805 20701-20701/com.cartier.croxel.cartierApp D/InputTransport: Input channel destroyed: fd=83
06-21 11:34:40.813 20701-20701/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-21 11:34:40.815 20701-20701/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-21 11:34:40.815 20701-20701/com.cartier.croxel.cartierApp D/BluetoothLeScanner: Start Scan
06-21 11:34:40.816 20701-20701/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-21 11:34:40.817 20701-20701/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-21 11:34:40.819 20701-20701/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-21 11:34:40.821 20701-20701/com.cartier.croxel.cartierApp D/BluetoothAdapter: STATE_ON
06-21 11:34:40.926 20701-20715/com.cartier.croxel.cartierApp D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5 mClientIf=0
06-21 11:34:42.592 20701-20714/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-21 11:34:42.592 20701-20701/com.cartier.croxel.cartierApp I/arrive: start scanning
06-21 11:34:42.592 20701-20701/com.cartier.croxel.cartierApp I/arrive: rxBleScanResult is not null
06-21 11:34:42.592 20701-20701/com.cartier.croxel.cartierApp I/arrive: bleDevice is not null
06-21 11:34:42.595 20701-20701/com.cartier.croxel.cartierApp I/arrive: name is not null
06-21 11:34:42.654 20701-20715/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-21 11:34:42.654 20701-20715/com.cartier.croxel.cartierApp D/ScanRecord: first manudata for manu ID
06-21 11:34:42.655 20701-20701/com.cartier.croxel.cartierApp I/arrive: start scanning
06-21 11:34:42.655 20701-20701/com.cartier.croxel.cartierApp I/arrive: rxBleScanResult is not null
[... 218 lines cut ...]
06-21 11:08:21.562 17455-17455/com.cartier.croxel.cartierApp I/arrive: bleDevice is not null
06-21 11:08:21.592 17455-17469/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-21 11:08:21.592 17455-17469/com.cartier.croxel.cartierApp D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=00:0B:57:15:88:1F, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=Empty Example], mRssi=-64, mTimestampNanos=11257648229346}
06-21 11:08:21.592 17455-17455/com.cartier.croxel.cartierApp I/arrive: start scanning
06-21 11:08:21.592 17455-17455/com.cartier.croxel.cartierApp I/arrive: rxBleScanResult is not null
06-21 11:08:21.592 17455-17455/com.cartier.croxel.cartierApp I/arrive: bleDevice is not null
06-21 11:08:21.592 17455-17455/com.cartier.croxel.cartierApp I/arrive: name is not null
06-21 11:08:21.602 17455-17455/com.cartier.croxel.cartierApp W/arrive: do unSubscribe
06-21 11:08:21.622 17455-17469/com.cartier.croxel.cartierApp D/ScanRecord: parseFromBytes
06-21 11:08:21.622 17455-17469/com.cartier.croxel.cartierApp D/ScanRecord: first manudata for manu ID
06-21 11:08:21.622 17455-17469/com.cartier.croxel.cartierApp D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=1C:E9:71:2E:4C:C6, mScanRecord=ScanRecord [mAdvertiseFlags=-1, mServiceUuids=null, mManufacturerSpecificData={6=[1, 9, 32, 0, -79, -14, -79, 23, -99, -59, 29, -44, 6, -58, -82, -111, 89, -6, 4, 15, 74, -83, -114, 101, 41, 67, 8]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-67, mTimestampNanos=11257679326758}

# Asus P01Z(Android 5.0.2) Log
06-21 11:48:36.557 7650-7650/com.cartier.croxel.cartierApp I/arrive: to the method
06-21 11:48:36.557 7650-7650/com.cartier.croxel.cartierApp I/arrive: to the if
06-21 11:48:36.558 7650-7650/com.cartier.croxel.cartierApp I/arrive: do subscribe
06-21 11:48:36.583 7650-7669/com.cartier.croxel.cartierApp D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6
06-21 11:48:36.658 7650-7670/com.cartier.croxel.cartierApp D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=60:D0:12:3D:33:DB, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={76=[16, 2, 11, 0]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-66, mTimestampNanos=11625251599585}
06-21 11:48:36.668 7650-7650/com.cartier.croxel.cartierApp I/arrive: start scanning
06-21 11:48:36.669 7650-7650/com.cartier.croxel.cartierApp I/arrive: rxBleScanResult is not null
06-21 11:48:36.669 7650-7650/com.cartier.croxel.cartierApp I/arrive: bleDevice is not null
#### Actual result
As you can see in the logs when the name if dont execute the asus tablet stops the scan for no reasons, but the other devices works fine with the same code
#### Expected result
Continue scanning in the asus tab

dariuszseweryn, 2017-06-21:
First of all you can create a `ScanFilter` with a name filtering. Secondly I do not see anything in the log from the Asus. No `Stop Scan` or any error. I do not have this device to check it by myself here either. Feel free to dig deeper.

leof23, 2017-06-21:
I implement the ScanFilter but is like you say I can't see any Stop Scan or something similar in the logs, I don't know why is not scanning correctly

dariuszseweryn, 2017-06-22:
It looks like a model specific Android BLE implementation bug. I do not have access to this model so unfortunately I can do nothing remotely. You can try to find a solution yourself and create a pull request or post your findings here.

lukevr, 2017-06-23:
Hi, is there a list of supported(tested) and unsupported devices/Android versions?

dariuszseweryn, 2017-06-23:
Hello @lukevr
No — there is no such list.
We are doing our best to make the library as generic as possible without the need to maintain a model/vendor specific workarounds in the code.

closed 2017-06-29 by dariuszseweryn