dariuszseweryn/RxAndroidBle #230: RxRadio Queue Stalls

kenwdelong, 2017-06-21:
### Summary
Entering the bootloader mode for a Nordic device causes the RxBle#Radio queue to stop executing.

#### Preconditions
We are using the Nordic nRF51822 chip for our BLE devices, SDK version 0.8.  In order to do a firmware upgrade, we need to put the device into bootloader mode.  This involves writing [0x01, 0x04] to the DFU Control Characteristic 00001531-1212-efde-1523-785feabcd123, which causes the Soft Device to shut down and the chip to reboot.  The connection is dropped immediately.

#### Steps to reproduce actual result
<br />1.  Write 0x01, 0x04 to 00001531-1212-efde-1523-785feabcd123.
<br />2.  Soft Device shuts down and chip reboots. 20 seconds later Android detects the connection loss.
<br />3.  We proceed with the DFU firmware upgrade using Nordic's library https://github.com/NordicSemiconductor/Android-DFU-Library
<br />4. After it returns, we cannot scan or connect to any devices because the RxBle#Radio queue no longer executes.

#### Minimum code snippet reproducing the issue
`rxBleConnection.writeCharacteristic(characteristic, new byte[] { 0x01, 0x04 })`

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
	06-21 16:08:59.421 18175-18175/com.hatchbaby.rest.qa I/Nightlight: [main] Starting bootloader
	06-21 16:08:59.427 18175-18733/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED RxBleRadioOperationCharacteristicWrite(78713072)
	06-21 16:08:59.429 18175-18209/com.hatchbaby.rest.qa D/RxBle#Radio:  STARTED RxBleRadioOperationCharacteristicWrite(78713072)
	06-21 16:09:19.491 18175-18261/com.hatchbaby.rest.qa D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=00001531-1212-efde-1523-785feabcd123 status=133
	06-21 16:09:19.492 18175-18261/com.hatchbaby.rest.qa D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=E5:D6:FC:68:FB:55
	06-21 16:09:19.493 18175-18261/com.hatchbaby.rest.qa D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
	06-21 16:09:19.496 18175-18261/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(109194973)

Here we see our write to the DFU Control Point is STARTED by the RxBle#Radio as operation 78713072.  It goes through QUEUED, STARTED, then it is never heard from again.  20 seconds later, Android realizes that the connection is broken, and a disconnect operation is QUEUED as operation 109194973, but it never appears again in the logs - never STARTED, never even REMOVED.

After we successfully upgrade the firmware, we try to scan and reconnect to the device.  But the operations never start.

	06-21 16:10:01.690 18175-18213/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED RxBleRadioOperationScanApi21(261606157)
	06-21 16:10:04.690 18175-18314/com.hatchbaby.rest.qa W/Nightlight: [RxComputationScheduler-6] Unable to scan device
	06-21 16:10:04.701 18175-18314/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED RxBleRadioOperationConnect(218129354)
	06-21 16:10:04.702 18175-18175/com.hatchbaby.rest.qa D/RxBle#Radio$2:  REMOVED RxBleRadioOperationScanApi21(261606157)
	06-21 16:10:16.696 18175-18732/com.hatchbaby.rest.qa W/NightlightManager: [RxIoScheduler-31] Connection process failed! java.util.concurrent.TimeoutException: Timeout connecting to nightlight (15 SECONDS)
	06-21 16:10:16.708 18175-18732/com.hatchbaby.rest.qa I/Nightlight: [RxIoScheduler-31] Disconnecting: Nightlight.disconnect()
	06-21 16:10:16.712 18175-18732/com.hatchbaby.rest.qa D/RxBle#Radio:   QUEUED RxBleRadioOperationDisconnect(128512662)
	06-21 16:10:16.714 18175-18732/com.hatchbaby.rest.qa D/RxBle#Radio$2:  REMOVED RxBleRadioOperationConnect(218129354)
	06-21 16:10:16.719 18175-18732/com.hatchbaby.rest.qa I/Nightlight: [RxIoScheduler-31] Nightlight state: RxBleConnectionState{DISCONNECTED}

#### Actual result
RxBle#Radio is broken by the suddenly lost connection

#### Expected result
Hopefully we can recover from the broken connection, either automatically or through an API.

dariuszseweryn, 2017-06-22:
What library version do you use?

kenwdelong, 2017-06-22:
Sorry, version 1.3.1

kenwdelong, 2017-06-22:
No, we were on 1.3.0.  I will upgrade immediately!

kenwdelong, 2017-06-22:
Also, we tried a workaround last night that worked:  we wrote the DFU Control Characteristic by using a RxBleRadioOperationCustom implementation like this:

    public Observable<Void> asObservable(BluetoothGatt bluetoothGatt, RxBleGattCallback rxBleGattCallback, Scheduler scheduler) throws Throwable {

        BluetoothGattService dfuControlSvc = bluetoothGatt.getService(UUID.fromString(Nightlight.Service.DFU_SERVICE));
        if(dfuControlSvc != null) {
            BluetoothGattCharacteristic dfuControlChar = dfuControlSvc.getCharacteristic(UUID.fromString(Nightlight.Characteristic.DFU_CONTROL));
            if(dfuControlChar != null) {
                dfuControlChar.setValue(new byte[] {0x01, 0x04});
                boolean status = bluetoothGatt.writeCharacteristic(dfuControlChar);
                Timber.i("Wrote bootloader start command: status [%b]", status);
                // No need to wait for callbacks or nothin', this kills the connection immediately
            }
            else {
                Timber.w("No DFU Control Characteristic");
            }
        }
        else {
            Timber.w("No DFU Control Service");
        }
        return Observable.just(null);
    }

Using this method, the connection still dies a sudden death (because the device resets) but the RxBle#Radio queue continues to process requests correctly.

kenwdelong, 2017-06-22:
The bug is not there in 1.3.1, everything continues to work fine!

Sorry for the noise.

closed 2017-06-22 by kenwdelong

dariuszseweryn, 2017-06-22:
No problem. So this was a duplicate of #218 
Good to hear that it is solved.
Best regards