dariuszseweryn/RxAndroidBle #231: `Observable` emitted from `RxBleConnection.setupNotification()`/`RxBleConnection.setupIndication()` does not com

dariuszseweryn, 2017-06-23:
### Summary
Currently when the user will unsubscribe from `.setupNotification()`/`.setupIndication()` the notification/indication is teardown but the (possibly) emitted `Observable<byte[]>` does not complete.
### Library version
`1.3.1`
#### Steps to reproduce actual result
<br />1. Connect to a device
<br />2. Setup notification (or indication)
<br />3. Wait for it being setup
<br />4. Get emitted `Observable<byte[]>`
<br />5. Un-subscribe from the `.setup*()`
<br />6. Observe that emitted `Observable<byte[]>` has not completed
#### Minimum code snippet reproducing the issue
```
rxBleDevice.establishConnection(false)
                .flatMap(connection -> connection.setupNotification(characteristicUuid)
                        .takeUntil(Observable.timer(15, TimeUnit.SECONDS))
                        .flatMap(observable -> observable)
                        .doOnCompleted(() -> Log.d("SHOULD", "COMPLETE"))
                )
                .subscribe();
```
#### Actual result
`SHOULD COMPLETE` is not logged after 15 seconds from the notification setup.
#### Expected result
`SHOULD COMPLETE` should be logged after 15 seconds from the notification setup.

RobLewis, 2017-06-25:
Haven't thought this through, but will this fix require any changes to `ConnectionSharingAdapter`?

dariuszseweryn, 2017-06-26:
@RobLewis No.

closed 2017-06-26 by dariuszseweryn