dariuszseweryn/RxAndroidBle #233: rxBleClient.observeStateChanges() behaviour - not returning the current state

Jogosb, 2017-06-26:
### Summary
When I chain any ble operation with the observeStateChanges observable and the state is not changed the operation is not executed. 

### Library version
`1.3.1`

#### Preconditions
I enter the ScanActivity for example (from the sample), I chain the scanning with the observeStateChanges observable while permissions are granted, bluetooth and location services are on. 

#### Actual result
Scanning doesn't start

#### Expected result
Scanning starts 

Basically when I first saw this observable I thought it would behave like BehaviourSubject and will return the current value immediately after subscription, while it emits values only on changes. 
I assume this is intentional and my question is how You guys solve such issues, when all the requirements are fulfilled and the operation is chained with this observable. Do You use getState() method before performing operations?

dariuszseweryn, 2017-06-26:
From Javadoc:

    /**
     * Returns an observable emitting state _changes_ of the RxBleClient environment which may be helpful in deciding if particular
     * functionality should be used at a given moment.
     *
     * @see #getState() for {@link State} precedence order
     *
     * Examples:
     * - If the device is in {@link State#READY} and the user will turn off the bluetooth adapter then {@link State#BLUETOOTH_NOT_ENABLED}
     * will be emitted.
     * - If the device is in {@link State#BLUETOOTH_NOT_ENABLED} then changing state of Location Services will not cause emissions
     * because of the checks order
     * - If the device is in {@link State#BLUETOOTH_NOT_AVAILABLE} then this {@link Observable} will complete because any other checks
     * will not be performed as devices are not expected to obtain bluetooth capabilities during runtime
     *
     * To get the initial {@link State} and then observe changes you can use: `observeStateChanges().startWith(getState())`.
     *
     * @return the observable
     */
    public abstract Observable<State> observeStateChanges();

TL;DR Yes.

closed 2017-06-26 by dariuszseweryn

Jogosb, 2017-06-26:
I missed that Javadoc, thanks for a quick response!