dariuszseweryn/RxAndroidBle #236: BleGattCallbackTimeoutException / SERVICE_DISCOVERY on Nougat

Ezor, 2017-06-28:
### Summary

Hi,

I'm getting a `BleGattCallbackTimeoutException` exception when trying to setupNotification on my bluetooth device (error log below).

My Configuration is : 
* Sony XPERIA L1
* Android 7.0

The same code works on an Samsung Note 8 with Android 4.4.2.

### Library version
`1.3.2`

#### Minimum code snippet reproducing the issue

Here is the code of my app : 

```java
public void connectToDevice(String macAddress, BluetoothClientDeviceDelegate delegate) {
        RxBleLog.setLogLevel(RxBleLog.VERBOSE);
        this.deviceDelegate = delegate;
        final RxBleDevice device = rxBleClient.getBleDevice(macAddress);
        connectionObservable = device.establishConnection(false)
                .retry(3)
                .takeUntil(disconnectTriggerSubject)
                .compose(new ConnectionSharingAdapter());
        connectionObservable.subscribe(rxBleConnection -> setupNotification(),
                throwable -> {
                    Log.e(TAG, "Connection error", throwable);
                    deviceDelegate.onBluetoothFailed(throwable);
                });
    }

private void setupNotification() {
        currentBuffer = "";
        connectionObservable.flatMap(rxBleConnection -> rxBleConnection.setupIndication(BluetoothClient.POCKET_PRO_READ_UUID, NotificationSetupMode.DEFAULT))
                .doOnNext(notificationObservable -> {
                    Log.d(TAG, "Notification set");
                    setModeData();
                })
                .flatMap(notificationObservable -> notificationObservable)
                .takeUntil(disconnectTriggerSubject)
                .subscribe(
                        bytes -> {
                            String newData = new String(bytes);
                            for(int i=0; i<newData.length(); i++) {
                                String character = newData.substring(i, i+1);
                                if(character.equals("\r")) {
                                    Log.d(TAG, currentBuffer);
                                    handleData(currentBuffer);
                                    currentBuffer = "";
                                } else if(!character.equals("\n")) {
                                    currentBuffer += character;
                                }
                            }
                        },
                        throwable -> {
                            Log.e(TAG, "Notification Error", throwable);
                            deviceDelegate.onBluetoothFailed(throwable);
                        }
                );
    }
```

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`

```
E/BluetoothClient: Notification Error
[... 29 lines cut ...]
06-28 20:25:06.927 17680-17723/mypackage D/BluetoothGatt: onGetService() - Device=EC:FE:7E:15:91:3A UUID=00001800-0000-1000-8000-00805f9b34fb
06-28 20:25:06.927 17680-17691/mypackage D/BluetoothGatt: onGetService() - Device=EC:FE:7E:15:91:3A UUID=00001801-0000-1000-8000-00805f9b34fb
06-28 20:25:06.932 17680-17692/mypackage D/BluetoothGatt: onGetService() - Device=EC:FE:7E:15:91:3A UUID=0000180f-0000-1000-8000-00805f9b34fb
06-28 20:25:06.932 17680-17723/mypackage D/BluetoothGatt: onGetService() - Device=EC:FE:7E:15:91:3A UUID=da2b84f1-6279-48de-bdc0-afbea0226079
06-28 20:25:06.932 17680-17691/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=00002a00-0000-1000-8000-00805f9b34fb
06-28 20:25:06.932 17680-17692/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=00002a01-0000-1000-8000-00805f9b34fb
06-28 20:25:06.932 17680-17723/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=00002a02-0000-1000-8000-00805f9b34fb
06-28 20:25:06.932 17680-17691/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=00002a03-0000-1000-8000-00805f9b34fb
06-28 20:25:06.937 17680-17692/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=00002a04-0000-1000-8000-00805f9b34fb
06-28 20:25:06.937 17680-17723/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=00002a05-0000-1000-8000-00805f9b34fb
06-28 20:25:06.937 17680-17691/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=00002a19-0000-1000-8000-00805f9b34fb
06-28 20:25:06.937 17680-17692/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=99564a02-dc01-4d3c-b04e-3bb1ef0571b2
06-28 20:25:06.937 17680-17723/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=a87988b9-694c-479c-900e-95dfa6c00a24
06-28 20:25:06.937 17680-17691/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=bf03260c-7205-4c25-af43-93b1c299d159
06-28 20:25:06.942 17680-17692/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=18cda784-4bd3-4370-85bb-bfed91ec86af
06-28 20:25:06.942 17680-17723/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=0a1934f5-24b8-4f13-9842-37bb167c6aff
06-28 20:25:06.942 17680-17691/mypackage D/BluetoothGatt: onGetCharacteristic() - Device=EC:FE:7E:15:91:3A UUID=fdd6b4d3-046d-4330-bdec-1fd0c90cb43b
06-28 20:25:06.942 17680-17692/mypackage D/BluetoothGatt: onGetDescriptor() - Device=EC:FE:7E:15:91:3A UUID=00002902-0000-1000-8000-00805f9b34fb
06-28 20:25:06.942 17680-17723/mypackage D/BluetoothGatt: onGetDescriptor() - Device=EC:FE:7E:15:91:3A UUID=00002902-0000-1000-8000-00805f9b34fb
06-28 20:25:06.947 17680-17691/mypackage D/BluetoothGatt: onGetDescriptor() - Device=EC:FE:7E:15:91:3A UUID=00002902-0000-1000-8000-00805f9b34fb
06-28 20:25:06.947 17680-17692/mypackage D/BluetoothGatt: onGetDescriptor() - Device=EC:FE:7E:15:91:3A UUID=00002902-0000-1000-8000-00805f9b34fb
06-28 20:25:06.947 17680-17723/mypackage D/BluetoothGatt: onSearchComplete() = Device=EC:FE:7E:15:91:3A Status=0
06-28 20:25:06.952 17680-17723/mypackage D/RxBle#BluetoothGatt: onServicesDiscovered status=0
06-28 20:25:06.952 17680-17801/mypackage D/BluetoothGatt: setCharacteristicNotification() - uuid: 18cda784-4bd3-4370-85bb-bfed91ec86af enable: true
```

dariuszseweryn, 2017-06-28:
It looks like the service discovery fails on your `Sony XPERIA L1`. You can try the vanilla API but I expect the same result as the library does nothing fancy. It is weird that no other logs are available.

Ezor, 2017-06-28:
It seems you are right there is something wrong with the XPERIA L1...

And for the log it seems i'm not the only one to have this issue : 
https://stackoverflow.com/questions/36554215/lodcat-not-displaying-debug-messages-when-using-specific-device

I close the issue as it's probably only related to my device.

Thanks for your fast feedback anyway.

closed 2017-06-28 by Ezor