dariuszseweryn/RxAndroidBle #242: discoverServices() retrurning null service?

Esperanz0, 2017-07-05:
Hi. I'm facing with discoverServices() error.

BLE logs below. Just tell if You need something wore.. Or what is wrong in this code?

```
final Subscription subscription = bleDevice.establishConnection(false)
                .takeUntil(disconnectTriggerSubject)
                .flatMap(new Func1<RxBleConnection, Observable<RxBleDeviceServices>>() {
                    @Override
                    public Observable<RxBleDeviceServices> call(RxBleConnection rxBleConnection) {
                        Log.d(TAG, "BLE connection to " + bleMac + " done!");*

                    // Add this connection to a map for later use
                    connectionMap.put(bleMac, rxBleConnection);

                    return rxBleConnection.discoverServices();
                }
            })
            .map(new Func1<RxBleDeviceServices, ConnectedDevice>() {
                @Override
                public ConnectedDevice call(RxBleDeviceServices rxBleDeviceServices) {
                    Log.d(TAG, "Got service BLE from " + bleMac);

                    BluetoothGattService ngService = null;
                    for (BluetoothGattService service : rxBleDeviceServices.getBluetoothGattServices()) {
                        Log.e(TAG, "ngServices UUID: " + service.getUuid());
                        Log.e(TAG, "ngServices Instance Id: " + service.getInstanceId());
                        if (NG_SERVICE_UUID.equals(service.getUuid())) {
                            ngService = service;
                            break;
                        }
                    }

                    // If there are no proper service, just throw an error
                    if (ngService == null) {
                        throw new IllegalArgumentException("Connected device not a proper ng device");
                    }
```

BLE LOGS:

```
E/BleManager: isConnected: RxBleConnectionState{DISCONNECTED}
D/BluetoothGatt: connect() - device: E5:07:96:5E:94:CD, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=04b2518b-01e4-4160-9296-461bfea33036
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=14
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=14 device=E5:07:96:5E:94:CD
D/BleManager: BLE connection to E5:07:96:5E:94:CD done!
D/BluetoothGatt: discoverServices() - device: E5:07:96:5E:94:CD
D/BluetoothGatt: onSearchComplete() = Device=E5:07:96:5E:94:CD Status=0
D/BleManager: Got service BLE from E5:07:96:5E:94:CD
E/BleManager: Connected device not a proper ng device
```

dariuszseweryn, 2017-07-05:
Hello @Esperanz0 
First of all could you give info what device do you use, OS version, library version does your peripheral is changing services or you use `BluetoothGatt.refresh()` and how reproducible is this scenario? So far it seems that you are facing a native Android BLE bug.

Esperanz0, 2017-07-05:
Hello @dariuszseweryn.
BUG: Sony Xperia Z5 - Android 7.0 ( Updated to 7.1 and still the same problem)
OK: Sony Xperia Z5 Premium - Android 6.0.1

Library:  'com.polidea.rxandroidble:rxandroidble:1.2.3'

I'm not using BluetoothGatt.refresh().

Reproduction:
Connect Phone with Ble device. Restart both ( app and ble device). Try connect again.
After 15-20s its working again OK.

Looks like phone issue? Not a platform?  Any idea how to handle this?

dariuszseweryn, 2017-07-05:
`0x85` = `133` which is the dreaded `GATT_ERROR`. There is no difference in the discovery mechanism in between OS versions. Does your peripheral use `Service Changed` characteristic?

Esperanz0, 2017-07-07:
To be honest. I'm not responsible for peripheral device and don't know.

dariuszseweryn, 2017-07-07:
But you can check on the device that does not exhibit the but if a `Generic Attribute Service` is discovered.

dariuszseweryn, 2017-07-12:
@Esperanz0 Any luck? Have you tried to postpone service discovery for at least 2 seconds after getting the connection?

Esperanz0, 2017-07-17:
@dariuszseweryn Hey. Sorry have no time for this issue now. For now I'm just trying connect again. But will check it soon.

dariuszseweryn, 2017-09-01:
Closing due to inactivity. Too small amount of info. Feel free to reopen if something new will appear.

closed 2017-09-01 by dariuszseweryn

mahmoud-masmoudi-dev, 2019-03-12:
@dariuszseweryn Speaking of `Generic Attribute Service`, strangely, when I try to discover services, this one is not found on Nexus 5. But it is found on an Android 8. How could it be possible to find some services but not the `Generic Attribute Service` ?

dariuszseweryn, 2019-03-12:
Never seen this behaviour