dariuszseweryn/RxAndroidBle #244: Radio is not released after operation completes and/or is unsubscribed from

prefanatic, 2017-07-06:
I've just recently bumped from 1.1 to 1.3.2 - and have been noticing weird behavior when executing a series of many writeCharacteristic operations.  Every once in a while, the radio interface is not released from one of the operations, causing the queue to hang and no more operations to run.

Bear with me - for I can't fully supply any code to test this, but I do have some cleaned logs that show the queue hanging.
```            RxBle#Radio  D  FINISHED RxBleRadioOperationCharacteristicWrite(145751808)
                         D    QUEUED RxBleRadioOperationCharacteristicWrite(88156208)
                         D   STARTED RxBleRadioOperationCharacteristicWrite(88156208)
    RxBle#BluetoothGatt  D  onCharacteristicChanged characteristic=1cf987ae-7529-4f5c-9af3-f89256ab9370
    RxBle#BluetoothGatt  D  onCharacteristicWrite characteristic=c0555234-90a1-4905-abc4-33462a4ad25f status=0
            RxBle#Radio  D  FINISHED RxBleRadioOperationCharacteristicWrite(88156208)
            RxBle#Radio  D    QUEUED RxBleRadioOperationCharacteristicWrite(26120076)
                         D   STARTED RxBleRadioOperationCharacteristicWrite(26120076)
    RxBle#BluetoothGatt  D  onCharacteristicChanged characteristic=1cf987ae-7529-4f5c-9af3-f89256ab9370
    RxBle#BluetoothGatt  D  onCharacteristicWrite characteristic=c0555234-90a1-4905-abc4-33462a4ad25f status=0
            RxBle#Radio  D  FINISHED RxBleRadioOperationCharacteristicWrite(26120076)
            RxBle#Radio  D    QUEUED RxBleRadioOperationCharacteristicWrite(245035964)
                         D   STARTED RxBleRadioOperationCharacteristicWrite(245035964)
    RxBle#BluetoothGatt  D  onCharacteristicChanged characteristic=1cf987ae-7529-4f5c-9af3-f89256ab9370
    RxBle#BluetoothGatt  D  onCharacteristicWrite characteristic=c0555234-90a1-4905-abc4-33462a4ad25f status=0
=== We'd expect to see a FINISHED here.  Throwing breakpoints in the emitters show the radio is never released after the operation completes. ===
            RxBle#Radio  D    QUEUED RxBleRadioOperationCharacteristicWrite(252352159)

```

I noticed that from within ```RadioReleasingEmitterWrapper``` - when the queue would hang, the previous operation would not release the radio.  The emitter would perform ```onComplete```, and then ```cancel```.  From within ```RxBleRadioOperation``` - the ```.doOnTerminate``` attached in the ```run``` method would not be called.  Changing the ```.doOnTerminate``` to ```.doOnUnsubscribe``` fixes this problem I'm having, but I'm not sure this is the correct solution.

To be quite honest, I'm unsure why I'm unsubscribing from the operation before it completes.  The code used to reproduce this is:
```
Observable.range(0, 200)
                        .concatMap {
                            val control = connection!!.writeCharacteristic(CHARAC_CONTROL_DECIDER, ByteArray(1, { i -> command }))
                            connection!!.writeCharacteristic(CHARAC_DATA_OUT, byteBuffer)
                                    .concatMap {
                                        Observable.combineLatest(dataOutSubject, control) {
                                            o1, o2 ->
                                            o1
                                        }.first()
                                    }
                        }
```
In essence, I'm combining the latest between a writeCharacteristic operation, and a response from an indication from another characteristic, and I'm doing this about 200 times until it hangs.

dariuszseweryn, 2017-07-06:
Hey @prefanatic 
It indeed looks like a bug in the library. But `RadioReleasingEmitterWrapper` was made exactly for this use-case - to always release the radio after completion. 
Could you add a `synchronized` keyword to `onCompleted()` and `onError()` of `RadioReleasingEmitterWrapper`? I think there might be a race condition there.
I am off the office and I cannot perform the tests myself.

prefanatic, 2017-07-06:
@dariuszseweryn 

Unfortunately, adding ```synchronized``` to ```onCompleted()``` and ```onError()``` does not prevent the queue hanging.

dariuszseweryn, 2017-07-06:
Yup it is still possible that the race condition could happen depending on what is the implementation of the `Emitter`. If you could add logs to `onComplete()`, `onError()`, `onCancel()` and paste a log with timestamps displayed to prove if this may be the cause.
It would need two synchronization flags to be sure that the `RadioReleaseInterface.release()` was called properly.
As a side thought I wonder what would happen if you would change this:
```
Observable.combineLatest(dataOutSubject, control) {
  o1, o2 -> o1
}.first()
```
to this:
```
Observable.combineLatest(dataOutSubject.first(), control) {
  o1, o2 -> o1
}
```

dariuszseweryn, 2017-07-07:
@prefanatic Could you test this branch `fix/issue_245_radio_release_race_in_operation` if it would fix your use-case?

prefanatic, 2017-07-07:
@dariuszseweryn 
Looks good - working great now.  Thanks for the fast turn around!

dariuszseweryn, 2017-07-07:
Should be available in `1.3.3-SNAPSHOT` shortly and I will try to release `1.3.3` before the weekend.

closed 2017-07-07 by dariuszseweryn