dariuszseweryn/RxAndroidBle #245: I am not getting any log output in the latest builds.

Emeritus-DarranKelinske, 2017-07-07:
### Disclaimer
Before you will post a new issue to the RxAndroidBle library consider if you are filling a bug/feature request or a general question 
about:

I am following the getting started steps that are part of the readme (i put in a pr to fix a typo in it that was accepted)

I am putting log statements to help follow the connection process and no logging is being generated even though breakpoints are being hit after the log statements.

Below is a snippet of the code:

```
package com.laelaps.rxandroidbletest;

import android.os.Bundle;
import android.support.v7.app.AppCompatActivity;
import android.util.Log;

import com.polidea.rxandroidble.RxBleClient;
import com.polidea.rxandroidble.internal.RxBleLog;
import com.polidea.rxandroidble.scan.ScanFilter;
import com.polidea.rxandroidble.scan.ScanSettings;

import rx.Observable;
import rx.Subscription;
import rx.android.schedulers.AndroidSchedulers;

public class MainActivity extends AppCompatActivity {

    Subscription flowSubscription;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        RxBleClient rxBleClient = RxBleClient.create(this);
        RxBleClient.setLogLevel(RxBleLog.VERBOSE);

        Log.d("DARRAN", "is this thing on?");

        flowSubscription = rxBleClient.observeStateChanges()
                .switchMap(state -> {
                    switch (state) {
                        case READY:
                            Log.d("DARRAN", "Bluetooth is ready");
                            // everything should work
                            return rxBleClient.scanBleDevices(new ScanSettings.Builder()
                                    .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                                    .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                                    .build(), new ScanFilter.Builder()
                                    .build());
                        case BLUETOOTH_NOT_AVAILABLE:
                            Log.d("DARRAN", "Bluetooth not available.");
                            // basically no functionality will work here
                        case LOCATION_PERMISSION_NOT_GRANTED:
                            Log.d("DARRAN", "Location not granted.");
                            // scanning and connecting will not work
                        case BLUETOOTH_NOT_ENABLED:
                            Log.d("DARRAN", "Bluetooth not enabled.");
                            // scanning and connecting will not work
                        case LOCATION_SERVICES_NOT_ENABLED:
                            Log.d("DARRAN", "Location services not enabled.");
                            // scanning will not work
                        default:
                            return Observable.empty();
                    }
                })
                .observeOn(AndroidSchedulers.mainThread())
                .subscribe(
                        rxBleScanResult -> {
                            // Process scan result here.
                            Log.d("DARRAN", "Scan result is: " +rxBleScanResult.toString());
                        },
                        throwable -> {
                            // Handle an error here.
                            Log.d("DARRAN", "Scan result is: " +throwable.toString());
                        }
                );
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        flowSubscription.unsubscribe();
    }
}

```

Thank you!

dariuszseweryn, 2017-07-07:
Hello @Emeritus-DarranKelinske 
Correct me if I am wrong. Your problem is that the logs are not showing in the `logcat` but if you put a breakpoint on them it fires properly?

Emeritus-DarranKelinske, 2017-07-07:
yes sir

dariuszseweryn, 2017-07-07:
Then it seems that you have some problem with your `Android Studio`. Have you tried searching `Stackoverflow`? For instance [this](https://stackoverflow.com/questions/28434901/why-are-log-d-and-log-v-not-printing) may be helpful.

Emeritus-DarranKelinske, 2017-07-07:
Will do. Does logging work for you on the latest builds?

dariuszseweryn, 2017-07-07:
I did not encountered any problems with logging.

dariuszseweryn, 2017-07-07:
As this is not an issue with the library - closing.

closed 2017-07-07 by dariuszseweryn