dariuszseweryn/RxAndroidBle #247: Can't find printer device.

ramanbranavitski, 2017-07-12:
Hi there. I'am trying to find printer using your library and can't do it. I use following `scanSettings` and `scanFilter`. As a result I receive list of nearby mobiles and tablets, but printer isn't visible. How can I tackle my problem? Thanks.

`var scanSettings = ScanSettings.Builder()
                .setScanMode(ScanSettings.SCAN_MODE_BALANCED)
                .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                .build()
var scanFilter = ScanFilter.Builder()
                .build()`

dariuszseweryn, 2017-07-12:
Hello @ramanbranavitski 
As far as I understand you are not reporting an issue with the library but you ask a question about the usage of scanning and filtering of Bluetooth Low Energy devices. 
Those questions and the knowledge held in the answers is easier to be found by other users on www.stackoverflow.com with a tag `rxandroidble`. Please post a question with appropriate tags there and someone will try to help you.
Best Regards

closed 2017-07-12 by dariuszseweryn

dariuszseweryn, 2017-07-12:
[Reference](https://stackoverflow.com/questions/45057047/rxandroidble-work-with-not-low-energy-devices).