dariuszseweryn/RxAndroidBle #250: Two rxBleClient instances to handle two devices simultaneously

Jogosb, 2017-07-14:
### Summary
It’s not exactly an issue, more like a architecture-ish question. 
Basically in app, we perform some kind of a real time control on two ble devices simultaneously.
Ble operations are performed very fast so as long there are no errors everything works smoothly. 
The problem appears when one of the two devices vanish, and operation hangs. It’s waiting for a timeout, and all the operations from the 
second device are waiting in the queue. 
We were able to overcome this issue by adding second rxBleClient so every client handles communication with single device. 
However, in  all Your docs You strongly emphsise that there should only be one rxBleClient instance in app. 
Basically the question is: 
is there any way to handle two devices using one rxBleClient without problem I described before? 
If not, are there any crucial downsides of using two rxBleClients?

Thanks in advance for any feedback!

dariuszseweryn, 2017-07-17:
Hello @Jogosb 
The single `RxBleClient` instance is recommended because it is serializing all bluetooth actions to make the experience more stable. It is possible that the library serializes too much when it comes to multiple connections but it was not tested and the experience may vary between implementations of the BLE stack from different phone vendors. This use-case is not defined / documented in any way.
> is there any way to handle two devices using one rxBleClient without problem I described before?

Unfortunately no — currently it is not possible
> If not, are there any crucial downsides of using two rxBleClients?

It is not tested in any way and you may experience weird behaviour / bugs.

For sure actions on a single `BluetoothGatt` are needed to be synchronized or the next action will not start (i.e. `BluetoothGatt.readCharacteristic(BluetoothGattCharacteristic)` will return `false` unless the previous action will finish with a corresponding `BluetoothGattCallback`). If you could if actions on two or more `BluetoothGatt`s have to be serialized in the same way as on a single `BluetoothGatt` if would be great and could make the library better.

Best Regards

hoanglm4, 2017-07-26:
I using 4 RxBleClient for connect 4 device in the same time. Currently, I do see anything problem.

dariuszseweryn, 2017-08-30:
I have just merged a pull request that should make additional `RxBleClient`'s obsolete — now the same behaviour should be possible to get with only just one. Handling of multiple connections should now be better — stalling of one connection should not interact with others.

The change should be available in `1.4.0-SNAPSHOT` shortly.

Best Regards

closed 2017-08-30 by dariuszseweryn

hoanglm4, 2017-09-01:
Dear @dariuszseweryn ,
When is 1.4.0 released ?

Thanks

dariuszseweryn, 2017-09-01:
There are some minor things to finish. Maybe I will be able to release today. If not then early next week. Still waiting to clarify #275 :)