dariuszseweryn/RxAndroidBle #252: Turn Bluetooth ON and OFF

vpratfr, 2017-07-20:
### Summary

I did not see any documentation to turn on/off the bluetooth connectivity on the device.

Provided we have the Bluetooth Admin permission, that should be possible (ex. via the BluetoothAdapter.enable methods)

Or is there a method I did not see somewhere?

### Library version
`1.3.3`

dariuszseweryn, 2017-07-20:
Hello @vpratfr 
This is intended. As you can see in the `Javadoc` of `BluetoothAdapter.enable()`:
```
Turn on the local Bluetooth adapter—do not use without explicit user action to turn on Bluetooth.
```
Having a wrapper for `BluetoothAdapter.enable()` and `BluetoothAdapter.disable()` could encourage users to play with the adapter state by themselves whereas it is more polite to ask the application user to change the state of the adapter via an intent:
```
Intent enableBtIntent = new Intent(BluetoothAdapter.ACTION_REQUEST_ENABLE);
int REQUEST_ENABLE_BT = 1;
context.startActivityForResult(enableBtIntent, REQUEST_ENABLE_BT);
```
Alternatively if the library user does want to violate the application user's right to decide about the state of their phone it is possible to call:
```
BluetoothAdapter.getDefaultAdapter().enable();
```
Best Regards

vpratfr, 2017-07-20:
Hi,

Yes, in our application this is acceptable and I am calling `BluetoothAdapter.getDefaultAdapter().enable()` after getting user agreement from a dialog (avoids going to settings & all)

I would have expected a method which avoids using `BluetoothAdapter.getDefaultAdapter()` but using your adapter wrapper instead. That seemed logical to me.

I however fully agree with you that some warning should be shown in that method documentation, just like in the javadoc of `BluetoothAdapter.enable()`

Rioner123, 2017-07-24:
Related to this... i am having issues where the bluetooth disconnects and will not reconnect. I've tried manually turning the adapter on and off, but it does not work. The only thing working is if I do a hard reset on the device. Unfortunately I have not been able to reproduce this myself, but many of my users are having this problem. Any solution or advice?

dariuszseweryn, 2017-07-25:
@Rioner123 Unfortunately I have no idea what issue you are facing. I would need more information and I do not actually get what is the relation to the original post?

dariuszseweryn, 2017-08-03:
With @uKL we have agreed that managing of the Bluetooth Adapter state is out of scope of this library. I have added a few words about turning the adapter on/off in [Readme](https://github.com/Polidea/RxAndroidBle/blob/master/README.md).
Best Regards

closed 2017-08-03 by dariuszseweryn