dariuszseweryn/RxAndroidBle #254: SCAN_FAILED_INTERNAL_ERROR on API21

dagi12, 2017-07-23:
### Summary
Library throw "SCAN_FAILED_INTERNAL_ERROR" error on scan on phones with Android 5.0 (API21). After updating Android 6.0 problem gone.

### Preconditions
Checked for 2 devices namely: Sony Xperia E3 with Android 5.0 and Sony XPERIA E3 with Android 5.0.  I do not know if other models experiencing error as well.

### Library version
`1.3.3` but experiencing the same issue on older releases as well.

#### Steps to reproduce actual result
Just clone RxAndroidBle sample and toch "START SCAN"

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
```
07-23 16:40:54.077 16409-16409/com.polidea.rxandroidble.sample D/RxBle#Radio:   QUEUED RxBleRadioOperationScanApi21(780412642)
07-23 16:40:54.077 16409-17838/com.polidea.rxandroidble.sample D/RxBle#Radio:  STARTED RxBleRadioOperationScanApi21(780412642)
07-23 16:40:54.093 16409-17838/com.polidea.rxandroidble.sample D/RxBle#Radio: FINISHED RxBleRadioOperationScanApi21(780412642)
07-23 16:40:54.099 16409-16409/com.polidea.rxandroidble.sample D/BluetoothLeScanner: could not find callback wrapper
07-23 16:40:54.103 16409-16409/com.polidea.rxandroidble.sample E/THIS: tal
                                                                       BleScanException{reason=SCAN_FAILED_INTERNAL_ERROR}
                                                                           at com.polidea.rxandroidble.internal.operations.RxBleRadioOperationScanApi21$1.onScanFailed(RxBleRadioOperationScanApi21.java:76)
                                                                           at android.bluetooth.le.BluetoothLeScanner$1.run(BluetoothLeScanner.java:379)
                                                                           at android.os.Handler.handleCallback(Handler.java:739)
                                                                           at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                           at android.os.Looper.loop(Looper.java:135)
                                                                           at android.app.ActivityThread.main(ActivityThread.java:5254)
                                                                           at java.lang.reflect.Method.invoke(Native Method)
                                                                           at java.lang.reflect.Method.invoke(Method.java:372)
                                                                           at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:898)
                                                                           at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:693)

```

dariuszseweryn, 2017-07-24:
Hello @dagi12 
Why do you think it is an issue with the library and not native Android BLE stack? The library just starts a scan and translates the result from the OS which in this case was `SCAN_FAILED_INTERNAL_ERROR`.
Could you paste the exact code that you are using?

I do not have access to the said phone with 5.0 so I cannot test it myself. It is possible that the other  applications use either the legacy API. Which other applications have you checked against?

dariuszseweryn, 2017-08-01:
Closing due to inactivity. Feel free to add more info when possible

closed 2017-08-01 by dariuszseweryn