dariuszseweryn/RxAndroidBle #256: MissingBackpressureException even .onBackpressureDrop() is added?

Esperanz0, 2017-07-28:
Hey. I'm using RxAndroidBle for connection / writing data etc stuffs. 
And with long stream operation like subscribe to resource ( 1-3min) I got MissingBackpressureException ( bottom of source code with Log.e).
Any idea what is wrong here?

 // Setup notifications for incoming data
                            RxBleConnection connection = connectionMap.get(bleMac);
                            Subscription notifySubscription = connection.setupNotification(notifyCharacteristic)
                                    .doOnNext(new Action1<Observable<byte[]>>() {
                                        @Override
                                        public void call(Observable<byte[]> observable) {
                                            Log.d(TAG, "Notifications set, calling bypassConnect()");

                                            // Bypass connect in WB to make it aware of this new device
                                            String wbAddress = addressMap.getOrCreateWbAddress(bleMac);
                                            bleWrapper.bypassConnect(wbAddress);
                                        }
                                    })
                                    .flatMap(new Func1<Observable<byte[]>, Observable<byte[]>>() {
                                        @Override
                                        public Observable<byte[]> call(Observable<byte[]> observable) {
                                            return observable;
                                        }
                                    })
                                    .onBackpressureDrop()
                                    .subscribe(new Action1<byte[]>() {
                                        @Override
                                        public void call(byte[] bytes) {
                                            dataAvailable(bleMac, bytes);
                                        }
                                    }, new Action1<Throwable>() {
                                        @Override
                                        public void call(Throwable throwable) {
                                            Log.e(TAG, "dataAvailable() Error: " + throwable);
                                        }
                                    });

dariuszseweryn, 2017-07-28:
Hello @Esperanz0 
Unfortunately I do not see anything at first glance and I am a little short on time to debug this code. Looks weird that even with `.onBackpressureDrop()` it still have this exception.
You can try to print the whole stacktrace and check it yourself.

Esperanz0, 2017-07-28:
Hey @dariuszseweryn . 

By print whole stacktrace You mean this? ->   RxJavaHooks.enableAssemblyTracking();
Or logs from RxAndroidBle?

Esperanz0, 2017-07-28:
More logs :

```
D/BleManager: Received data: 
E/MdsHeader: headerObj: {"Content-Type":"application\/json","Status":0,"Reason":"CUSTOM_STATUS","Uri":"MDS\/EventListener\/20","Content-Length":421}
D/Mds:  Mds SDSInternalCallback()  taskId:20 sdsCallType:2 header:SdsHeader{status=0, uri='MDS/EventListener/20', reason='CUSTOM_STATUS', contentLength=421, contentType='null', location='', taskId=0} dataBody:{"Body": {"Timestamp": 242559, "ArrayAcc": [{"x": -0.21315674483776093, "y": -0.30656251311302185, "z": 10.027947425842285}, {"x": -0.25387206673622131, "y": -0.23231691122055054, "z": 9.9441213607788086}, {"x": -0.24189698696136475, "y": -0.28740236163139343, "z": 10.027947425842285}, {"x": -0.18681153655052185, "y": -0.27782228589057922, "z": 10.030342102050781}]}, "Uri": "ECKI89CB9A98/Meas/Acc/52", "Method": "PUT"}
D/LinearAccelerationTestActivity: onSuccess(): {"Body": {"Timestamp": 242559, "ArrayAcc": [{"x": -0.21315674483776093, "y": -0.30656251311302185, "z": 10.027947425842285}, {"x": -0.25387206673622131, "y": -0.23231691122055054, "z": 9.9441213607788086}, {"x": -0.24189698696136475, "y": -0.28740236163139343, "z": 10.027947425842285}, {"x": -0.18681153655052185, "y": -0.27782228589057922, "z": 10.030342102050781}]}, "Uri": "ECKI89CB9A98/Meas/Acc/52", "Method": "PUT"}
I/art: Background partial concurrent mark sweep GC freed 295544(11MB) AllocSpace objects, 0(0B) LOS objects, 40% free, 18MB/31MB, paused 6.197ms total 93.206ms
D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=34802252-7185-4d5d-b431-630e7050e8f0
D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=34802252-7185-4d5d-b431-630e7050e8f0
D/BleManager: Received data: 
E/MdsHeader: headerObj: {"Content-Type":"application\/json","Status":0,"Reason":"CUSTOM_STATUS","Uri":"MDS\/EventListener\/20","Content-Length":421}
D/Mds:  Mds SDSInternalCallback()  taskId:20 sdsCallType:2 header:SdsHeader{status=0, uri='MDS/EventListener/20', reason='CUSTOM_STATUS', contentLength=421, contentType='null', location='', taskId=0} dataBody:{"Body": {"Timestamp": 242636, "ArrayAcc": [{"x": -0.21076172590255737, "y": -0.29219239950180054, "z": 10.008787155151367}, {"x": -0.19878663122653961, "y": -0.34248781204223633, "z": 10.147698402404785}, {"x": -0.19878663122653961, "y": -0.28021728992462158, "z": 10.037527084350586}, {"x": -0.17723144590854645, "y": -0.27303224802017212, "z": 10.054292678833008}]}, "Uri": "ECKI89CB9A98/Meas/Acc/52", "Method": "PUT"}
D/LinearAccelerationTestActivity: onSuccess(): {"Body": {"Timestamp": 242636, "ArrayAcc": [{"x": -0.21076172590255737, "y": -0.29219239950180054, "z": 10.008787155151367}, {"x": -0.19878663122653961, "y": -0.34248781204223633, "z": 10.147698402404785}, {"x": -0.19878663122653961, "y": -0.28021728992462158, "z": 10.037527084350586}, {"x": -0.17723144590854645, "y": -0.27303224802017212, "z": 10.054292678833008}]}, "Uri": "ECKI89CB9A98/Meas/Acc/52", "Method": "PUT"}
D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=34802252-7185-4d5d-b431-630e7050e8f0
D/BleManager: Received data: 
E/MdsHeader: headerObj: {"Content-Type":"application\/json","Status":0,"Reason":"CUSTOM_STATUS","Uri":"MDS\/EventListener\/20","Content-Length":419}
D/Mds:  Mds SDSInternalCallback()  taskId:20 sdsCallType:2 header:SdsHeader{status=0, uri='MDS/EventListener/20', reason='CUSTOM_STATUS', contentLength=419, contentType='null', location='', taskId=0} dataBody:{"Body": {"Timestamp": 242713, "ArrayAcc": [{"x": -0.25626710057258606, "y": -0.25147706270217896, "z": 10.090217590332031}, {"x": -0.20597168803215027, "y": -0.25866213440895081, "z": 10.037527084350586}, {"x": -0.21076172590255737, "y": -0.26105713844299316, "z": 10.03513240814209}, {"x": -0.22034180164337158, "y": -0.26824221014976501, "z": 10.03513240814209}]}, "Uri": "ECKI89CB9A98/Meas/Acc/52", "Method": "PUT"}
D/LinearAccelerationTestActivity: onSuccess(): {"Body": {"Timestamp": 242713, "ArrayAcc": [{"x": -0.25626710057258606, "y": -0.25147706270217896, "z": 10.090217590332031}, {"x": -0.20597168803215027, "y": -0.25866213440895081, "z": 10.037527084350586}, {"x": -0.21076172590255737, "y": -0.26105713844299316, "z": 10.03513240814209}, {"x": -0.22034180164337158, "y": -0.26824221014976501, "z": 10.03513240814209}]}, "Uri": "ECKI89CB9A98/Meas/Acc/52", "Method": "PUT"}
D/RxBle#BluetoothGatt: onCharacteristicChanged characteristic=34802252-7185-4d5d-b431-630e7050e8f0
D/BleManager: Received data: 
E/MdsHeader: headerObj: {"Content-Type":"application\/json","Status":0,"Reason":"CUSTOM_STATUS","Uri":"MDS\/EventListener\/20","Content-Length":419}
D/Mds:  Mds SDSInternalCallback()  taskId:20 sdsCallType:2 header:SdsHeader{status=0, uri='MDS/EventListener/20', reason='CUSTOM_STATUS', contentLength=419, contentType='null', location='', taskId=0} dataBody:{"Body": {"Timestamp": 242790, "ArrayAcc": [{"x": -0.24668702483177185, "y": -0.31853762269020081, "z": 10.003996849060059}, {"x": -0.22992187738418579, "y": -0.30656251311302185, "z": 9.99920654296875}, {"x": -0.21794678270816803, "y": -0.31853762269020081, "z": 10.063872337341309}, {"x": -0.19878663122653961, "y": -0.31614258885383606, "z": 10.075847625732422}]}, "Uri": "ECKI89CB9A98/Meas/Acc/52", "Method": "PUT"}
D/LinearAccelerationTestActivity: onSuccess(): {"Body": {"Timestamp": 242790, "ArrayAcc": [{"x": -0.24668702483177185, "y": -0.31853762269020081, "z": 10.003996849060059}, {"x": -0.22992187738418579, "y": -0.30656251311302185, "z": 9.99920654296875}, {"x": -0.21794678270816803, "y": -0 [... cut ...] "y": -0.32572266459465027, "z": 10.039922714233398}, {"x": -0.16765137016773224, "y": -0.33051270246505737, "z": 10.011181831359863}]}, "Uri": "ECKI89CB9A98/Meas/Acc/52", "Method": "PUT"}
D/RxBle#Radio:   QUEUED RxBleRadioOperationDescriptorWrite(168985183)
D/RxBle#Radio:  STARTED RxBleRadioOperationDescriptorWrite(168985183)
D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
D/RxBle#Radio: FINISHED RxBleRadioOperationDescriptorWrite(168985183)

```

dariuszseweryn, 2017-07-28:
Yeah, assembly tracking could be helpful. Plus the full stacktrace of the exception.

Esperanz0, 2017-07-31:
```

rx.exceptions.MissingBackpressureException
at rx.internal.util.RxRingBuffer.onNext(RxRingBuffer.java:325)
at rx.internal.operators.OperatorMerge$MergeSubscriber.queueScalar(OperatorMerge.java:379)
at rx.internal.operators.OperatorMerge$MergeSubscriber.tryEmit(OperatorMerge.java:361)
at rx.internal.operators.OperatorMerge$InnerSubscriber.onNext(OperatorMerge.java:846)
at rx.subjects.PublishSubject$PublishSubjectProducer.onNext(PublishSubject.java:304)
at rx.subjects.PublishSubject$PublishSubjectState.onNext(PublishSubject.java:219)
at rx.subjects.PublishSubject.onNext(PublishSubject.java:72)
at rx.observers.SerializedObserver.onNext(SerializedObserver.java:91)
at rx.subjects.SerializedSubject.onNext(SerializedSubject.java:67)
at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$3.onCharacteristicChanged(RxBleGattCallback.java:130)
at android.bluetooth.BluetoothGatt$1.onNotify(BluetoothGatt.java:438)
at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:399)
at android.os.Binder.execTransact(Binder.java:453)
Caused by: rx.exceptions.AssemblyStackTraceException: Assembly trace:
at rx.Observable.unsafeCreate(Observable.java:162)
at rx.Observable.lift(Observable.java:299)
at rx.Observable.merge(Observable.java:2572)
at rx.Observable.merge(Observable.java:2914)
at rx.Observable.merge(Observable.java:2667)
at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.withHandlingStatusErrorAndDisconnection(RxBleGattCallback.java:247)
at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.getOnCharacteristicChanged(RxBleGattCallback.java:301)
at com.polidea.rxandroidble.internal.connection.NotificationAndIndicationManager.observeOnCharacteristicChangeCallbacks(NotificationAndIndicationManager.java:181)
at com.polidea.rxandroidble.internal.connection.NotificationAndIndicationManager.access$400(NotificationAndIndicationManager.java:30)
at com.polidea.rxandroidble.internal.connection.NotificationAndIndicationManager$1$1.call(NotificationAndIndicationManager.java:107)
at com.polidea.rxandroidble.internal.connection.NotificationAndIndicationManager$1$1.call(NotificationAndIndicationManager.java:104)
at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:69)
at rx.observers.Subscribers$5.onNext(Subscribers.java:235)
at rx.internal.operators.OperatorMerge$MergeSubscriber.emitScalar(OperatorMerge.java:395)
at rx.internal.operators.OperatorMerge$MergeSubscriber.tryEmit(OperatorMerge.java:355)
at rx.internal.operators.OperatorMerge$InnerSubscriber.onNext(OperatorMerge.java:846)
at rx.observers.SerializedObserver.onNext(SerializedObserver.java:91)
at rx.observers.SerializedSubscriber.onNext(SerializedSubscriber.java:94)
at rx.internal.operators.OnSubscribeConcatMap$ConcatMapSubscriber.innerNext(OnSubscribeConcatMap.java:182)
at rx.internal.operators.OnSubscribeConcatMap$ConcatMapInnerScalarProducer.request(OnSubscribeConcatMap.java:366)
at rx.internal.producers.ProducerArbiter.setProducer(ProducerArbiter.java:126)
at rx.internal.operators.OnSubscribeConcatMap$ConcatMapSubscriber.drain(OnSubscribeConcatMap.java:278)
at rx.internal.operators.OnSubscribeConcatMap$ConcatMapSubscriber.onNext(OnSubscribeConcatMap.java:144)
at rx.internal.operators.OnSubscribeFromArray$FromArrayProducer.slowPath(OnSubscribeFromArray.java:100)
at rx.internal.operators.OnSubscribeFromArray$FromArrayProducer.request(OnSubscribeFromArray.java:63)
at rx.Subscriber.setProducer(Subscriber.java:211)
at rx.Subscriber.setProducer(Subscriber.java:205)
at rx.internal.operators.OnSubscribeFromArray.call(OnSubscribeFromArray.java:32)
at rx.internal.operators.OnSubscribeFromArray.call(OnSubscribeFromArray.java:24)
at rx.Observable.unsafeSubscribe(Observable.java:10256)
at rx.internal.operators.OnSubscribeConcatMap.call(OnSubscribeConcatMap.java:94)
at rx.internal.operators.OnSubscribeConcatMap.call(OnSubscribeConcatMap.java:42)
at rx.Observable.unsafeSubscribe(Observable.java:10256)
at rx.

```

Esperanz0, 2017-07-31:
@dariuszseweryn I am not sure I can post SO link here. 
But meybe it will also help You. 

https://stackoverflow.com/questions/45409986/missingbackpressureexception-even-onbackpressuredrop-is-added

dariuszseweryn, 2017-07-31:
Yeah, I am monitoring stackoverflow for questions related to the library. :) TIL a new Thing™. I would add it by myself if you would not link it.
Anyway — I am closing this issue.
I do not think that the library should handle back-pressure by itself as there are many possible ways of handling it (and each use-case could use a different way of handling) and it would need another knob in the interface. Maybe with the new API #239 I will add a back-pressure handling switch.

TeemuStenhammar, 2017-07-31:
@dariuszseweryn We are experiencing this same issue with RxAndroidBle. No matter what we do, we cannot have backpressure handled in such a way that it doesn't break after couple minutes. So, I do agree that handling of backpressure should not be done by the library, but to me it looks like the library is doing some internal subscription that I have no access to. This issue is beginning to be a blocker for us.

We have tried two approaches to circumvent the backpressure:

```
    Subscription notifySubscription = connection.setupNotification(notifyCharacteristic)
        .flatMap(new Func1<Observable<byte[]>, Observable<byte[]>>() {
            @Override
            public Observable<byte[]> call(final Observable<byte[]> observable) {
                observable.onBackpressureDrop();
            }
        })
        .subscribe(new Action1<byte[]>() {
            @Override
            public void call(byte[] bytes) {
                dataAvailable(bytes);
            }
        }, new ThrowableLoggingAction(TAG, "Error while receiving data"));
```
And:
```
Subscription notifySubscription = connection.setupNotification(notifyCharacteristic)
    .flatMap(new Func1<Observable<byte[]>, Observable<byte[]>>() {
        @Override
        public Observable<byte[]> call(final Observable<byte[]> observable) {
            return Observable.create(new Action1<Emitter<byte[]>>() {
                @Override
                public void call(final Emitter<byte[]> emitter) {
                    Subscription s = observable
                            .subscribe(new Action1<byte[]>() {
                                @Override
                                public void call(byte[] bytes) {
                                    emitter.onNext(bytes);
                                }
                            }, new Action1<Throwable>() {
                                @Override
                                public void call(Throwable throwable) {
                                    Log.e(TAG, "Emitter failure: ", throwable);
                                }
                            });

                    emitter.setSubscription(s);
                }
            }, Emitter.BackpressureMode.LATEST);
        }
    })
    .subscribe(new Action1<byte[]>() {
        @Override
        public void call(byte[] bytes) {
            dataAvailable(bytes);
        }
    }, new ThrowableLoggingAction(TAG, "Error while receiving data"));
```

Neither approach works so any help with circumventing this issue would be welcome.

TeemuStenhammar, 2017-07-31:
@dariuszseweryn We'll get those logs to as fast as possible, but that will most likely take until tomorrow.

RobLewis, 2017-07-31:
FWIW, I'm using the following without seeing any exceptions: 
```
frameRelayerSubscription = frameRelayer
                .observeOn( AndroidSchedulers.mainThread() ) 
                .onBackpressureLatest()
                .subscribe( frameRelayerSubscriber );  
```
(`frameRelayer` is a `BehaviorSubject`)

TeemuStenhammar, 2017-08-01:
@dariuszseweryn Here is the stacktrace. It came from MissingBackpressureException generated in the first case I described:

```
rx.exceptions.MissingBackpressureException
    at rx.internal.util.RxRingBuffer.onNext(RxRingBuffer.java:325)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.queueScalar(OperatorMerge.java:379)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.tryEmit(OperatorMerge.java:361)
    at rx.internal.operators.OperatorMerge$InnerSubscriber.onNext(OperatorMerge.java:846)
    at com.jakewharton.rxrelay.RelaySubscriptionManager$RelayObserver.onNext(RelaySubscriptionManager.java:205)
    at com.jakewharton.rxrelay.PublishRelay.call(PublishRelay.java:47)
    at com.jakewharton.rxrelay.SerializedAction1.call(SerializedAction1.java:84)
    at com.jakewharton.rxrelay.SerializedRelay.call(SerializedRelay.java:20)
    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$4.onCharacteristicChanged(RxBleGattCallback.java:139)
    at android.bluetooth.BluetoothGatt$1.onNotify(BluetoothGatt.java:438)
    at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:399)
    at android.os.Binder.execTransact(Binder.java:453)
Caused by: rx.exceptions.AssemblyStackTraceException: Assembly trace:
    at rx.Observable.unsafeCreate(Observable.java:162)
    at rx.Observable.lift(Observable.java:299)
    at rx.Observable.merge(Observable.java:2572)
    at rx.Observable.merge(Observable.java:2914)
    at rx.Observable.merge(Observable.java:2637)
    at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.getOnCharacteristicChanged(RxBleGattCallback.java:311)
    at com.polidea.rxandroidble.internal.connection.NotificationAndIndicationManager.observeOnCharacteristicChangeCallbacks(NotificationAndIndicationManager.java:18
    at com.polidea.rxandroidble.internal.connection.NotificationAndIndicationManager.access$400(NotificationAndIndicationManager.java:31)
    at com.polidea.rxandroidble.internal.connection.NotificationAndIndicationManager$1$1.call(NotificationAndIndicationManager.java:110)
    at com.polidea.rxandroidble.internal.connection.NotificationAndIndicationManager$1$1.call(NotificationAndIndicationManager.java:107)
    at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:69)
    at rx.observers.Subscribers$5.onNext(Subscribers.java:235)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.emitScalar(OperatorMerge.java:395)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.tryEmit(OperatorMerge.java:355)
    at rx.internal.operators.OperatorMerge$InnerSubscriber.onNext(OperatorMerge.java:846)
    at rx.observers.SerializedObserver.onNext(SerializedObserver.java:91)
    at rx.observers.SerializedSubscriber.onNext(SerializedSubscriber.java:94)
    at rx.internal.operators.OnSubscribeConcatMap$ConcatMapSubscriber.innerNext(OnSubscribeConcatMap.java:182)
    at rx.internal.operators.OnSubscribeConcatMap$ConcatMapInnerScalarProducer.request(OnSubscribeConcatMap.java:366)
    at rx.internal.producers.ProducerArbiter.setProducer(ProducerArbiter.java:126)
    at rx.internal.operators.OnSubscribeConcatMap$ConcatMapSubscriber.drain(OnSubscribeConcatMap.java:278)
    at rx.internal.operators.OnSubscribeConcatMap$ConcatMapSubscriber.onNext(OnSubscribeConcatMap.java:144)
    at rx.internal.operators.OnSubscribeFromArray$FromArrayProducer.slowPath(OnSubscribeFromArray.java:100)
    at rx.internal.operators.OnSubscribeFromArray$FromArrayProducer.request(OnSubscribeFromArray.java:63)
    at rx.Subscriber.setProducer(Subscriber.java:211)
    at rx.Subscriber.setProducer(Subscriber.java:205)
    at rx.internal.operators.OnSubscribeFromArray.call(OnSubscribeFromArray.java:32)
    at rx.internal.operators.OnSubscribeFromArray.call(OnSubscribeFromArray.java:24)
    at rx.Observable.unsafeSubscribe(Observable.java:10256)
    at rx.internal.operators.OnSubscribeConcatMap.call(OnSubscribeConcatMap.java:94)
    at rx.internal.operators.OnSubscribeConcatMap.call(OnSubscribeConcatMap.java:42)
    at rx.Observable.unsafeSubscribe(Observable.java:10256)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.onNext(OperatorMerge.java:248)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.onNext(OperatorMerge.java:148)
    at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:77)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.emitScalar(OperatorMerge.java:395)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.tryEmit(OperatorMerge.java:355)
    at rx.internal.operators.OperatorMerge$InnerSubscriber.onNext(OperatorMerge.java:846)
    at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:77)
    at rx.internal.operators.OnSubscribeSwitchIfEmpty$ParentSubscriber.onNext(OnSubscribeSwitchIfEmpty.java:121)
    at rx.internal.operators.OperatorOnErrorResumeNextViaFunction$4.onNext(OperatorOnErrorResumeNextViaFunction.java:154)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.emitScalar(OperatorMerge.java:395)
    at rx.internal.operators.OperatorMerge$MergeSubscriber.tryEmit(OperatorMerge.java:355)
    at rx.internal.operators.OperatorMerge$InnerSubscriber.onNext(OperatorMerge.java:846)
    at rx.internal.operators.OnSubscribeCreate$NoneEmitter.onNext(OnSubscribeCreate.java:173)
    at rx.internal.util.ObserverSubscriber.onNext(ObserverSubscriber.java:34)
    at rx.observers.SafeSubscriber.onNext(SafeSubscriber.java:134)
[... cut ...]
 at com.jakewharton.rxrelay.PublishRelay.call(PublishRelay.java:47)
 at com.jakewharton.rxrelay.SerializedAction1.call(SerializedAction1.java:84)
 at com.jakewharton.rxrelay.SerializedRelay.call(SerializedRelay.java:20)
 at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$4.onCharacteristicChanged(RxBleGattCallback.java:139)
 at android.bluetooth.BluetoothGatt$1.onNotify(BluetoothGatt.java:438)
 at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:399)
 at android.os.Binder.execTransact(Binder.java:453)
```

dariuszseweryn, 2017-09-13:
@TeemuStenhammar Sorry for such a delay (I was on holiday)—could you add on what phone did the problem surfaced? How quick and of what size the notifications are?

dariuszseweryn, 2017-09-20:
Finally I had time to dig into this issue. I have prepared a peripheral that sends out ~400 notifications per second and it worked as long as the consumer was quick:
```
device.establishConnection(false)
        .flatMap(connection -> connection.setupNotification(uuid))
        .flatMap(observable -> observable)
        .subscribe(
                bytes -> Log.i("Received", "."),
                throwable -> Log.e("Error", "Whoops!", throwable)
        );
```
Then I have made the consumer slower by adding a `Thread.sleep(10)` and I quickly got `MissingBackpressureException`. I have then started to play with `.onBackpressureDrop()` and adding it right before `.subscribe()` did not fixed the issue. I have checked the `Javadoc` of `.onBackpressureDrop()` and it says:
```
     * If the downstream request count hits 0 then the Observable will refrain from calling {@code onNext} until
     * the observer invokes {@code request(n)} again to increase the request count.
```
Then I have checked the `Javadoc` of `.subscribe()`:
```
     *  <dd>The operator consumes the source {@code Observable} in an unbounded manner (i.e., no
     *  backpressure is applied to it).</dd>
```
This means that applying `.onBackpressureDrop()` just before `.subscribe()` does nothing as the `.subscribe()` is not applying backpressure to the upstream. `.onBackpressure*()` should be applied before an operator that applies backpressure to the upstream i.e. `.observeOn(Scheduler)` which `Javadoc` says:
```
     *  <dd>This operator honors backpressure from downstream and expects it from the source {@code Observable}. Violating this
     *  expectation will lead to {@code MissingBackpressureException}. This is the most common operator where the exception
     *  pops up; look for sources up the chain that don't support backpressure,
     *  such as {@code interval}, {@code timer}, {code PublishSubject} or {@code BehaviorSubject} and apply any
     *  of the {@code onBackpressureXXX} operators <strong>before</strong> applying {@code observeOn} itself.</dd>
```
i.e. this will work:
```
device.establishConnection(false)
        .flatMap(connection -> connection.setupNotification(uuid))
        .flatMap(observable -> observable)
        .onBackpressureDrop()
        .observeOn(Schedulers.computation())
        .subscribe(
                bytes -> {
                    Log.i("Received", ".");
                    try {
                        Thread.sleep(10);
                    } catch (InterruptedException e) {
                        e.printStackTrace();
                    }
                },
                throwable -> Log.e("Error", "Whoops!", throwable)
        );
```
but this will not:
```
device.establishConnection(false)
        .flatMap(connection -> connection.setupNotification(uuid))
        .flatMap(observable -> observable)
        .observeOn(Schedulers.computation())
        .onBackpressureDrop()
        .subscribe(
                bytes -> {
                    Log.i("Received", ".");
                    try {
                        Thread.sleep(10);
                    } catch (InterruptedException e) {
                        e.printStackTrace();
                    }
                },
                throwable -> Log.e("Error", "Whoops!", throwable)
        );
```
Summary: the behaviour of the library is correct. The aim of `RxAndroidBle` is to provide all BLE events to the user. Applying backpressure mechanisms inside the library would make it harder for the user to understand why they would not get all the notifications the device has sent if they consume them too slow.

closed 2017-09-20 by dariuszseweryn