dariuszseweryn/RxAndroidBle #257: Make scan filters more flexible

vpratfr, 2017-08-01:
Currently, scan filters can, for instance, reject results when name does not match a constant name. Same goes for other attributes.

Would be nice to be able to customize the match logic with a callback / lambda.

For instance we could reject scan results when name does not start with a given string (we are naming our devices XYZ-1234, where 1234 is a user friendly identifier for the device).

Could be an API such as (Kotlin, but would be equivalent in Java with Single Method interfaces):

```
ScanFilter.Builder()
    .matchNameWhen(name -> name.startsWith("XYZ"))
    .matchManufacturerIdWhen(id -> listOf(123, 5345).contains(id))
    .build()  
```

Each lambda passed there taken the input data (name, manufacturer, etc.) and returns a boolean.

Java equivalent could be

```
interface Predicate<T> { 
    boolean matches(T input);
}

ScanFilter.Builder()
    .matchNameWhen(new Predicate<String>() {
        boolean matches(String input) { return input.startsWith("XYZ") }
    })
    .matchManufacturerIdWhen(new Predicate<Integer>() {
        boolean matches(Integer input) { return input==123 || input==5345; }
    })
    .build()  
```

dariuszseweryn, 2017-08-01:
Hello @vpratfr 
The thing is that the built-in filtering may be offloaded to the native part. What you would like to have could not be offloaded and is easy to achieve by utilising the `RxJava` `.filter()` operator.

vpratfr, 2017-08-01:
Thanks for the insight. Indeed using Rx filter is a good choice.

closed 2017-08-01 by vpratfr