dariuszseweryn/RxAndroidBle #261: Decreasing of performance for long background scanning process inside service

lukevr, 2017-08-03:
Seems that when scanBleDevices works for a long time 20-30 min inside background/foreground service the performance(number of frames per second) dramatically goes down.

I'm using **RxAndroidBle 1.3.3** and have:

`
        scanResults = rxBleClient.scanBleDevices(
                new ScanSettings.Builder()
                        .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                        .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                        .build())
                .share();
`

that I share  it then for 2 subscribers :

`
        beaconSubscription = scanResults.
                observeOn(AndroidSchedulers.mainThread()).
                subscribe(rxBleScanResult -> {
                            checkBeaconCandidat(rxBleScanResult);
                        },
                        throwable -> {
                            Log.e(TAG, throwable.toString());
                            throwable.printStackTrace();
                        });
 `

and the second one is almost the same.

**10-15 frames per sec after start
2-3 frames per second after 20-30 min.**

**I have made an investigation into memory leaks, as before for early version, but didn't find any yet.**

After several hours of scanning, it can completely stop and the only logs I can see is the following:

08-03 17:28:05.088 26745-26745/? D/BluetoothAdapter: isLeEnabled(): ON
08-03 17:28:05.090 27582-27604/? D/BtGatt.GattService: stopScan() - queue size =2
08-03 17:28:05.097 27582-27632/? D/BtGatt.GattService: unregisterClient() - clientIf=6
08-03 17:28:05.102 26745-26745/? D/BluetoothAdapter: isLeEnabled(): ON
08-03 17:28:05.103 27582-27602/? D/BtGatt.GattService: onScanFilterParamsConfigured() - clientIf=6, status=0, action=1, availableSpace=30
08-03 17:28:05.103 27582-27602/? D/BtGatt.ScanManager: callback done for clientIf - 6 status - 0
08-03 17:28:05.103 27582-27606/? D/BtGatt.ScanManager: configureRegularScanParams() - queue=1
08-03 17:28:05.103 27582-27606/? D/BtGatt.ScanManager: configureRegularScanParams() - ScanSetting Scan mode=0 mLastConfiguredScanSetting=0
08-03 17:28:05.106 27582-27637/? D/BtGatt.GattService: registerClient() - UUID=f0d77122-dd21-4fa1-bc2a-65633b70e5c3
08-03 17:28:05.108 27582-27602/? D/BtGatt.GattService: onClientRegistered() - UUID=f0d77122-dd21-4fa1-bc2a-65633b70e5c3, clientIf=6
08-03 17:28:05.108 26745-26758/? D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6 mClientIf=0
08-03 17:28:05.109 27582-27593/? D/BtGatt.GattService: start scan with filters
08-03 17:28:05.117 27582-27606/? D/BtGatt.ScanManager: handling starting scan
08-03 17:28:05.121 27582-27602/? D/BtGatt.GattService: onScanFilterEnableDisabled() - clientIf=6, status=0, action=1
08-03 17:28:05.121 27582-27602/? D/BtGatt.ScanManager: callback done for clientIf - 6 status - 0
08-03 17:28:05.122 27582-27606/? D/BtGatt.ScanManager: configureFilterParamter 500 10000 1 0
08-03 17:28:05.124 27582-27602/? D/BtGatt.GattService: onScanFilterParamsConfigured() - clientIf=6, status=0, action=0, availableSpace=29
08-03 17:28:05.124 27582-27602/? D/BtGatt.ScanManager: callback done for clientIf - 6 status - 0
08-03 17:28:05.124 27582-27606/? D/BtGatt.ScanManager: configureRegularScanParams() - queue=2
08-03 17:28:05.124 27582-27606/? D/BtGatt.ScanManager: configureRegularScanParams() - ScanSetting Scan mode=2 mLastConfiguredScanSetting=0
08-03 17:28:05.124 27582-27606/? D/BtGatt.ScanManager: configureRegularScanParams - scanInterval = 8000configureRegularScanParams - scanWindow = 8000
08-03 17:28:05.125 27582-27602/? D/BtGatt.GattService: onScanParamSetupCompleted : 0
08-03 17:28:10.126 26745-26745/? D/BluetoothAdapter: isLeEnabled(): ON
08-03 17:28:10.128 27582-27632/? D/BtGatt.GattService: stopScan() - queue size =2
08-03 17:28:10.131 27582-27594/? D/BtGatt.GattService: unregisterClient() - clientIf=6
08-03 17:28:10.135 27582-27602/? D/BtGatt.GattService: onScanFilterParamsConfigured() - clientIf=6, status=0, action=1, availableSpace=30
08-03 17:28:10.135 27582-27602/? D/BtGatt.ScanManager: callback done for clientIf - 6 status - 0
08-03 17:28:10.135 26745-26745/? D/BluetoothAdapter: isLeEnabled(): ON
08-03 17:28:10.136 27582-27606/? D/BtGatt.ScanManager: configureRegularScanParams() - queue=1
08-03 17:28:10.136 27582-27606/? D/BtGatt.ScanManager: configureRegularScanParams() - ScanSetting Scan mode=0 mLastConfiguredScanSetting=2
08-03 17:28:10.136 27582-27606/? D/BtGatt.ScanManager: configureRegularScanParams - scanInterval = 8000configureRegularScanParams - scanWindow = 800
08-03 17:28:10.138 27582-27602/? D/BtGatt.GattService: onScanParamSetupCompleted : 0
08-03 17:28:10.144 27582-27615/? D/BtGatt.GattService: registerClient() - UUID=0df5f750-1aba-4147-9270-cabf44a9dc40
08-03 17:28:10.146 27582-27602/? D/BtGatt.GattService: onClientRegistered() - UUID=0df5f750-1aba-4147-9270-cabf44a9dc40, clientIf=6
08-03 17:28:10.147 26745-704/? D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6 mClientIf=0
08-03 17:28:10.147 27582-27617/? D/BtGatt.GattService: start scan with filters
08-03 17:28:10.154 27582-27606/? D/BtGatt.ScanManager: handling starting scan
08-03 17:28:10.159 27582-27602/? D/BtGatt.GattService: onScanFilterEnableDisabled() - clientIf=6, status=0, action=1
08-03 17:28:10.159 27582-27602/? D/BtGatt.ScanManager: callback done for clientIf - 6 status - 0
08-03 17:28:10.159 27582-27606/? D/BtGatt.ScanManager: configureFilterParamter 500 10000 1 0
08-03 17:28:10.162 27582-27602/? D/BtGatt.GattService: onScanFilterParamsConfigured() - clientIf=6, status=0, action=0, availableSpace=29
08-03 17:28:10.162 27582-27602/? D/BtGatt.ScanManager: callback done for clientIf - 6 status - 0
08-03 17:28:10.162 27582-27606/? D/BtGatt.ScanManager: configureRegularScanParams() - queue=2
08-03 17:28:10.162 27582-27606/? D/BtGatt.ScanManager: configureRegularScanParams() - ScanSetting Scan mode=0 mLastConfiguredScanSetting=0

dariuszseweryn, 2017-08-04:
Hello @lukevr 
You are probably testing on Android 5.0+. I cannot find the source right now but the Android considers BLE scanning to be a battery expensive operation and after a longer period of time it switches back to a more relaxed timings for scan or even stops the scan. It has nothing to do with the library itself.
I see that you are trying to create a beacon centered application — maybe it would be better to use some kind of beacon centered library? For instance [ReactiveBeacons](https://github.com/pwittchen/ReactiveBeacons)?

lukevr, 2017-08-04:
Hi @dariuszseweryn,
I'm testing on Nexus 6P, Android 7.1.2 as the main device and also tried on other phones of different vendors and Android versions. 
Same behavior exists everywhere.

dariuszseweryn, 2017-08-04:
https://stackoverflow.com/questions/43833904/android-bluetooth-le-scanner-stops-after-a-time/44488991#44488991
The degraded performance has nothing to do directly with the library and beacons use-case is out of scope to be honest. Closing.

closed 2017-08-04 by dariuszseweryn