dariuszseweryn/RxAndroidBle #262: BleCharacteristicNotFoundException when mocking characteristic write with MockRxAndroidBle

jdrider, 2017-08-03:
### Summary

I posted this issue to [StackOverflow](https://stackoverflow.com/questions/44894337/mocking-ble-characteristic-write-with-mockrxandroidble) but haven't received any responses.  There are more details there, but the main issue is that an exception is being thrown when attempting to write to a characteristic using MockRxAndroidBle.

Example code that reproduces this issue can be found in this [gist](https://gist.github.com/jdrider/9bf2d52950f2faf1f84d6b6cf62413a7)

### Library version
`1.3.2, 1.3.3`

#### Actual result
BleCharacteristicNotFoundException is thrown.
#### Expected result
The write operation is mocked successfully.

dariuszseweryn, 2017-08-04:
Hello @jdrider 
It is possible that your issue is related to #128 
I am currently a bit constrained on my time and @mzgreen as well so I cannot give any estimates on when we will address this issue. If you have the time to dive into this — you are more than welcome.

dariuszseweryn, 2017-08-04:
Closed by mistake — missclick.

mapyo, 2017-08-30:
@jdrider @dariuszseweryn 
I solved this issue.

https://stackoverflow.com/questions/44894337/mocking-ble-characteristic-write-with-mockrxandroidble
> java.lang.RuntimeException: Method setValue in android.bluetooth.BluetoothGattCharacteristic not mocked.

This error seems that the mock of Android API is not working well.

I made a pull request to reproduce this problem.
https://github.com/mapyo/RxAndroidBle/pull/1

And I fixed it.
https://github.com/mapyo/RxAndroidBle/pull/2

I setted up [Robolectric](http://robolectric.org/).
Please confirm.

jdrider, 2017-08-30:
Thank you for investigating this issue @mapyo.  Ideally, I would like a solution that doesn't involve adding a dependency on Robolectric.

dariuszseweryn, 2017-08-30:
Unfortunately it is not possible to solve without fully wrapping the Android API in the library.
Currently the Android API is visible in the Library API — we have started internal discussion on wether / how to fully enclose the Android API.

The Mock Library was initially (at least by me) considered to be a drop-in replacement for in-app usage — not for using in tests but @uKL has memories that suggest it should work also in the tests.

Bottom line — in the current state the Mock Library will not work correctly if the Android sources are not available. A potential workaround is to use `Robolectric` in the tests. We are considering an API that will fully enclose the Android API and would make it possible to use the Mock Library virtually everywhere.

jdrider, 2017-09-01:
Thank you for the explanation @dariuszseweryn.  It would be nice to have the mock library support unit tests.  I will rethink how to perform my tests or use Robolectric as a workaround and consider this issue resolved.

Side Note - The README for this project is misleading as it indicates that the MockRxAndroidBle module can be used for unit testing.  It would be good to add a note that Robolectric is required at this time.

closed 2017-09-01 by jdrider

dariuszseweryn, 2017-09-01:
I agree - this is misleading. We will add the note and we are evaluating an idea to enhance the new API proposition #239 to wrap the Android API fully - it would allow to use the mock library in the unit tests without the need of `Robolectric`.