dariuszseweryn/RxAndroidBle #263: Tag getName() as Nullable in RxBleDevice

vpratfr, 2017-08-07:
When using Kotlin, bleDevice.name is a property which is assumed to be null-safe.

This is not the case as name can be null.

An annotation `@Nullable` should hint Kotlin that the generated name property can have null values and should be considered as `String?` instead of simply `String`

Noticed that when getting a NPE in my code:

`scanResult.bleDevice.name.startsWith("EXAMPLE-", true)` give an exception: 

```
java.lang.IllegalArgumentException: Parameter specified as non-null is null: method kotlin.jvm.internal.Intrinsics.checkParameterIsNotNull, parameter $receiver
    at kotlin.text.StringsKt__StringsJVMKt.startsWith(StringsJVM.kt)
```

uKL, 2017-08-21:
Thanks for letting us know! It will be added soon.

dariuszseweryn, 2017-08-28:
Should be available in `1.3.4-SNAPSHOT` shortly.

closed 2017-08-28 by dariuszseweryn