dariuszseweryn/RxAndroidBle #264: Disconnect callback

toaderandrei, 2017-08-08:
Is there a possible way to get a callback after the disconnect or at least a method to manually disconnect. The way I see it, it will disconnect after the unsubscription.
I want to be able to do some operations before or after the disconnect.

closed 2017-08-08 by toaderandrei

dariuszseweryn, 2017-08-08:
Hello @toaderandrei 
I assume you have already managed to achieve what you need?

toaderandrei, 2017-08-08:
Hi, kind of, it would still be ok to have a method to disconnect, that returns an Observable or Flowable because my use case is to do a bit more as I have a custom bluetooth board.

dariuszseweryn, 2017-08-08:
You would probably could use `RxBleDevice.observeConnectionStateChanges()` on version `1.4.0-SNAPSHOT` which emits at almost exact moments when `BluetoothGatt` changes states.

Unfortunately there will not be any explicit `.disconnect()` method / function.

toaderandrei, 2017-08-08:
Hi, thanks for the update, I already discovered this method and so far it fits my needs so to say. That is why I closed the issue.

toaderandrei, 2017-08-08:
@dariuszseweryn , side question, when will you update to rxjava 2?, or is there any plan to update to rxjava 2?

dariuszseweryn, 2017-08-08:
> is there any plan to update to rxjava 2?

Yes
> when will you update to rxjava 2?

Don't know yet — first we are developing a better suited API #239 to test it before major version update
[RxJava 2](https://github.com/Polidea/RxAndroidBle/issues/72) related topic 

Best Regards

GuillaumeBo, 2018-03-01:
I was trying to make a "reconnect" function based on the observeConnectionStateChanges, but the Close function is called by RxBle after Disconnected status (logical case).This cause force disconnection if we already relaunch a connection between the the Disconnected event and the close.
There is any workarround to be notified that RxBle lib has done every actions to close properly a connection ? Adding the Close state to the observeConnectionStateChanges maybe ?

dariuszseweryn, 2018-03-02:
Reconnect function should be based on the `RxBleDevice.establishConnection()` as this is the point which is dealing with the connection. `RxBleDevice.observeConnectionStateChanges()` is just a shortcut if one would need to display the state (the only value that is emitted from this observable) in the UI

GuillaumeBo, 2018-03-07:
I am agree with you and use also this function for ui :)
With a light delay to ensure close() was called before call reconnection it's working. It's not ideal but it's rare case (need to refresh gatt during a connection, the self method is not enough). I will switch if a better solution appears in the futur

dariuszseweryn, 2018-03-08:
I do not get your use-case nor what you are trying to achieve unfortunately

GuillaumeBo, 2018-03-16:
Some time when the connection between the smartphone and the device is not closed properly, at the next connection, the smartphone is not able to write/read on the device (Generally this error : GATT_INVALID_HANDLE, and sometime these : GATT_READ_NOT_PERMIT, GATT_WRITE_NOT_PERMIT)
The only solution to fix it is to refresh BLEGatt and disconnect/reconnect. So i made this process at the run time when i meet this error to finally retry the write/read :)
(I cannot know at the connection if it will be bugged or not, only when i trying write/read)