dariuszseweryn/RxAndroidBle #266: Disable bluetooth

toaderandrei, 2017-08-11:
I have a weird problem when disabling bluetooth while connected.
Although I have an error subscriber it still crashes my app.
**Full log below:**
08-11 16:19:53.861 31863-3753/cc.xxx.yyyy.sdk D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=18:93:D7:5B:65:EE
08-11 16:19:53.862 31863-31863/cc.xxx.yyyy.sdk E/cc.xxx.yyyy.CustomViewModel: exception:BleDisconnectedException{bluetoothDeviceAddress='18:93:D7:5B:65:EE', cause=rx.exceptions.OnErrorThrowable$OnNextValue: OnError while emitting onNext value: android.util.Pair.class}
08-11 16:19:53.862 31863-31863/cc.robart.bluetooth.sdk E/cc.xxx.yyyy.app: throwable:BleDisconnectedException{bluetoothDeviceAddress='18:93:D7:5B:65:EE', cause=rx.exceptions.OnErrorThrowable$OnNextValue: OnError while emitting onNext value: android.util.Pair.class}
08-11 16:19:53.864 31863-31955/cc.robart.bluetooth.sdk E/AndroidRuntime: FATAL EXCEPTION: pool-1-thread-1
                                                                         Process: cc.xxx.yyyy.sdk, PID: 31863
                                                                         Theme: themes:{default=overlay:system, iconPack:system, fontPkg:system, com.android.systemui=overlay:system, com.android.systemui.navbar=overlay:system}
                                                                         java.lang.IllegalStateException: Exception thrown on Scheduler.Worker thread. Add `onError` handling.
                                                                             at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:57)
                                                                             at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run(ExecutorScheduler.java:107)
                                                                             at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                                             at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                                             at java.lang.Thread.run(Thread.java:818)
                                                                          Caused by: rx.exceptions.OnErrorNotImplementedException
                                                                             at rx.internal.util.InternalObservableUtils$ErrorNotImplementedAction.call(InternalObservableUtils.java:386)
                                                                             at rx.internal.util.InternalObservableUtils$ErrorNotImplementedAction.call(InternalObservableUtils.java:383)
                                                                             at rx.internal.util.ActionSubscriber.onError(ActionSubscriber.java:44)
                                                                             at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:153)
                                                                             at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
                                                                             at rx.observers.SerializedObserver.onError(SerializedObserver.java:152)
                                                                             at rx.observers.SerializedSubscriber.onError(SerializedSubscriber.java:78)
                                                                             at rx.internal.operators.OperatorTakeUntil$1.onError(OperatorTakeUntil.java:50)
                                                                             at rx.internal.operators.OnSubscribeMap$MapSubscriber.onError(OnSubscribeMap.java:88)
                                                                             at rx.internal.operators.OnSubscribeFilter$FilterSubscriber.onError(OnSubscribeFilter.java:90)
                                                                             at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.checkTerminated(OperatorObserveOn.java:273)
                                                                             at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.call(OperatorObserveOn.java:216)
                                                                             at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:55)
                                                                             at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run(ExecutorScheduler.java:107) 
                                                                             at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113) 
                                                                             at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588) 
                                                                             at java.lang.Thread.run(Thread.java:818) 
                                                                          Caused by: BleDisconnectedException{bluetoothDeviceAddress='18:93:D7:5B:65:EE', cause=rx.exceptions.OnErrorThrowable$OnNextValue: OnError while emitting onNext value: android.util.Pair.class}
                                                                             at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.call(RxBleGattCallback.java:64)
                                                                             at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.call(RxBleGattCallback.java:61)
                                                                             at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:69)
                                                                             at rx.internal.operators.OnSubscribeFilter$FilterSubscriber.onNext(OnSubscribeFilter.java:76)
                                                                             at com.jakewharton.rxrelay.RelaySubscriptionManager$RelayObserver.onNext(RelaySubscriptionManager.java:205)
                                                                             at com.jakewharton.rxrelay.PublishRelay.call(PublishRelay.java:47)
                                                                             at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$4.onConnectionStateChange(RxBleGattCallback.java:92)
                                                                             at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:181)
                                                                             at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
                                                                             at android.os.Binder.execTransact(Binder.java:453)
                                                                          Caused by: rx.exceptions.OnErrorThrowable$OnNextValue: OnError while emitting onNext value: android.util.Pair.class
                                                                             at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:73)
                                                                             at rx.internal.operators.OnSubscribeFilter$FilterSubscriber.onNext(OnSubscribeFilter.java:76) 
                                                                             at com.jakewharton.rxrelay.RelaySubscriptionManager$RelayObserver.onNext(RelaySubscriptionManager.java:205) 
                                                                             at com.jakewharton.rxrelay.PublishRelay.call(PublishRelay.java:47) 
                                                                             at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$4.onConnectionStateChange(RxBleGattCallback.java:92) 
                                                                             at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:181) 
                                                                             at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70) 
                                                                             at android.os.Binder.execTransact(Binder.java:453) 
[... cut ...]
08-11 16:19:53.862 31863-31863/cc.xxx.yyyy.sdk E/cc.xxx.yyyy.CustomViewModel: exception:BleDisconnectedException{bluetoothDeviceAddress='18:93:D7:5B:65:EE', cause=rx.exceptions.OnErrorThrowable$OnNextValue: OnError while emitting onNext value: android.util.Pair.class}
08-11 16:19:53.862 31863-31863/cc.robart.bluetooth.sdk E/cc.xxx.yyyy.app: throwable:BleDisconnectedException{bluetoothDeviceAddress='18:93:D7:5B:65:EE', cause=rx.exceptions.OnErrorThrowable$OnNextValue: OnError while emitting onNext value: android.util.Pair.class}

closed 2017-08-12 by toaderandrei

craigzour, 2017-08-12:
@toaderandrei Did you find a solution ? I encountered the same issue yesterday. It is related to the fact that I do a `retry()` on the `establishConnection()`.

toaderandrei, 2017-08-12:
@craigzour , I did and it seems I had a code inside a flatMap that was missing the error block. This code was after establishConnection. In my case the crash was happening because of a device disconnect from bluetooth(turned off bluetooth on the device). I could catch the disconnect callback but, because of the missing error block, I had a crash.