dariuszseweryn/RxAndroidBle #269: BleGattException status return 133

evansherry, 2017-08-21:
When I connect ble after I try to open the notification, but often return BleGattException status = 133, sometimes it will sometimes not .
this is my code , i set it after 500 ms open notify:
mConnectingDevice.establishConnection(false)
                    .doOnNext(this::storeConnect)
                    .delay(500, TimeUnit.MILLISECONDS)
                    .flatMap(rxBleConnection -> rxBleConnection.setupNotification(notifyUuid))
                    .observeOn(AndroidSchedulers.mainThread())
                    .flatMap(notificationObservable -> notificationObservable)
                    .observeOn(AndroidSchedulers.mainThread())
                    .subscribe(this::onNotify, this::connectFail); 
connectFail return BleGattException status = 133

Ruifen9, 2017-08-22:
I also often encounter this problem when connecting the same device,you must call "BluetoothGatt.close" method in Google api.

evansherry, 2017-08-22:
中文：我使用的是RxAndroidBle，没有使用android api，我断开连接是调用connectSubscription.unsubscribe()，这个方法内部肯定是gatt.close了吧？

english:
I am using RxAndroidBle, do not use the android api, i disconnect is called connectSubscription.unsubscribe ();  is this method internally calling gatt.close?

evansherry, 2017-08-22:
This is the print log    // behind my comments：
08-22 14:12:00.777 31399-31399/com.evan.yeeuusample D/BluetoothAdapter: STATE_ON
08-22 14:12:00.781 31399-31399/com.evan.yeeuusample D/BluetoothGatt: connect() - device: F0:65:A0:1A:20:16, auto: false                //mConnectingDevice.establishConnection(false)
08-22 14:12:00.781 31399-31399/com.evan.yeeuusample D/BluetoothGatt: registerApp()
08-22 14:12:00.782 31399-31399/com.evan.yeeuusample D/BluetoothGatt: registerApp() - UUID=58cec16e-bb91-4603-bd28-11da0e5e17c0
08-22 14:12:00.785 31399-31410/com.evan.yeeuusample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5

08-22 14:12:02.965 31399-31410/com.evan.yeeuusample D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=F0:65:A0:1A:20:16        // connectstatus 133
08-22 14:12:02.969 31399-31399/com.evan.yeeuusample D/BluetoothManager: getConnectionState()
08-22 14:12:02.970 31399-31399/com.evan.yeeuusample D/BluetoothManager: getConnectedDevices
08-22 14:12:02.997 31399-31399/com.evan.yeeuusample D/BluetoothGatt: close()      // connect fail   
 call connectSubscription.unsubscribe()
08-22 14:12:02.997 31399-31399/com.evan.yeeuusample D/BluetoothGatt: unregisterApp() - mClientIf=5

Continue to try to connect:

08-22 14:17:35.278 31399-31399/com.evan.yeeuusample D/BluetoothGatt: connect() - device: F0:65:A0:1A:20:16, auto: false
08-22 14:17:35.278 31399-31399/com.evan.yeeuusample D/BluetoothGatt: registerApp()
08-22 14:17:35.278 31399-31399/com.evan.yeeuusample D/BluetoothGatt: registerApp() - UUID=b8fa445c-aab2-46d1-b2c2-e67bb475caf9
08-22 14:17:35.280 31399-31411/com.evan.yeeuusample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
08-22 14:17:43.666 31399-31441/com.evan.yeeuusample D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=F0:65:A0:1A:20:16
Failed to connect again！
Is there any solution? i do not know why the connection status is 133, gatt has been closed! Now this state happens frequently, help me thank you .

uKL, 2017-08-22:
Hey!

Thank you for contacting us. This seams to be a general question rather than an issue. Please ask on stackoverflow.com and add tag `rxandroidble`

closed 2017-08-22 by uKL

dariuszseweryn, 2017-09-18:
@evansherry the `status=133` is a generic `GATT_ERROR` message. It is caused mostly by how the BLE stack works internally on your phone. The issue is most probably related to how your peripheral—phone pair works.
If you would like to have more information you would need to get detailed [HCI logs](https://blog.bluetooth.com/debugging-bluetooth-with-an-android-app) from a Xiaomi/Huawei device that shows the problem and another from a phone that works fine.