dariuszseweryn/RxAndroidBle #273: CONNECTING forever

hoanglm4, 2017-08-30:
### Summary
**CONNECTING state forever.**

I create unit test as below:
```
public class CharacteristicOperationExampleActivity extends RxAppCompatActivity {
    private static final UUID characteristicUuid = UUID.fromString("25A80001-6478-11E6-BDF4-AD3BCC77759F");
    private PublishSubject<Void> disconnectTriggerSubject = PublishSubject.create();
    private Observable<RxBleConnection> connectionObservable;
    private RxBleDevice bleDevice;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_example4);
        ButterKnife.bind(this);
        String macAddress = "DC:00:DC:98:2B:4D";
        bleDevice = SampleApplication.getRxBleClient(this).getBleDevice(macAddress);
        connectionObservable = prepareConnectionObservable();

        bleDevice.observeConnectionStateChanges().subscribe(state -> {
            Log.i(getClass().getSimpleName(), "" + state.toString());
            if (state == RxBleConnection.RxBleConnectionState.DISCONNECTED) {
                connect();
            }
        });
    }

    @OnClick(R.id.connect)
    public void onConnectToggleClick() {
        for (int i = 0; i < 10; i++) {
            connect();
        }
    }

    private void connect() {
        Observable.timer(1, TimeUnit.SECONDS)
                .observeOn(AndroidSchedulers.mainThread())
                .subscribe(aLong -> {
                    triggerDisconnect();
                    getConnectionObservable()
                            .flatMap(RxBleConnection::discoverServices)
                            .flatMap(rxBleDeviceServices -> rxBleDeviceServices.getCharacteristic(characteristicUuid))
                            .observeOn(AndroidSchedulers.mainThread())
                            .doOnSubscribe(() -> Log.i(getClass().getSimpleName(), "start connect "))
                            .subscribe(
                                    characteristic -> {
                                        Log.i(getClass().getSimpleName(), "Hey, connection has been established!");
                                    },
                                    this::onConnectionFailure,
                                    this::onConnectionFinished
                            );
                });
    }

    private Observable<RxBleConnection> prepareConnectionObservable() {
        return bleDevice
                .establishConnection(false)
                .takeUntil(disconnectTriggerSubject)
                .compose(bindUntilEvent(PAUSE))
                .compose(new ConnectionSharingAdapter());
    }

    private Observable<RxBleConnection> getConnectionObservable() {
        if (isConnecting()) {
            return Observable.empty();
        }
        return connectionObservable;
    }

[... 70 lines cut ...]
> 08-30 15:30:10.441 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: start connect 
> 08-30 15:30:10.441 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: onConnectionFinished: 
> 08-30 15:30:10.461 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: check stateRxBleConnectionState{CONNECTING}
> 08-30 15:30:10.461 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: start connect 
> 08-30 15:30:10.461 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: onConnectionFinished: 
> 08-30 15:30:10.471 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: check stateRxBleConnectionState{CONNECTING}
> 08-30 15:30:10.471 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: start connect 
> 08-30 15:30:10.471 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: onConnectionFinished: 
> 08-30 15:30:10.561 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: check stateRxBleConnectionState{CONNECTING}
> 08-30 15:30:10.561 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: start connect 
> 08-30 15:30:10.561 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: onConnectionFinished: 
> 08-30 15:30:10.661 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: check stateRxBleConnectionState{CONNECTING}
> 08-30 15:30:10.661 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: start connect 
> 08-30 15:30:10.661 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: onConnectionFinished: 
**> 08-30 15:30:10.671 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: check stateRxBleConnectionState{CONNECTING}**
> 08-30 15:30:10.671 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: start connect 
> 08-30 15:30:10.671 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: onConnectionFinished: 
> 08-30 15:33:52.331 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: check stateRxBleConnectionState{CONNECTING}
> 08-30 15:33:52.331 13037-13037/com.polidea.rxandroidble.sample I/CharacteristicOperationExampleActivity: start connect 
```

#### Expected result
In step 5, about 35 seconds to get BleGattCallbackTimeoutException => DISCONNECTED state. However, DISCONNECTED state is not occur.
#### Library version
`1.3.3`

dariuszseweryn, 2017-08-30:
Hello @hoanglm4 
Are you sure that all subscriptions to `RxBleDevice.establishConnection()` are unsubscribed? In the `1.3.x` branch the connection state reporting is quite naive and report's `DISCONNECTED` state when the unsubscription happens.
Anyways a good finding as this would be definitely a problem in `1.4.0`

hoanglm4, 2017-08-30:
Dear @dariuszseweryn ,

> Are you sure that all subscriptions to RxBleDevice.establishConnection() are unsubscribed?

Yes. I'm sure. Because I call **triggerDisconnect()** before **subscribe** RxBleDevice.establishConnection()

Thank you so much. This issue is not occur in your fixed

dariuszseweryn, 2017-08-30:
It was probably some kind of a race condition. Anyways — I'm glad that it works in the `develop` branch.

Best Regards

closed 2017-08-30 by dariuszseweryn