dariuszseweryn/RxAndroidBle #275: Ble is not disconnected when BluetoothGatt: android.os.DeadObjectException is thrown

hoanglm4, 2017-09-01:
### Summary
CONNECTED state forever when BluetoothGatt: android.os.DeadObjectException is thrown

My code in below:
```
   private static final UUID characteristicUuid = UUID.fromString("25A80001-6478-11E6-BDF4-AD3BCC77759F");

   @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_example4);
        ButterKnife.bind(this);
        bleDevice = SampleApplication.getRxBleClient(this).getBleDevice(macAddress);
        connectionObservable = prepareConnectionObservable();
        bleDevice.observeConnectionStateChanges().subscribe(state -> {
            Log.i(getClass().getSimpleName(), "" + state.toString());
        });
        connect();
    }

    private Observable<RxBleConnection> prepareConnectionObservable() {
        return bleDevice
                .establishConnection(false)
                .takeUntil(disconnectTriggerSubject)
                .compose(bindUntilEvent(PAUSE))
                .compose(new ConnectionSharingAdapter());
    }

    @OnClick(R.id.connect)
    public void connect() {
        connectionObservable.flatMap(RxBleConnection::discoverServices)
                .flatMap(rxBleDeviceServices -> rxBleDeviceServices.getCharacteristic(characteristicUuid))
                .observeOn(AndroidSchedulers.mainThread())
                .doOnSubscribe(() -> Log.i(getClass().getSimpleName(), "start connect "))
                .subscribe(
                        characteristic -> {
                            Log.i(getClass().getSimpleName(), "Hey, connection has been established!");
                        },
                        this::onConnectionFailure,
                        this::onConnectionFinished
                );
    }

   @OnClick(R.id.dissconnect)
    public void triggerDisconnect() {
        disconnectTriggerSubject.onNext(null);
    }

     @OnClick(R.id.read_rssi)
    public void readRSSI() {
        if (bleDevice.getConnectionState() != RxBleConnection.RxBleConnectionState.CONNECTED) {
            return;
        }
        connectionObservable.map(rxBleConnection -> rxBleConnection.readRssi())
                .subscribe(rssi -> {
                    Log.i(getClass().getSimpleName(), "RSSI = " + rssi);
                }, throwable -> {
                    Log.i(getClass().getSimpleName(), "Connection error: " + throwable);
                });
    }
```

- Sometime, when Bluetooth in phone is crashed => I call **readRSSI** => **BluetoothGatt: android.os.DeadObjectException** is thrown. But RxBleDevice#getConnectionState() return RxBleConnectionState{CONNECTED}.
- When call **readRSSI** again, **BluetoothGatt: android.os.DeadObjectException** still thown.
- After, I call **triggerDisconnect**() method =>  **connect**() method=>  try to **readRSSI**, and it work fine.

#### Actual result
BluetoothGatt: android.os.DeadObjectException is thown
Please see log:
```
[... 31 lines cut ...]
                                                                              at android.bluetooth.IBluetoothGatt$Stub$Proxy.readRemoteRssi(IBluetoothGatt.java:1006)
                                                                              at android.bluetooth.BluetoothGatt.readRemoteRssi(BluetoothGatt.java:1121)
                                                                              at com.polidea.rxandroidble.internal.operations.RxBleRadioOperationReadRssi.startOperation(RxBleRadioOperationReadRssi.java:30)
                                                                              at com.polidea.rxandroidble.internal.RxBleSingleGattRadioOperation.protectedRun(RxBleSingleGattRadioOperation.java:56)
                                                                              at com.polidea.rxandroidble.internal.RxBleRadioOperation$1.call(RxBleRadioOperation.java:39)
                                                                              at com.polidea.rxandroidble.internal.RxBleRadioOperation$1.call(RxBleRadioOperation.java:35)
                                                                              at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:72)
                                                                              at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:32)
                                                                              at rx.Observable.unsafeSubscribe(Observable.java:10256)
                                                                              at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.call(OperatorSubscribeOn.java:100)
                                                                              at rx.android.schedulers.LooperScheduler$ScheduledAction.run(LooperScheduler.java:107)
                                                                              at android.os.Handler.handleCallback(Handler.java:751)
                                                                              at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                              at android.os.Looper.loop(Looper.java:154)
                                                                              at android.app.ActivityThread.main(ActivityThread.java:6126)
                                                                              at java.lang.reflect.Method.invoke(Native Method)
                                                                              at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
                                                                              at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
09-01 11:49:25.141 13204-13302/com.mysoftsource.vuzitech D/RxBle#Radio: FINISHED RxBleRadioOperationReadRssi(189325052)
```

#### Expected result
When BluetoothGatt: android.os.DeadObjectException is thrown => call disconnect and shown DISCONNECTED state
### Library version
`1.3.3`

dariuszseweryn, 2017-09-01:
Could you check against `1.4.0-SNAPSHOT` version? The code related to reporting state has changed there.

hoanglm4, 2017-09-01:
Dear @dariuszseweryn,

I have checked this issue in 1.4.0-SNAPSHOT, this bug is occur too. However, Bluetooth is crashed on Samsung Note II (rooted). I can not make bluetooth on Samsung J7(or HTC U11 because they are not rooted) is crashed.

Thanks

dariuszseweryn, 2017-09-11:
@hoanglm4 I have just pushed a new version of `1.4.0-SNAPSHOT` — it should now properly handle situation when BluetoothAdapter is turned off. Could you check it out?

hoanglm4, 2017-09-11:
Dear @dariuszseweryn , thank you so much. I will check your fixed in tomorrow (currently, my country in midnight, so I don't have device for check it)

hoanglm4, 2017-09-12:
Dear @dariuszseweryn,
 I can not connect to ble device. **"java.lang.NullPointerException: Cannot return null from a non-@Nullable @Provides method"** is thown when **connect()** method called

Version: 1.3.4-SNAPSHOT

Thanks

dariuszseweryn, 2017-09-12:
@hoanglm4 my bad — just pushed a fix. There should be a new version of `1.3.4-SNAPSHOT` available in around 10 minutes. Feel free to re-check.

hoanglm4, 2017-09-12:
Dear @dariuszseweryn,

I have tested this issue. "**DISCONNECTED**" state is shown when bluetooth is crashed. I think your fixed is OK.

Thanks

dariuszseweryn, 2017-09-12:
I am happy to hear that. :) Thank you for your report.

closed 2017-09-12 by dariuszseweryn

dariuszseweryn, 2017-09-12:
@hoanglm4 I have just released an official `1.3.4` version :)
Best Regards

hoanglm4, 2017-09-12:
Dear @dariuszseweryn,
Great! When you release 1.4.0?

Thanks

dariuszseweryn, 2017-09-12:
I will probably try to release it tomorrow.

dariuszseweryn, 2017-09-13:
FYI—it is released.

hoanglm4, 2017-09-13:
Great! Thank you so much :)

omeryounus, 2020-02-11:
I am getting this error when calling disconnect gatt.

` E/BluetoothGatt: android.os.DeadObjectException
        at android.os.BinderProxy.transactNative(Native Method)
        at android.os.BinderProxy.transact(Binder.java:503)
        at android.bluetooth.IBluetoothGatt$Stub$Proxy.clientDisconnect(IBluetoothGatt.java:925)
        at android.bluetooth.BluetoothGatt.disconnect(BluetoothGatt.java:754)
        at com.kasiel.solusguard.link.RxBluetoothGatt.completelyCloseBluetoothConnection(RxBluetoothGatt.kt:321)
        at com.kasiel.solusguard.link.RxBluetoothGatt$connectionFlowable$3.invoke(RxBluetoothGatt.kt:90)
        at com.kasiel.solusguard.link.RxBluetoothGatt$connectionFlowable$3.invoke(RxBluetoothGatt.kt:25)
        at com.kasiel.solusguard.link.RxBluetoothGattKt$sam$io_reactivex_functions_Action$0.run(RxBluetoothGatt.kt)
        at io.reactivex.internal.operators.flowable.FlowableDoOnLifecycle$SubscriptionLambdaSubscriber.cancel(FlowableDoOnLifecycle.java:115)
        at io.reactivex.internal.subscriptions.SubscriptionHelper.cancel(SubscriptionHelper.java:181)
        at io.reactivex.internal.operators.flowable.FlowableFlatMap$InnerSubscriber.dispose(FlowableFlatMap.java:695)
        at io.reactivex.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.disposeAll(FlowableFlatMap.java:587)
        at io.reactivex.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.cancel(FlowableFlatMap.java:354)
        at io.reactivex.internal.subscriptions.SubscriptionHelper.cancel(SubscriptionHelper.java:181)
        at io.reactivex.internal.subscribers.LambdaSubscriber.cancel(LambdaSubscriber.java:119)
        at io.reactivex.internal.subscribers.LambdaSubscriber.dispose(LambdaSubscriber.java:104)
        at io.reactivex.disposables.CompositeDisposable.dispose(CompositeDisposable.java:240)
        at io.reactivex.disposables.CompositeDisposable.dispose(CompositeDisposable.java:82)`

dariuszseweryn, 2020-02-11:
I am afraid that this has nothing to do with the library judging by the stacktrace.

omeryounus, 2020-02-11:
Okay I am going to remove it

omeryounus, 2020-02-11:
I have seen this exception happening on some low end Android phones when calling gatt.disconnect method
i:e - E/BluetoothGatt: android.os.DeadObjectException
I am sure if there is any properly handling in library that would prevent it.