dariuszseweryn/RxAndroidBle #280: Can't write characteristic on Xiaomi device

Ezor, 2017-09-12:
### Summary

Hi,

I get a 133 0x85 error when trying to write Characteristic from a Xiaomi Redmi 4X device.

I don't have this kind of problem using other devices (like Samsung ones) so i don't know where to look to fix this issue.

Any idea ?

Here is the error stack trace :

```
BleGattCharacteristicException{macAddress=EC:FE:7E:15:99:82, status=133 (0x85 -> https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h), bleGattOperationType=BleGattOperation{description='CHARACTERISTIC_WRITE'}}
at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateErrorIfOccurred(RxBleGattCallback.java:216)
at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$700(RxBleGattCallback.java:34)
at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$4.onCharacteristicWrite(RxBleGattCallback.java:122)
at android.bluetooth.BluetoothGatt$1.onCharacteristicWrite(BluetoothGatt.java:407)
at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:279)
at android.os.Binder.execTransact(Binder.java:453)
Caused by: rx.exceptions.OnErrorThrowable$OnNextValue: OnError while emitting onNext value: com.polidea.rxandroidble.exceptions.BleGattCharacteristicException.class
at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:73)
at com.jakewharton.rxrelay.RelaySubscriptionManager$RelayObserver.onNext(RelaySubscriptionManager.java:205)
at com.jakewharton.rxrelay.PublishRelay.call(PublishRelay.java:47)
at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusError(RxBleGattCallback.java:249)
... 6 more
rx.exceptions.OnErrorThrowable$OnNextValue: OnError while emitting onNext value: com.polidea.rxandroidble.exceptions.BleGattCharacteristicException.class
at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:73)
at com.jakewharton.rxrelay.RelaySubscriptionManager$RelayObserver.onNext(RelaySubscriptionManager.java:205)
at com.jakewharton.rxrelay.PublishRelay.call(PublishRelay.java:47)
at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusError(RxBleGattCallback.java:249)
at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateErrorIfOccurred(RxBleGattCallback.java:216)
at com.polidea.rxandroidble.internal.connection.RxBleGattCallback.access$700(RxBleGattCallback.java:34)
at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$4.onCharacteristicWrite(RxBleGattCallback.java:122)
at android.bluetooth.BluetoothGatt$1.onCharacteristicWrite(BluetoothGatt.java:407)
at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:279)
at android.os.Binder.execTransact(Binder.java:453)
```

### Library version
`1.3.3`

#### Code snippet triggering the issue

```java
final RxBleDevice device = rxBleClient.getBleDevice(macAddress);
        connectionObservable = device.establishConnection(false)
                .retry(3)
                .takeUntil(disconnectTriggerSubject)
                .compose(new ConnectionSharingAdapter());
        connectionObservable.subscribe(rxBleConnection -> setupNotification(),
                throwable -> {
                    Log.e(TAG, "Connection error", throwable);
                    deviceDelegate.onBluetoothFailed(throwable, "Connecting to device");
                });
```

```java

private static final int MODE_DATA = 1;

byte[] newMode = new byte[1];
        newMode[0] = (byte) BluetoothClient.MODE_DATA;
        Observable<byte[]> setDataMode = connectionObservable.flatMap(rxBleConnection -> rxBleConnection.writeCharacteristic(BluetoothClient.MODE_UUID, newMode));
        setDataMode.subscribe(modeValue -> {
            Log.d(TAG, "INIT STEP MODE DATA OK");
            initStepSendCommand(0);
        }, throwable -> {
            Log.e(TAG, "INIT STEP MODE DATA ERROR", throwable);
            deviceDelegate.onBluetoothFailed(throwable, "Setting MODE");
[... 60 lines cut ...]
09-12 16:44:51.863 E\/BluetoothClient(23064): \tat android.bluetooth.BluetoothGatt$1.onCharacteristicWrite(BluetoothGatt.java:407)
09-12 16:44:51.863 E\/BluetoothClient(23064): \tat android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:279)
09-12 16:44:51.863 E\/BluetoothClient(23064): \tat android.os.Binder.execTransact(Binder.java:453)
09-12 16:44:51.863 E\/BluetoothClient(23064): Caused by: rx.exceptions.OnErrorThrowable$OnNextValue: OnError while emitting onNext value: com.polidea.rxandroidble.exceptions.BleGattCharacteristicException.class
09-12 16:44:51.863 E\/BluetoothClient(23064): \tat rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:73)
09-12 16:44:51.863 E\/BluetoothClient(23064): \tat com.jakewharton.rxrelay.RelaySubscriptionManager$RelayObserver.onNext(RelaySubscriptionManager.java:205)
09-12 16:44:51.863 E\/BluetoothClient(23064): \tat com.jakewharton.rxrelay.PublishRelay.call(PublishRelay.java:47)
09-12 16:44:51.863 E\/BluetoothClient(23064): \tat com.polidea.rxandroidble.internal.connection.RxBleGattCallback.propagateStatusError(RxBleGattCallback.java:249)
09-12 16:44:51.863 E\/BluetoothClient(23064): \t... 6 more
09-12 16:44:51.868 D\/BluetoothGatt(23064): onClientConnectionState() - status=8 clientIf=5 device=EC:FE:7E:15:99:82
09-12 16:44:51.868 D\/RxBle#BluetoothGatt(23064): onConnectionStateChange newState=0 status=8
09-12 16:44:51.871 D\/RxBle#Radio(23064):   QUEUED RxBleRadioOperationDisconnect(137517779)
09-12 16:44:51.873 D\/RxBle#Radio(23064):  STARTED RxBleRadioOperationDisconnect(137517779)
09-12 16:44:51.874 D\/BluetoothGatt(23064): setCharacteristicNotification() - uuid: 18cda784-4bd3-4370-85bb-bfed91ec86af enable: false
09-12 16:44:51.879 D\/BluetoothManager(23064): getConnectionState()
09-12 16:44:51.879 D\/BluetoothManager(23064): getConnectedDevices
09-12 16:44:51.895 D\/BluetoothGatt(23064): close()
09-12 16:44:51.895 D\/BluetoothGatt(23064): unregisterApp() - mClientIf=5
09-12 16:44:51.898 D\/RxBle#Radio(23064): FINISHED RxBleRadioOperationDisconnect(137517779)
09-12 16:44:51.899 D\/RxBle#Radio(23064):   QUEUED RxBleRadioOperationDescriptorWrite(183014204)
09-12 16:44:51.900 D\/RxBle#Radio(23064):  STARTED RxBleRadioOperationDescriptorWrite(183014204)
09-12 16:44:51.914 D\/RxBle#Radio(23064): FINISHED RxBleRadioOperationDescriptorWrite(183014204)
09-12 16:44:51.931 V\/BoostFramework(23064): BoostFramework() : mPerf = com.qualcomm.qti.Performance@bf14a72
09-12 16:44:51.931 V\/BoostFramework(23064): BoostFramework() : mPerf = com.qualcomm.qti.Performance@979ec3
```

dariuszseweryn, 2017-09-12:
Hello @Ezor 

The `status=133` is the generic `GATT_ERROR`. So far it looks like there is a bug in Xiaomi BLE stack (which probably will not be fixed without OS update) but it could be possible to fix it by changing interaction with your BLE peripheral or by changing the firmware of your peripheral.

What you can do to gather more information is:

1. Check what properties the `BluetoothClient.MODE_UUID` has (`BluetoothGattCharacteristic.getProperties()`
2. Enable the [Bluetooth HCI snoop log](https://blog.bluetooth.com/debugging-bluetooth-with-an-android-app) and paste the output here so it would be possible to compare what Samsung does and what happens on Xiaomi

Ezor, 2017-09-12:
Well that's crazy, I just fixed the problem 2 minutes after sending this issue (I worked on it 4 hours...).

As you may notice in the log, the app is doing the following :

1. Connect to device (establishConnection)
2. Setup Indication for some "READ" Characteristic
3. Write some value on a "WRITE" Characteristic
4. Write & Read lot of data on both "READ" & "WRITE" Characteristic

I switched 2 & 3 it's now OK !!!?

1. Connect to device (establishConnection)
2. Write some value on a "WRITE" Characteristic
3. Setup Indication for some "READ" Characteristic
4. Write & Read lot of data on both "READ" & "WRITE" Characteristic

dariuszseweryn, 2017-09-12:
I would not consider this issue fixed but rather worked around. ;)
It may be because the peripheral is sending some packets at a particular time that makes the Xiaomi stack go nuts and changing the sequence makes it work.
I am curious what exactly is causing the BLE stack to return `GATT_ERROR` on Xiaomi. If you could please upload the HCI logs that show the problem from this Xiaomi and the Samsung so I could examine them and learn.

Ezor, 2017-09-12:
You are probably right, it's just a workaround.

I will post the HCI logs for the Samsung tomorrow but I'm not sure to be able to post the HCI logs for the Xiaomi because i don't have the device at home, it's the device of one of my client (working 500km far away). I will do my best.

dariuszseweryn, 2017-09-27:
Closing due to inactivity.

closed 2017-09-27 by dariuszseweryn

Lloyd0577, 2019-01-23:
I meet same problem in xiaomi device,but i use android native connect method ,it can connect,so i wonder if there is any bug in this lib,thanks