dariuszseweryn/RxAndroidBle #282: Add status to BleDisconnectedException

simonratner, 2017-09-14:
### Summary
The status param of `onConnectionStateChange` provides useful information even when new state is disconnected, since it holds the reason for disconnection, if supplied by the remote peer.

From _Bluetooth Core Specification v5.0, Vol 2, Part E_:
```
7.1.6  Disconnect Command

Command Parameters:

Reason: 
  Size: 1 octet

Value:
  0x05, 0x13-0x15, 0x1A, 0x29, 0x3B

Description:
  Authentication Failure error code (0x05), 
  Other End Terminated Connection error codes (0x13-0x15), 
  Unsupported Remote Feature error code (0x1A), 
  Pairing with Unit Key Not Supported error code (0x29),
  Unacceptable Connection Parameters error code (0x3B).
```
I believe it is also possible to get status `0x8 (Supervision Timeout)`, `0x3e (Connection Establish Timeout)`, and `0x16 (Connection Terminated By Local Host)` in a disconnect callback.

### Library version
`1.4.0`

#### Preconditions
Connection state change event indicating a disconnect with a non-zero status.
Relevant code is at [RxBleGattCallback.java#L76](https://github.com/Polidea/RxAndroidBle/blob/master/rxandroidble/src/main/java/com/polidea/rxandroidble/internal/connection/RxBleGattCallback.java#L69).

#### Actual result
`BleDisconnectedException` without a status value.

#### Expected result
`BleDisconnectedException` with a status value.

ThunderThrash, 2017-09-15:
Sorry for my English...
I would like to know if you can get the status 0x81 (GATT internal error) and 0x85 (GATT error) after closing this issue?

RobLewis, 2017-11-02:
Possibly irrelevant comment @ThunderThrash: I'm struggling right now with error 0x81, which is reportedly a "generic Bluetooth error", i.e., seems to have no diagnostic value whatever. Enlightenment appreciated!

hoanglm4, 2017-11-06:
I think BleDisconnectedException with a status value is great.

RobLewis, 2017-11-30:
Just to achieve a bit of "closure" with my previous comment: the "generic Bluetooth error" 0x81 turned out to be that I was attempting to enable notifications when I should have been using indications (or vice versa, forget which). Wouldn't it be swell if there were explicit causes given like this?

dariuszseweryn, 2017-11-30:
@RobLewis Since `1.4.0` operations that are requested but are not supported by the characteristic properties are logged as a warning. There is also an issue #259 which will be released with #239 - it will cause the above mentioned operation to emit `BleIllegalOperationException` with a explicit cause.

RobLewis, 2017-11-30:
I should have clarified that the above error was in some Android-native (non-RxAndroidBle) code I inherited. Rewriting it now with Rx. What a difference!

uKL, 2018-03-22:
@Cierpliwy Looks like this is the one we were talking about. Mentioning just to remember about it.

commit 3ae9ce36 (2018-06-04): - Added GATT status code to a BleDisconnectException
- Added GATT status to description mapping

Fixes #282
PR #405 (2018-06-08): Added GATT status code to a BleDisconnectException
- Added GATT status to description mapping

Fixes #282
commit 5787e3ca (2018-06-08): Added GATT status code to a BleDisconnectException (#405)

- Added GATT status code to a BleDisconnectException
- Added GATT status to description mapping

Fixes #282
closed 2018-06-08 by uKL