dariuszseweryn/RxAndroidBle #285: Why does it automatically reconnect after disconnecting?

XIANDONGZHI, 2017-09-18:
version is 1.4.0
This problem has been bothering me for a long time...
disconnect code:
public void disConnect() {
        disconnectTriggerSubject.onNext(null);
        bleDevice = null;
        connectionObservable = null;
}
i try call this method 
And the remote device immediately disconnects the connection
After a few seconds....It's will automatically reconnected... why? 
I try use:
xxx.establishConnection( false);//and call disConnect(); result is no effect, it still automatically reconnection
..I have broken down...
Can you tell me where the interface can be set up and not automatically reconnected?
like this:
public void disConnectAndNeverReconnectAutomatically();

this is my  code:

    private PublishSubject<Void> disconnectTriggerSubject = PublishSubject.create();
    private Observable<RxBleConnection> connectionObservable;
 
    protected RxBleDevice bleDevice;
    public void disConnect() {
        disconnectTriggerSubject.onNext(null);
        bleDevice = null;
        connectionObservable = null;
   }
    public void connectDevice(String address) {
        this.disConnect();
        this.bleDevice = this.rxBleClient.getBleDevice(address);
        if (connectionObservable == null) {
            connectionObservable = prepareConnectionObservable();
        }
        this.connectionObservable
                .observeOn(AndroidSchedulers.mainThread())
                .doOnSubscribe(new Action0() {
                    @Override
                    public void call() {
                      
                    }
                })
                .subscribe(new Action1<RxBleConnection>() {
            public void call(RxBleConnection rxBleConnection) {
 
            }
        }, new Action1<Throwable>() {
            public void call(Throwable throwable) {
                BLELog.e("connectDevice Throwable=" + throwable);
            }
        });
}

    private Observable<RxBleConnection> prepareConnectionObservable() {
        return bleDevice
                .establishConnection( true)
                .takeUntil(disconnectTriggerSubject)
                .doOnUnsubscribe(new Action0() {
                    @Override
                    public void call() {
                        connectionObservable = null;
                    }
                })
                .compose(new ConnectionSharingAdapter());
    }

uKL, 2017-09-18:
Hey,

this is a general question to the library usage. Please head to stackoverflow and tag your question with #rxandroidble tag.

closed 2017-09-18 by uKL

XIANDONGZHI, 2017-09-18:
I solved this serious problem....I rewrote and reflected...
`
    public void disConnect() {
        disconnectTriggerSubject.onNext(null);
        if (subscribe != null) {
            subscribe.unsubscribe();
        }

        bleDevice = null;
        connectionObservable = null;
    }

    public Observable<RxBleConnection> establishConnection(final boolean autoConnect) {
        ConnectionSetup options = new ConnectionSetup.Builder()
                .setAutoConnect(autoConnect)
                .setSuppressIllegalOperationCheck(true)
                .build();
        return establishConnection(options);
    }

    //    @Override
    public Observable<RxBleConnection> establishConnection(final ConnectionSetup options) {
        return Observable.defer(new Func0<Observable<RxBleConnection>>() {
            Connector connector = null;
            AtomicBoolean isConnected;

            @Override
            public Observable<RxBleConnection> call() {
                if (bleDevice==null) {
                    return Observable.error(new Exception("is disconnect!"));
                }
                try {
                    Field connectorField = bleDevice.getClass().getDeclaredField("connector");
                    final Field isConnectedField =  bleDevice.getClass().getDeclaredField("isConnected");
                    connectorField.setAccessible(true);
                    isConnectedField.setAccessible(true);
                    connector = (Connector) connectorField.get(bleDevice);
                    isConnected = (AtomicBoolean) isConnectedField.get(bleDevice);
                } catch (Exception e) {
                    e.printStackTrace();
                }
                if (isConnected.compareAndSet(false, true)) {
                    return connector.prepareConnection(options)
                            .doOnUnsubscribe(new Action0() {
                                @Override
                                public void call() {
                                    isConnected.set(false);
                                }
                            });
                } else {
                    return Observable.error(new BleAlreadyConnectedException(bleDevice.getMacAddress()));
                }

            }
        });
    }

    private Observable<RxBleConnection> prepareConnectionObservable() {
        Observable<RxBleConnection> rxBleConnectionObservable = establishConnection(false);
        return rxBleConnectionObservable
                .takeUntil(disconnectTriggerSubject)
                .doOnUnsubscribe(new Action0() {
                    @Override
                    public void call() {
                        connectionObservable = null;
                    }
                })
                .compose(new ConnectionSharingAdapter());
    }
`