dariuszseweryn/RxAndroidBle #288: Leak resulting from RxBleClient.scanBleDevices()

HolySamosa, 2017-09-27:
I've been pulling my hair out trying to debug a leak in my app when using `RxBleClient.scanBleDevices()`.  My subscribe onNext lambda holds a reference to the Activity, however I am unsubscribing when the Activity is destroyed, so I shouldn't be leaking.  The leak's GC root is a thread, which references `com.polidea.rxandroidble.internal.radio.FIFORunnableEntry.emitter`.

Maybe I'm just missing something here?  In that case, any insight is greatly appreciated.

I've created an example that demonstrates the issue:

```
import android.os.Bundle;
import android.support.v7.app.AppCompatActivity;

import com.google.common.base.Strings;
import com.polidea.rxandroidble.RxBleClient;
import com.polidea.rxandroidble.scan.ScanSettings;

import com.example.R;
import com.example.MyApplication;
import rx.Observable;
import rx.Subscription;
import timber.log.Timber;

public class DebugLeakActivity extends AppCompatActivity {

    RxBleClient bleClient;
    Subscription subscription = null;
    String someActivityField;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_debug_leak);

        // in my app, RxBleClient is provided by dependency injection and has the lifetime of the app.
        // simulating here.
        bleClient = ((MyApplication)getApplication()).bleClient;

    }

    @Override
    protected void onStart() {
        super.onStart();

        subscription = scan()
                .subscribe(
                        // reference a field in the activity to create a reference to the Activity from the lambda
                        deviceString -> someActivityField = deviceString,
                        throwable -> Timber.e(throwable));
    }

    @Override
    protected void onDestroy() {

        if (subscription != null) {
            subscription.unsubscribe();
            subscription = null;
        }
        super.onDestroy();
    }

    public Observable<String> scan() {
        ScanSettings scanSettings = new ScanSettings.Builder()
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                .build();

        return bleClient.scanBleDevices(scanSettings)
                .doOnUnsubscribe(() -> Timber.d("scanBleDevices() unsubscribed"))
                .distinct()
                .doOnNext(scanResult -> Timber.d(String.format("Found BLE device: %s, %s", scanResult.getBleDevice().getMacAddress(), Strings.nullToEmpty(scanResult.getBleDevice().getName()))))
                .map(rxBleScanResult -> rxBleScanResult.toString());
    }
}
```

Here is the reference chain from LeakCanary:
```
In com.example:0.20.0-SNAPSHOT:200000.
* com.example.presentation.DebugLeakActivity has leaked:
* GC ROOT thread java.lang.Thread.<Java Local> (named 'Thread-8')
* references com.polidea.rxandroidble.internal.radio.FIFORunnableEntry.emitter
* references rx.internal.operators.OnSubscribeCreate$NoneEmitter.actual
* references rx.internal.operators.OperatorUnsubscribeOn$1.val$subscriber (anonymous subclass of rx.Subscriber)
* references rx.internal.operators.OnSubscribeMap$MapSubscriber.actual
* references rx.internal.operators.OperatorMerge$InnerSubscriber.parent
* references rx.internal.operators.OperatorMerge$MergeSubscriber.child
* references rx.observers.Subscribers$5.val$subscriber (anonymous subclass of rx.Subscriber)
* references rx.observers.Subscribers$5.val$subscriber (anonymous subclass of rx.Subscriber)
* references rx.observers.Subscribers$5.val$subscriber (anonymous subclass of rx.Subscriber)
* references rx.internal.operators.OperatorDistinct$1.val$child (anonymous subclass of rx.Subscriber)
* references rx.internal.operators.OnSubscribeDoOnEach$DoOnEachSubscriber.subscriber
* references rx.internal.operators.OnSubscribeMap$MapSubscriber.actual
* references rx.observers.SafeSubscriber.actual
* references rx.internal.util.ActionSubscriber.onNext
* references com.example.presentation.DebugLeakActivity$$Lambda$1.arg$1 (anonymous implementation of rx.functions.Action1)
* leaks com.example.presentation.DebugLeakActivity instance
```

I've attached the full LeakCanary details for more information:
[In 2.txt](https://github.com/Polidea/RxAndroidBle/files/1338937/In.2.txt)

dariuszseweryn, 2017-09-28:
Are you sure that the activity is destroyed by the time LeakCanary fires? (I have not used id so I do not know).

One thing you can do is to put a log in `.onDestroy()` and check if it is available in the logcat before the LeakCanary statement.

The other possibility would be to move unsubscription to an opposite event of `.onStart()` -> `.onStop()`.

And as a matter of fact even calling `.scan()` makes a reference to your Activity. :)

dariuszseweryn, 2017-10-02:
Ping?

HolySamosa, 2017-10-02:
Thanks for the suggestions.  I did some memory dumps and it seems that in some cases *some* of the Activities do eventually get garbage collected, but never all.  I'm going to experiment with moving the subscription / unsubscription to `.onStart()` and `onStop()` and will fill you in on the results.

dariuszseweryn, 2017-10-04:
Impatiently waiting. :) 👍

streetsofboston, 2017-10-12:
Looking at this code snippet:
```
    @Override
    protected void onStart() {
        super.onStart();

        subscription = scan()
                .subscribe(
                        // reference a field in the activity to create a reference to the Activity from the lambda
                        deviceString -> someActivityField = deviceString,
                        throwable -> Timber.e(throwable));
    }

    @Override
    protected void onDestroy() {

        if (subscription != null) {
            subscription.unsubscribe();
            subscription = null;
        }
        super.onDestroy();
    }
```

The `onStart` can be called multiple times before the `onDestroy` is finally called.

Instead of unsubscribing in the `onDestroy`, override the `onStop` instead and unsubscribe there.

dariuszseweryn, 2017-10-13:
I think that @streetsofboston comment may be the cause of the observed leak. Closing for now. If new information will be available—feel free to reopen.

closed 2017-10-13 by dariuszseweryn